loco-3d/crocoddyl #774: gepetto-gui core dumps on Ubuntu 18.04

GlenHenshaw, 2020-05-15:
robotpkg *wants* to install gepetto-viewer=4.4, but gepetto-viewer-corab requires gepetto-viewer=4.2.

Manually specifying gepetto-viewer=4.2 installs fine, but trying to run gepetto-gui results in:

   Gtk-Message: 15:47:42.823: Failed to load module "canberra-gtk-module"
   libpng warning: iCCP: known incorrect sRGB profile
   terminate called after throwing an instance of 'CORBA::TRANSIENT'
   Aborted (core dumped)

nim65s, 2020-05-15:
Hi @GlenHenshaw 
gepetto-viewer 4.4 is really old.
I think you are looking for `sudo apt install robotpkg-py27-qt4-gepetto-viewer-corba`
You can adapt your python version, but for Qt, we are still bound to ubuntu's packaging of OSG with only qt4.

GlenHenshaw, 2020-05-15:
I'm just barely starting to learn about robotpkg, but I have a fresh Ubuntu 18.04 install and I installed the current robotpkg from their apt repository. And 4.4 is what it wants to install.

Building everything from source -- all the urdfdom dependencies, then gepetto -- seems to work. But the instructions indicate you can install via robotpkg, and that doesn't appear to work at least for me.

I'm not trying to do anything exotic, just get the examples running...

GlenHenshaw, 2020-05-15:
On the other hand, installing from source and then trying to run one of the crocoddyl exaples gives me this:

    python -m crocoddyl.examples.quadrupedal_gaits "display" "plot"
    *** SOLVE walking ***
    Traceback (most recent call last):
      File "/home/glenhenshaw/anaconda3/envs/crocoddyl/lib/python3.6/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
      File "/home/glenhenshaw/anaconda3/envs/crocoddyl/lib/python3.6/runpy.py", line 85, in _run_code
    exec(code, run_globals)
      File "/opt/openrobots/lib/python3.6/site-packages/crocoddyl/examples/quadrupedal_gaits.py", line 104, in <module>
    display = crocoddyl.GepettoDisplay(anymal, 4, 4, cameraTF, frameNames=[lfFoot, rfFoot, lhFoot, rhFoot])
      File "/opt/openrobots/lib/python3.6/site-packages/crocoddyl/__init__.py", line 57, in __init__
    self.addRobot()
      File "/opt/openrobots/lib/python3.6/site-packages/crocoddyl/__init__.py", line 124, in addRobot
    self.robot.initViewer(windowName="crocoddyl", loadModel=False)
      File "/opt/openrobots/lib/python3.6/site-packages/pinocchio/robot_wrapper.py", line 238, in initViewer
        self.viz.initViewer(*args, **kwargs)
      File "/opt/openrobots/lib/python3.6/site-packages/pinocchio/visualize/gepetto_visualizer.py", line 28, in initViewer
        import gepetto.corbaserver
    ModuleNotFoundError: No module named 'gepetto'

nim65s, 2020-05-15:
`robotpkg-gepetto-corba` and `robotpkg-gepetto-viewer-corba` are obsolete packages. You can now choose your python and (soon) qt versions, with eg. `robotpkg-py27-qt4-gepetto-viewer-corba`.

Your ModuleNotFoundError is probably because the python sitelib of your installation (eg. `/opt/openrobots/lib/python2.7/site-packages` for robotpkg & python 2.7) is not available in the `sys.path` of your python interpreter. Please add it, with (by order of preference) a `.pth` file, or using the `PYTHONPATH`environment variable, or by manually using `sys.path.append()`.

proyan, 2020-05-15:
Hi @GlenHenshaw 
`gepetto-gui` is used for visualizations, which are indeed good to have :).
If you've followed the installation instructions, you would have access to `crocoddyl` already.

For installing gepetto-gui, please follow the instructions given by @nim65s `sudo apt install robotpkg-py27-qt4-gepetto-viewer-corba`

proyan, 2020-05-15:
Looking at your error log from when you install by source, it looks like your only problem is the missing gepetto-gui python package. You seem to have everything else already.

proyan, 2020-05-17:
Hi @GlenHenshaw 
Could you please verify that you have a functioning installation?

GlenHenshaw, 2020-05-18:
So when I install from the robotpkg-py36-qt4-gepetto-viewer-corba package the install succeeds. I can run the resulting binary and it does not core dump.

However, I have been spectacularly unsuccessful in getting crocoddyl to communicate with the viewer. Neither the command-line tests nor the jupyter notebook scripts provided in the docs work.

From the command line:
    $ python -m crocoddyl.examples.quadrupedal_gaits "display" "plot"
    Traceback (most recent call last):
      File "/home/glenhenshaw/anaconda3/lib/python3.7/runpy.py", line 183, in _run_module_as_main
    mod_name, mod_spec, code = _get_module_details(mod_name, _Error)
      File "/home/glenhenshaw/anaconda3/lib/python3.7/runpy.py", line 109, in _get_module_details
    __import__(pkg_name)
      File "/opt/openrobots/lib/python3.6/site-packages/crocoddyl/__init__.py", line 1, in <module>
    from .libcrocoddyl_pywrap import *
    ModuleNotFoundError: No module named 'crocoddyl.libcrocoddyl_pywrap'

And from the jupyter notebook:

    display = crocoddyl.MeshcatDisplay(talos_legs, 4, 4, False)
    ddp.setCallbacks([crocoddyl.CallbackLogger(),
                  crocoddyl.CallbackVerbose(),
                  crocoddyl.CallbackDisplay(display)])

    ---------------------------------------------------------------------------
    AttributeError                            Traceback (most recent call last)
    <ipython-input-22-cdeb5909be20> in <module>
    ----> 1 display = crocoddyl.MeshcatDisplay(talos_legs, 4, 4, False)
            2 #display = crocoddyl.GepettoDisplay(talos_legs)
            3 ddp.setCallbacks([crocoddyl.CallbackLogger(),
            4                   crocoddyl.CallbackVerbose(),
            5                   crocoddyl.CallbackDisplay(display)])

    AttributeError: module 'crocoddyl' has no attribute 'MeshcatDisplay'

It is therefore very difficult for me to determine whether I have a "functioning installation" or not...

GlenHenshaw, 2020-05-18:
Update: when I manually add the path to libcrocoddyl_pywrap.cpython-36m-x86_64-linux-gnu.so to my LD_LOAD_PATH (in my case, /opt/openrobots/lib/python3.6/site-packages/crocoddyl), then the command-line example works.

The jupyter notebook, which is what I would prefer to use, still does not work.

proyan, 2020-05-19:
> Update: when I manually add the path to libcrocoddyl_pywrap.cpython-36m-x86_64-linux-gnu.so to my LD_LOAD_PATH (in my case, /opt/openrobots/lib/python3.6/site-packages/crocoddyl), then the command-line example works.

If you already have the `/opt/openrobots/lib/python3.6/site-packages` taken care of by your PYTHONPATH, I don't think you would have needed to manually specify the crocoddyl folder for linking.

Looking at your error log, I have a feeling that it is because you seem to be using `python3.7` instead of `python3.6` here. (You are calling `/home/glenhenshaw/anaconda3/lib/python3.7/runpy.py`)

Could you please fix this and try again?

GlenHenshaw, 2020-05-19:
"Could you please fix this and try again?"

Yes -- I apologize, that was operator error. When I properly activate the Conda environment with python 3.6 *and* I properly install the most up-to-date crocoddyl packages, the command line example *mostly* works.

The problem I get now is that with the ANYmal example, it complains about not having the right plugin to read the model files.

The humanoid waling example works fine though.

proyan, 2020-05-19:
@GlenHenshaw You are turning out to be a good beta user :)

I think the issue is a missing `robotpkg-osg-dae`. Please install this package and let me know if there is any further issue?

GlenHenshaw, 2020-05-20:
Yes, that works!

GlenHenshaw, 2020-05-20:
I do still get the following error:

    Traceback (most recent call last):
      File "/home/glenhenshaw/anaconda3/envs/crocoddyl/lib/python3.6/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
      File "/home/glenhenshaw/anaconda3/envs/crocoddyl/lib/python3.6/runpy.py", line 85, in _run_code
    exec(code, run_globals)
      File "/opt/openrobots/lib/python3.6/site-packages/crocoddyl/examples/quadrupedal_gaits.py", line 143, in <module>
    title = phase.keys()[0] + " (phase " + str(i) + ")"
    TypeError: 'dict_keys' object does not support indexing

proyan, 2020-05-20:
#778 should take care of this last issue. It is because of change in the API between python 2 and 3.

Thanks for the reports @GlenHenshaw, please let us know if there are further issues. If not, could you please close this issue?

closed 2020-05-21 by GlenHenshaw