loco-3d/crocoddyl #775: Example notebook bipedal_walking.ipyn fails

GlenHenshaw, 2020-05-15:
display = crocoddyl.MeshcatDisplay(talos_legs, 4, 4, False)

    ddp.setCallbacks([crocoddyl.CallbackLogger(),

                  crocoddyl.CallbackVerbose(),

                  crocoddyl.CallbackDisplay(display)])

    ---------------------------------------------------------------------------
    AttributeError                            Traceback (most recent call last)
    <ipython-input-7-5eb8da7a93d3> in <module>
    ----> 1 display = crocoddyl.MeshcatDisplay(talos_legs, 4, 4, False)
      2 ddp.setCallbacks([crocoddyl.CallbackLogger(),
      3                   crocoddyl.CallbackVerbose(),
      4                   crocoddyl.CallbackDisplay(display)])

    AttributeError: module 'crocoddyl' has no attribute 'MeshcatDisplay'

proyan, 2020-05-19:
Thanks for the report @GlenHenshaw 
MeshcatDisplay was recently integrated to make the display easier with jupyter notebooks. However, it looks like the recent release `v1.2.1r3` in robotpkg only updated the examples, and not the python bindings. 
@nim65s Do you know why this is? This looks to be broken after the changes in #759.

proyan, 2020-05-19:
From #776 , you mentioned that you also installed crocoddyl from source. Are you able to run the examples with installation from source?

proyan, 2020-05-19:
My bad, robotpkg doesn't have meshcat support yet.
@GlenHenshaw do you get this error when running from source?

nim65s, 2020-05-19:
This has nothing to do with robotpkg. Meshcat is not available in any release yet.

I think @GlenHenshaw just cloned the project to get the notebook. And the clone is on devel by default. So the notebooks from a freshly cloned project are broken by default, because they are ahead of the binaries.

@GlenHenshaw : to fix this, you can use `git checkout master`.

@cmastalli @proyan : I don't think using devel as the main branch is a good idea. This will obviously keep creating this kind of situations, where a new user is totally mislead before even being able to check the most basic stuff about our software. This creates frustration because he doesn't understand if the errors are coming from his actions or if our software is broken. And here, we are guilty. Moreover, @GlenHenshaw (rightfully) need for our time, while he should be able to work on his own to get the examples work.

GlenHenshaw, 2020-05-19:
"you mentioned that you also installed crocoddyl from source. Are you able to run the notebooks with installation from source?"

I did it from source originally because I couldn't get it to install properly via robotpkg. But I never successfully got everything to build, I got hung up on building gepetto-viewer-corba. Then I figured out what was wrong with my robotpkg install, so I removed the built-from-source version and went with the apt-get robotpkg version.

That is the version that resulted in the bug reported above.

GlenHenshaw, 2020-05-19:
"So the notebooks from a freshly cloned project are broken by default, because they are ahead of the binaries."

This is precisely what I did - I wanted to the notebooks and I am trying to run them against the robotpkg binary.

cmastalli, 2020-05-19:
> @cmastalli @proyan : I don't think using devel as the main branch is a good idea. This will obviously keep creating this kind of situations, where a new user is totally mislead before even being [...]

@nim65s I understand your point, however, your propose solution doesn't necessary avoids these user mistakes. For instance, other problems could arise by exchanging versions of `robotpkg` binaries with source files. Maybe a solution could be to install the Jupyter notebooks as well, do @nim65s agree with it? If so, could you help me with this?

Regarding to the default branch topic, I agree that it is a partial plausible solution. However, I would like to keep the devel branch as default one. The reason is that I expect developers to create PRs directly to this branch. And to be honest, we have made this kind of mistakes in the past (inside gepgilab).

> This is precisely what I did - I wanted to the notebooks and I am trying to run them against the robotpkg binary.

I am happy that you have solved your problem. I hope you can find our source code helpful. @GlenHenshaw could we close this issue? Do you have other questions / issues?

GlenHenshaw, 2020-05-20:
Well, hold on - I still don't have a working notebook.

Is the recommendation that I install crocoddyl from source instead of binary? Or is there an older example notebook I could use?

proyan, 2020-05-20:
As @nim65s mentioned, the examples in the `devel` branch, (which you currently use) are ahead of the current binary release (which I suspect you are using). So please simply try to do 
```
git checkout master
```
and try the examples, using the current binary installation from robotpkg.

cmastalli, 2020-05-20:
> Well, hold on - I still don't have a working notebook.
> 
> Is the recommendation that I install crocoddyl from source instead of binary? Or is there an older example notebook I could use?

There are recent features that haven't been  released yet in robotpkg, e.g Meshcat viewer. If you cannot wait for the release, then you could install by source the devel branch. Note that for this particular case it would be enough with replacing the display with GepettoDisplay.

GlenHenshaw, 2020-05-21:
OK, so... I checkout out the master branch and used that notebook.

It doesn't work.

The notebook (bipedal_walking.ipyn) runs. It doesn't throw any errors. The plots look good. Gepetto-viewer shows the robot on the screen.

But it doesn't actually move. There's no animation. I assume there is supposed to be?

cmastalli, 2020-05-21:
> But it doesn't actually move. There's no animation. I assume there is supposed to be?

No, you should see the robot motion when you type `display.displayFromSolver(ddp, rate)`.

GlenHenshaw, 2020-05-21:
Nope... I get a static robot.

When I run examples from the command line I see a moving robot. When I run similar examples from the jupyter notebook I see a static robot.

GlenHenshaw, 2020-05-21:
In case it helps, when I run gepetto-viewer I get the following output

```Gtk-Message: 11:37:13.021: Failed to load module "canberra-gtk-module"
libpng warning: iCCP: known incorrect sRGB profile
Object::connect: No such signal QComboBox::currentTextChanged(QString)
Object::connect:  (receiver name: 'Highlight/State')
Object::connect: No such signal QComboBox::currentTextChanged(QString)
Object::connect:  (receiver name: 'LightingMode')
Object::connect: No such signal QComboBox::currentTextChanged(QString)
Object::connect:  (receiver name: 'Visibility')
Object::connect: No such signal QComboBox::currentTextChanged(QString)
Object::connect:  (receiver name: 'WireframeMode')
Object::connect: No such signal QComboBox::currentTextChanged(QString)
Object::connect:  (receiver name: 'Highlight/State')
Object::connect: No such signal QComboBox::currentTextChanged(QString)
Object::connect:  (receiver name: 'LightingMode')
Object::connect: No such signal QComboBox::currentTextChanged(QString)
Object::connect:  (receiver name: 'Visibility')
Object::connect: No such signal QComboBox::currentTextChanged(QString)
Object::connect:  (receiver name: 'WireframeMode')
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
'''

GlenHenshaw, 2020-05-21:
ANd when I run the notebook, I get the following output in the terminal:

```Object::connect: No such signal QComboBox::currentTextChanged(QString)
Object::connect:  (receiver name: 'Highlight/State')
Object::connect: No such signal QComboBox::currentTextChanged(QString)
Object::connect:  (receiver name: 'LightingMode')
Object::connect: No such signal QComboBox::currentTextChanged(QString)
Object::connect:  (receiver name: 'Visibility')
Object::connect: No such signal QComboBox::currentTextChanged(QString)
Object::connect:  (receiver name: 'WireframeMode')
Object::connect: No such signal QComboBox::currentTextChanged(QString)
Object::connect:  (receiver name: 'Highlight/State')
Object::connect: No such signal QComboBox::currentTextChanged(QString)
Object::connect:  (receiver name: 'LightingMode')
Object::connect: No such signal QComboBox::currentTextChanged(QString)
Object::connect:  (receiver name: 'Visibility')
Object::connect: No such signal QComboBox::currentTextChanged(QString)
Object::connect:  (receiver name: 'WireframeMode')
Object::connect: No such signal QComboBox::currentTextChanged(QString)
Object::connect:  (receiver name: 'Highlight/State')
Object::connect: No such signal QComboBox::currentTextChanged(QString)
Object::connect:  (receiver name: 'LightingMode')
Object::connect: No such signal QComboBox::currentTextChanged(QString)
Object::connect:  (receiver name: 'Visibility')
Object::connect: No such signal QComboBox::currentTextChanged(QString)
Object::connect:  (receiver name: 'WireframeMode')
Object::connect: No such signal QComboBox::currentTextChanged(QString)
Object::connect:  (receiver name: 'Highlight/State')
Object::connect: No such signal QComboBox::currentTextChanged(QString)
Object::connect:  (receiver name: 'LightingMode')
Object::connect: No such signal QComboBox::currentTextChanged(QString)
Object::connect:  (receiver name: 'Visibility')
Object::connect: No such signal QComboBox::currentTextChanged(QString)
Object::connect:  (receiver name: 'WireframeMode')
Object::connect: No such signal QComboBox::currentTextChanged(QString)
Object::connect:  (receiver name: 'Highlight/State')
Object::connect: No such signal QComboBox::currentTextChanged(QString)
Object::connect:  (receiver name: 'LightingMode')
Object::connect: No such signal QComboBox::currentTextChanged(QString)
Object::connect:  (receiver name: 'Visibility')
Object::connect: No such signal QComboBox::currentTextChanged(QString)
Object::connect:  (receiver name: 'WireframeMode')
Object::connect: No such signal QComboBox::currentTextChanged(QString)
Object::connect:  (receiver name: 'Highlight/State')
Object::connect: No such signal QComboBox::currentTextChanged(QString)
Object::connect:  (receiver name: 'LightingMode')
Object::connect: No such signal QComboBox::currentTextChanged(QString)
Object::connect:  (receiver name: 'Visibility')
Object::connect: No such signal QComboBox::currentTextChanged(QString)
Object::connect:  (receiver name: 'WireframeMode')
Object::connect: No such signal QComboBox::currentTextChanged(QString)
Object::connect:  (receiver name: 'Highlight/State')
Object::connect: No such signal QComboBox::currentTextChanged(QString)
Object::connect:  (receiver name: 'LightingMode')
Object::connect: No such signal QComboBox::currentTextChanged(QString)
Object::connect:  (receiver name: 'Visibility')
Object::connect: No such signal QComboBox::currentTextChanged(QString)
Object::connect:  (receiver name: 'WireframeMode')
Object::connect: No such signal QComboBox::currentTextChanged(QString)
Object::connect:  (receiver name: 'Highlight/State')
Object::connect: No such signal QComboBox::currentTextChanged(QString)
Object::connect:  (receiver name: 'LightingMode')
Object::connect: No such signal QComboBox::currentTextChanged(QString)
Object::connect:  (receiver name: 'Visibility')
Object::connect: No such signal QComboBox::currentTextChanged(QString)
Object::connect:  (receiver name: 'WireframeMode')
Object::connect: No such signal QComboBox::currentTextChanged(QString)
Object::connect:  (receiver name: 'Highlight/State')
Object::connect: No such signal QComboBox::currentTextChanged(QString)
Object::connect:  (receiver name: 'LightingMode')
[... 675 lines cut ...]
Object::connect:  (receiver name: 'LightingMode')
Object::connect: No such signal QComboBox::currentTextChanged(QString)
Object::connect:  (receiver name: 'Visibility')
Object::connect: No such signal QComboBox::currentTextChanged(QString)
Object::connect:  (receiver name: 'WireframeMode')
Object::connect: No such signal QComboBox::currentTextChanged(QString)
Object::connect:  (receiver name: 'Highlight/State')
Object::connect: No such signal QComboBox::currentTextChanged(QString)
Object::connect:  (receiver name: 'ImmediateMode')
Object::connect: No such signal QComboBox::currentTextChanged(QString)
Object::connect:  (receiver name: 'LightingMode')
Object::connect: No such signal QComboBox::currentTextChanged(QString)
Object::connect:  (receiver name: 'Visibility')
Object::connect: No such signal QComboBox::currentTextChanged(QString)
Object::connect:  (receiver name: 'WireframeMode')
Object::connect: No such signal QComboBox::currentTextChanged(QString)
Object::connect:  (receiver name: 'Highlight/State')
Object::connect: No such signal QComboBox::currentTextChanged(QString)
Object::connect:  (receiver name: 'ImmediateMode')
Object::connect: No such signal QComboBox::currentTextChanged(QString)
Object::connect:  (receiver name: 'LightingMode')
Object::connect: No such signal QComboBox::currentTextChanged(QString)
Object::connect:  (receiver name: 'Visibility')
Object::connect: No such signal QComboBox::currentTextChanged(QString)
Object::connect:  (receiver name: 'WireframeMode')

GlenHenshaw, 2020-05-21:
...but I get similar errors when running the example from the command line.

cmastalli, 2020-05-21:
@GlenHenshaw it seems there is something wrong in the gepetto-viewer installation.
It is hard to me understand how it is going on.

Could you check that you're not mixing Python versions of installed libraries?

GlenHenshaw, 2020-05-21:
I suspect that the issue may be how my python installation transfers from conda to my jupyter notebooks. I have a conda environment configured for crocoddyl. I instantiate jupyter from a terminal with that environment activated but I am not convinced notebook actually sees the python paths the same as the terminal does.

I am new at conda and fairly new at jupyter as well, so this might be pilot error on my part.

Glen
_________________________________________________________________

"A ship in port is safe, but that's not what ships are built for.
Sail out to sea and do new things."

                                - Admiral "Amazing" Grace Hopper, USN

nim65s, 2020-05-21:
Conda is cancer. Now, you are aware :)

cmastalli, 2020-05-21:
@GlenHenshaw we have decided to do not support Conda. In future we want to release Crocoddyl using pip, @nim65s is working hard on it.

I believe this issue is out of our scope, and we unfortunately have our hands tied.
I hope you can understand our point. To a smooth transition for you, I suggest you to clean up your installation and follow the instructions written in the REAME file.

GlenHenshaw, 2020-05-21:
1) Re: conda support. Fine. Personally I think that all python package management is cancer but I'm certainly not going to argue over that, I'll use whatever you guys support.

2) I seem to have solved my problem. I'm not sure exactly *how* I've solved it. What I did was open the bipedal_walking.py file as a notebook instead of opening the bipedal_walking.ipyn file. The code in these two files is note the same. In the first case the walking problem is divided up by gait phases and each gait phase is solved individually; the entire trajectory is the plotted and send to gepetto-viewer.

This works. The trajectory gets solved, and gepetto-viewer shows me an animation of it from the notebook.

I'm not sure why the ipyn version does not show any animation, but I note that in that file the walking problem is not divided up into gaits, and I wonder if perhaps what is actually being solved in that file is in fact a balancing problem? Or something?

In any case, I think I have enough working now that I can start my project.

cmastalli, 2020-05-22:
@GlenHenshaw I have a similar message error after updating `gepetto-viewer` and `gepett-viewer-corba` in my system (reported in https://github.com/Gepetto/gepetto-viewer-corba/issues/147).

Could you try to revert the versions (this works for me)?
```bash
sudo apt install robotpkg-py27-qt4-gepetto-viewer=4.8.0 robotpkg-py27-qt4-gepetto-viewer-corba=5.3.2
```

cmastalli, 2020-06-08:
@GlenHenshaw  do you still have this issue?

GlenHenshaw, 2020-06-08:
Yes and no. I have determined that the issue seems to be with the python code and not the notebook-gepetto interface. The code in the notebook example does not result in the robot moving, no matter where it is run. Other code (the example provided for humanoid walking that is a python script and not a notebook file) does result in robot motion whether it is run from the command line or from within a notebook.

cmastalli, 2020-10-02:
I will close this issue since I cannot reproduce it and there is no activity.
@GlenHenshaw feel free to re-open it :)

closed 2020-10-02 by cmastalli