loco-3d/crocoddyl #776: ValueError: Mesh package://example-robot-data/robots/ ...

syundo0730, 2020-05-17:
Hello. Error raises when I run the command bellow. Actually I couldn't launch any models.
```
ksyundo@ksyundo-MacBookPro:~$ python3 -m crocoddyl.examples.quadrupedal_gaits "display" "plot"

Traceback (most recent call last):
  File "/usr/lib/python3.5/runpy.py", line 184, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/lib/python3.5/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/opt/openrobots/lib/python3.5/site-packages/crocoddyl/examples/quadrupedal_gaits.py", line 17, in <module>
    anymal = example_robot_data.loadANYmal()
  File "/opt/openrobots/lib/python3.5/site-packages/example_robot_data/robots_loader.py", line 66, in loadANYmal
    robot = RobotWrapper.BuildFromURDF(modelPath + URDF_SUBPATH, [modelPath], pinocchio.JointModelFreeFlyer())
  File "/opt/openrobots/lib/python3.5/site-packages/pinocchio/robot_wrapper.py", line 17, in BuildFromURDF
    robot.initFromURDF(filename, package_dirs, root_joint, verbose, meshLoader)
  File "/opt/openrobots/lib/python3.5/site-packages/pinocchio/robot_wrapper.py", line 21, in initFromURDF
    model, collision_model, visual_model = buildModelsFromUrdf(filename, package_dirs, root_joint, verbose, meshLoader)
  File "/opt/openrobots/lib/python3.5/site-packages/pinocchio/shortcuts.py", line 46, in buildModelsFromUrdf
    geom_model = pin.buildGeomFromUrdf(model, filename, geometry_type, package_dirs)
ValueError: Mesh package://example-robot-data/robots/anymal_b_simple_description/meshes/base/anymal_base.dae could not be found.
```

Model files exists
```
ksyundo@ksyundo-MacBookPro:~$ ls /opt/openrobots/share/example-robot-data/robots
anymal_b_simple_description  hyq_description   kinova_description  talos_data
double_pendulum_description  icub_description  romeo_description   tiago_description
hector_description           iris_description  solo_description    ur_description
```

I'm using Ubuntu xenial and python 3.5.2
```
ksyundo@ksyundo-MacBookPro:~$ lsb_release -c
Codename:	xenial
ksyundo@ksyundo-MacBookPro:~$ python3 --version
Python 3.5.2
```

How can I fix the file path error?

cmastalli, 2020-05-17:
You need to define the ROS_PACKAGE_PATH. For a more detailed instruction, see the README file of the example-robot-data repository.

syundo0730, 2020-05-17:
it worked. Thank you!

closed 2020-05-17 by syundo0730

amirrezarazmjoo, 2021-01-22:
@syundo0730 How did you fix it? I have the same problem

syundo0730, 2021-01-22:
@amirrezarazmjoo I did what cmastalli told https://github.com/loco-3d/crocoddyl/issues/776#issuecomment-629758146

hjw-1014, 2021-05-17:
> it worked. Thank you!

Hi, can you tell me how did you define the ROS_PACKAGE_PATH? I don't find guidance in example-robot-data... Thank you very much.

cmastalli, 2021-05-18:
> Hi,@syundo0730, can you tell me how did you define the ROS_PACKAGE_PATH? I don't find guidance in example-robot-data... Thank you very much.

You need to include the folder where the mesh files are installed, i.e.
```bash
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:$INSTALLATION_PATH/share
```
where `$INSTALLATION_PATH` is `/opt/openrobots/` if you've installed the binaries with robotpkg.

hjw-1014, 2021-05-18:
@cmastalli Hi, thank you for your reply. I add this into `~/.bashrc` and resource it. And then I ran` gepetto-gui,` and tried to load robot like [this official video](https://homepages.laas.fr/jmirabel/raw/videos/hpp-gui-example.mp4), but there is no robot I can choose. What did I miss?

cmastalli, 2021-05-18:
> @cmastalli Hi, thank you for your reply. I add this into `~/.bashrc` and resource it. And then I ran` gepetto-gui,` and tried to load robot like [this official video](https://homepages.laas.fr/jmirabel/raw/videos/hpp-gui-example.mp4), but there is no robot I can choose. What did I miss?

I am not aware of how to do it. Perhaps @nim65s (or @jmirabel) knows the answer.

jmirabel, 2021-05-27:
The button is still here but the functionality should not be used. It was
meant to be used by plugins. It hasn't been updated when switching to
plugins.

Loading a robot shall be done using the python interface. If you want to
add buttons to the GUI, check the python python interface.