loco-3d/crocoddyl #777: Verifying Satisfaction of Constraints

julesser, 2020-05-18:
Hi all, 

I was wondering if there are already tools available in order to check if (or how good) constraints are actually satisfied from the optimal solution. It would be particularly interesting to know for

1. Inequality constraints (friction cone and joint limits)
2. Task constraints (e.g. tracking of desired foot placement)

Or do you validate the constraint satisfaction via simple post-processing?

cmastalli, 2020-05-18:
We don't have yet any code that post-process the constraint satisfaction.
However, you could retrieve it by accessing the value of the residual vector inside the cost functions.

julesser, 2020-07-21:
Does someone know how to retrieve the value of the residual vector for a specific cost term? 
I can access the cost terms via `solver.problem.runningModels[i].differential.costs.costs`

cmastalli, 2020-07-21:
> Does someone know how to retrieve the value of the residual vector for a specific cost term?
> I can access the cost terms via `solver.problem.runningModels[i].differential.costs.costs`

`solver.problem.runningDatas[i].differential.costs["myCostName"].r`

cmastalli, 2020-07-21:
Is this answer your question? Could we close this issue?

julesser, 2020-07-21:
Thanks for your convenient answers!

closed 2020-07-21 by julesser