loco-3d/crocoddyl #781: Release v1.3.0

cmastalli, 2020-05-21:
Hi @nim65s,

In this release, we have done the following tasks:
 1. Minor improvement in state-base class (enabling limits once we define only one component).
 2. Added functions to print the active/inactive status of costs, contacts and impulses.
 3. Running computations in the contact / impulse dynamics using the active number of contacts / impulses, respectively.
 4. Used NumPy Array in examples / notebooks.
 5. Initial integration of Meshcat viewer and used in notebooks.
 6. Improved efficiency of `Jdiff`, `Jintegrate` in multibody state.
 7. Added an extra operator in `Jintegrate` signature (state classes)
 8. Added an extra function parallel transport in the state class (i.e. `JintegrateTransport`)
 9. Added functions to retrieve the inactive costs, contacts and impulses.
 10. Fixed target_link_libraries use
11. Improved efficiency of few activation models.
12. Added the action-base class for code generation + unittest.
13. Added codegen bench for 4 dof arm and biped
14. Renamed all bench files.
15. Updated dependency versions for Pinocchio and example-robot-data.
16. Fixed bug in the impulse dynamics that appears in multiple threads.
17. Fixed various issues regarding data alignment.
 18. Added doxygen file with the documentation state-base class.
19. Fixed an issue when we updates reference in frame-placement and frame-rotation costs.
20. Fixed an issue with Python3 compatibility in few examples.
21. Proper display of friction cones + do not display when the contact / impulse is inactive/
22. Extended the computation of impulses derivatives.
23. Added a friction cost for impulse dynamics.

Note that I have created the branch as we used to do: release/v1.2.0. Additionally, @proyan and I have decided to follow your suggestion (i.e. defining the master branch as default). I would handle this task once these stuffs are released in robotpkg.

Yours,
Carlos

nim65s, 2020-05-21:
Thanks. This is holliday here, so I'll handle that on Monday :)

cmastalli, 2020-05-21:
> Thanks. This is holliday here, so I'll handle that on Monday :)

Sounds good! I hope you're enjoying your holidays!

closed 2020-05-27 by nim65s

cmastalli, 2020-05-27:
Thank you @nim65s :)