loco-3d/crocoddyl #785: Cannot install robotpkg-collada-dom to visualize the quadruped example

zzhou387, 2020-06-01:
Hello,

So I got the same issue in #774 when I was trying to visualize the quadruped example:
`File: /opt/openrobots/share/example-robot-data/robots/anymal_b_simple_description/meshes/base/anymal_base.dae could not be loaded
You may try to convert the file with the following command:
osgconv /opt/openrobots/share/example-robot-data/robots/anymal_b_simple_description/meshes/base/anymal_base.dae /opt/openrobots/share/example-robot-data/robots/anymal_b_simple_description/meshes/base/anymal_base.dae.osgb
File /opt/openrobots/share/example-robot-data/robots/anymal_b_simple_description/meshes/base/anymal_base.dae found but could not be opened. Check that a plugin exist.
`
And I tried to install `robotpkg-osg-dae` but got this error:
`The following packages have unmet dependencies:
 robotpkg-osg-dae : Depends: robotpkg-collada-dom (= 2.3.1r2) but 2.4.0 is to be installed
E: Unable to correct problems, you have held broken packages.
`
Is there a way to fix it? Thanks!

cmastalli, 2020-06-01:
This issue is related with recent problem in the binaries of `gepetto-viewer`.
Hopefully, it is already solved, right @nim65s?

@zzhou387 if you're urgent, then you could try to convert the dae files manually as the error message suggests you, i.e.
```
sudo osgconv /opt/openrobots/share/example-robot-data/robots/anymal_b_simple_description/meshes/base/anymal_base.dae /opt/openrobots/share/example-robot-data/robots/anymal_b_simple_description/meshes/base/anymal_base.dae.osgb
```

The biped / humanoid examples don't use collada files. So, they should work for you :)

nim65s, 2020-06-01:
@zzhou387 : can you try `sudo apt install robotpkg-collada-dom=2.3.1r2`, and see if this is better ?

@cmastalli : I asked you to try that, to check if this can be a workaround until the bigger issue is fixed. ref https://github.com/Gepetto/gepetto-viewer/pull/143

zzhou387, 2020-06-01:
> @zzhou387 : can you try `sudo apt install robotpkg-collada-dom=2.3.1r2`, and see if this is better ?
> 
> @cmastalli : I asked you to try that, to check if this can be a workaround until the bigger issue is fixed. ref [Gepetto/gepetto-viewer#143](https://github.com/Gepetto/gepetto-viewer/pull/143)

Yea it works when I install 2.3.1r2. Thanks!

cmastalli, 2020-06-04:
@nim65s sorry for the delay. I was extremely busy.

This solution doesn't work for me, this is the error message:
```
COMM_FAILURE                              Traceback (most recent call last)
/home/cmastalli/devel/loco-mpc/examples/test_quadruped.py in <module>()
     71     ddp.solve(xs_0, us_0, 1, False, reg)
     72     reg = ddp.u_reg
---> 73     display.displayFromSolver(ddp)

/usr/local/lib/python2.7/dist-packages/crocoddyl/__init__.py in displayFromSolver(self, solver, factor)
     44         models = solver.problem.runningModels + [solver.problem.terminalModel]
     45         dts = [m.dt if hasattr(m, "differential") else 0. for m in models]
---> 46         self.display(solver.xs, fs, ps, dts, factor)
     47         if numpy_conversion:
     48             numpy_conversion = False

/usr/local/lib/python2.7/dist-packages/crocoddyl/__init__.py in display(self, xs, fs, ps, dts, factor)
    149                         self._setConeMu(key, f["mu"])
    150                         self.robot.viewer.gui.applyConfiguration(
--> 151                             frictionName, list(np.array(pinocchio.SE3ToXYZQUAT(position)).squeeze()))
    152                         self.robot.viewer.gui.setVisibility(frictionName, "ON")
    153                         self.activeContacts[key] = True

/opt/openrobots/lib/python2.7/site-packages/gepetto/corbaserver/graphical_interface_idl.pyc in applyConfiguration(self, *args)
    484 
    485     def applyConfiguration(self, *args):
--> 486         return self._obj.invoke("applyConfiguration", _0_gepetto.corbaserver.GraphicalInterface._d_applyConfiguration, args)
    487 
    488     def applyConfigurations(self, *args):
```
and if I downgrade gepetto-viewer and gepetto-viewer-corba as
```
sudo apt install robotpkg-py27-qt4-gepetto-viewer=4.8.0 robotpkg-py27-qt4-gepetto-viewer-corba=5.3.2
```
then it doesn't crash but it cannot render the DAE files (no robot is displayed).

nim65s, 2020-06-04:
`COMM_FAILURE` ? This has nothing to do with collada. Can you just try again ?

cmastalli, 2020-06-08:
> `COMM_FAILURE` ? This has nothing to do with collada. Can you just try again ?

I tried again after upgrading `gepetto-viewer` and `gepetto-viewer-corba` to 4.9.0 and 5.4.0, respectively, and it didn't work. Concretely speaking, I don't have `COMM_FAILURE` but still I cannot display the collada files.

I don't have any idea what it is going on.

zzhou387, 2020-07-23:
Hello, thanks for all the help before. Right now the downgraded collada-dom=2.3.1r2 and osg-dae works fine for Anymal example. However, when I was trying to import my own robot model, I can only visualize part of the links and the mesh files are all dae files. Geptto-gui is reporting errors similar to before in terms of the broken links:
`Unable to locate visual scene!
Load failed in COLLADA DOM conversion
Warning: Error in reading to "/home/ziyi/crocoddyl/examples/A1/laikago_description/meshes/trunk.dae".
File: /home/ziyi/crocoddyl/examples/A1/laikago_description/meshes/trunk.dae could not be loaded
You may try to convert the file with the following command:
osgconv /home/ziyi/crocoddyl/examples/A1/laikago_description/meshes/trunk.dae /home/ziyi/crocoddyl/examples/A1/laikago_description/meshes/trunk.dae.osgb
File /home/ziyi/crocoddyl/examples/A1/laikago_description/meshes/trunk.dae found but could not be opened. Check that a plugin exist.
Reset up_axis to Z_UP.
You must initialize a Geode on world/robot/visuals/FL_hip_0 to use Alpha
Reset up_axis to Z_UP.
You must initialize a Geode on world/robot/visuals/FL_thigh_0 to use Alpha
`
I am pretty sure the dae files should work because it works well in Rviz. So I tried to convert the dae files to osgb as @cmastalli  suggested before but I failed to convert them after installing openscenegraph. I tried both `sudo apt-get install openscenegraph`  and `sudo apt-get install robotpkg-openscenegraph` (I have to remove osg-dae package when installing robotpkg-openscenegraph)
`ziyi@ziyi-XPS-8930:~/crocoddyl/examples/A1$ osgconv /home/ziyi/crocoddyl/examples/A1/laikago_description/meshes/calf.dae /home/ziyi/crocoddyl/examples/A1/laikago_description/meshes/calf.dae.osgb
Unable to locate visual scene!
Load failed in COLLADA DOM conversion
Warning: Error in reading to "/home/ziyi/crocoddyl/examples/A1/laikago_description/meshes/calf.dae".
Error no data loaded.
`
I am not quite familiar with openscenegraph. Am I doing the correct way to convert dae files?  Is there any other workaround to visualize them? Really Appreciate!

cmastalli, 2020-07-23:
From this message:
```Reset up_axis to Z_UP.```
It seems the normal vectors that describes each triangle in the mesh are going inward. You can change them using blender.

However, @nim65s might provide you precise information about it.

zzhou387, 2020-07-23:
> From this message:
> `Reset up_axis to Z_UP.`
> It seems the normal vectors that describe each triangle in the mesh are going inward. You can change them using blender.
> 
> However, @nim65s might provide you precise information about it.

Thanks for the prompt reply. I checked the dae files in blender and it seems like it is already Z up. The good thing is that I exported the dae files to be obj files and the geptto-gui can visualize it now! I think there is still a problem there when parsing the dae files in geptto-gui but everything works okay now. Thanks anyway.

cmastalli, 2020-07-24:
I will close this issue. If you are still having problems with gepetto viewer, then I suggest you to create an issue there. Definitely you will have a better support, none of the Crocoddyl developers knows the implementation details of Gepetto viewer neither are maintainers of it.

Feel free to reopen it if you have troubles with the Crocoddyl interface to it ;)

closed 2020-07-24 by cmastalli