loco-3d/crocoddyl #786: Load Model: How to Fix Joints at Specified Values?

julesser, 2020-06-05:
Hi all, 

I was wondering whether it's possible to load a full urdf model and then selectively fix some joints at some specified values?

Right now I am using the robot wrapper that comes along with Pinnochio in order to build a floating base model of a humanoid robot like this:
```
rh5_robot = RobotWrapper.BuildFromURDF(modelPath + URDF_SUBPATH, [modelPath], pinocchio.JointModelFreeFlyer())
rmodel = rh5_robot.model
addFreeFlyerJointLimits(rmodel)
```

jcarpent, 2020-06-05:
This issue is related to Pinocchio much more. You may have a look to the `uildReducedModel`.

cmastalli, 2020-06-05:
Yes, indeed, I suggest you to post it in Pinocchio if you still have questions regarding this topic.

I will close to this issue.

closed 2020-06-05 by cmastalli