loco-3d/crocoddyl #791: pybind / python unittests fail when building from source

skleff1994, 2020-06-19:
Hello,

I am trying to install Crocoddyl from source on Ubuntu 18.04 with the multi-threading option. I cloned the `devel` branch and did

```
cmake .. -DBUILD_WITH_MULTITHREADS=ON
make
make install
make test
```
I get the following output

```
Running tests...
/usr/bin/ctest --force-new-ctest-process 
Test project /home/sebastien/crocoddyl/cmake
      Start  1: test_states
 1/39 Test  #1: test_states ...................................   Passed    0.20 sec
      Start  2: test_actuation
 2/39 Test  #2: test_actuation ................................   Passed    0.04 sec
      Start  3: test_activations
 3/39 Test  #3: test_activations ..............................   Passed    0.02 sec
      Start  4: test_costs
 4/39 Test  #4: test_costs ....................................***Failed    2.28 sec
      Start  5: test_cost_sum
 5/39 Test  #5: test_cost_sum .................................   Passed    3.98 sec
      Start  6: test_contacts
 6/39 Test  #6: test_contacts .................................   Passed    0.14 sec
      Start  7: test_impulses
 7/39 Test  #7: test_impulses .................................   Passed    0.10 sec
      Start  8: test_multiple_contacts
 8/39 Test  #8: test_multiple_contacts ........................   Passed    1.23 sec
      Start  9: test_multiple_impulses
 9/39 Test  #9: test_multiple_impulses ........................   Passed    1.23 sec
      Start 10: test_actions
10/39 Test #10: test_actions ..................................   Passed    0.02 sec
      Start 11: test_diff_actions
11/39 Test #11: test_diff_actions .............................   Passed    0.39 sec
      Start 12: test_friction_cone
12/39 Test #12: test_friction_cone ............................   Passed    0.00 sec
      Start 13: test_boxqp
13/39 Test #13: test_boxqp ....................................   Passed    0.00 sec
      Start 14: test_solvers
14/39 Test #14: test_solvers ..................................   Passed    3.44 sec
      Start 15: pybinds-states
15/39 Test #15: pybinds-states ................................***Failed    0.49 sec
      Start 16: pybinds-actuations
16/39 Test #16: pybinds-actuations ............................***Failed    0.23 sec
      Start 17: pybinds-actions
17/39 Test #17: pybinds-actions ...............................***Failed    0.23 sec
      Start 18: pybinds-shooting
18/39 Test #18: pybinds-shooting ..............................***Failed    0.19 sec
      Start 19: pybinds-solvers
19/39 Test #19: pybinds-solvers ...............................***Failed    0.22 sec
      Start 20: pybinds-costs
20/39 Test #20: pybinds-costs .................................***Failed    0.24 sec
      Start 21: pybinds-contacts
21/39 Test #21: pybinds-contacts ..............................***Failed    0.23 sec
      Start 22: pybinds-impulses
22/39 Test #22: pybinds-impulses ..............................***Failed    0.23 sec
      Start 23: pybinds-force_cost
23/39 Test #23: pybinds-force_cost ............................***Failed    0.18 sec
      Start 24: pybinds-friction_cone_cost
24/39 Test #24: pybinds-friction_cone_cost ....................***Failed    0.18 sec
      Start 25: pybinds-momemtum_cost
25/39 Test #25: pybinds-momemtum_cost .........................***Failed    0.24 sec
      Start 26: pybinds-impulse_com_cost
26/39 Test #26: pybinds-impulse_com_cost ......................***Failed    0.19 sec
      Start 27: pybinds-squashing
27/39 Test #27: pybinds-squashing .............................   Passed    0.26 sec
      Start 28: example-python-double_pendulum
28/39 Test #28: example-python-double_pendulum ................***Failed    0.28 sec
      Start 29: example-python-boxfddp_vs_boxddp
29/39 Test #29: example-python-boxfddp_vs_boxddp ..............***Failed    0.32 sec
      Start 30: example-python-quadrotor
30/39 Test #30: example-python-quadrotor ......................***Failed    0.24 sec
      Start 31: example-python-quadrotor_ubound
31/39 Test #31: example-python-quadrotor_ubound ...............***Failed    0.23 sec
      Start 32: example-python-arm_manipulation
32/39 Test #32: example-python-arm_manipulation ...............***Failed    0.22 sec
      Start 33: example-python-quadrupedal_gaits
33/39 Test #33: example-python-quadrupedal_gaits ..............***Failed    0.31 sec
      Start 34: example-python-quadrupedal_walk_ubound
34/39 Test #34: example-python-quadrupedal_walk_ubound ........***Failed    0.31 sec
      Start 35: example-python-bipedal_walk
35/39 Test #35: example-python-bipedal_walk ...................***Failed    0.34 sec
      Start 36: example-python-bipedal_walk_ubound
36/39 Test #36: example-python-bipedal_walk_ubound ............***Failed    0.34 sec
      Start 37: example-python-humanoid_manipulation
37/39 Test #37: example-python-humanoid_manipulation ..........***Failed    0.35 sec
      Start 38: example-python-humanoid_manipulation_ubound
38/39 Test #38: example-python-humanoid_manipulation_ubound ...***Failed    0.37 sec
      Start 39: example-python-humanoid_taichi
39/39 Test #39: example-python-humanoid_taichi ................***Failed    0.38 sec

36% tests passed, 25 tests failed out of 39

Total Test time (real) =  19.91 sec

The following tests FAILED:
	  4 - test_costs (Failed)
	 15 - pybinds-states (Failed)
	 16 - pybinds-actuations (Failed)
	 17 - pybinds-actions (Failed)
	 18 - pybinds-shooting (Failed)
	 19 - pybinds-solvers (Failed)
	 20 - pybinds-costs (Failed)
	 21 - pybinds-contacts (Failed)
	 22 - pybinds-impulses (Failed)
	 23 - pybinds-force_cost (Failed)
	 24 - pybinds-friction_cone_cost (Failed)
	 25 - pybinds-momemtum_cost (Failed)
	 26 - pybinds-impulse_com_cost (Failed)
	 28 - example-python-double_pendulum (Failed)
	 29 - example-python-boxfddp_vs_boxddp (Failed)
	 30 - example-python-quadrotor (Failed)
	 31 - example-python-quadrotor_ubound (Failed)
	 32 - example-python-arm_manipulation (Failed)
	 33 - example-python-quadrupedal_gaits (Failed)
	 34 - example-python-quadrupedal_walk_ubound (Failed)
	 35 - example-python-bipedal_walk (Failed)
	 36 - example-python-bipedal_walk_ubound (Failed)
	 37 - example-python-humanoid_manipulation (Failed)
	 38 - example-python-humanoid_manipulation_ubound (Failed)
	 39 - example-python-humanoid_taichi (Failed)
Errors while running CTest
Makefile:109: recipe for target 'test' failed
make: *** [test] Error 8

```

Then I tried the same installation _without_ the multi-threading option and it still fails. I also (naively) tried with an additional -DPYTHON_EXECUTABLE='/usr/bin/python2' to see if it would change anything, but it doesn't. I guess it's somehow related to my installation, but I'm not sure how to debug this. Do you have any idea where the problem may come from ? Should I use the master branch instead ?

cmastalli, 2020-06-19:
All unittests are passing in non multithreading mode. Are you sure you don't have multiple installed copies of Crocoddyl.

For the multhreading mode, there is a known issue that needs to be tackle. The problem is that some of the python unittests creates internal data (not possible to avoid). However, it could be possible to create different copies of a Python model, which could be handle inside the code after creating objects that are copyable or pickleable.

skleff1994, 2020-06-19:
I made sure to uninstall Crocoddyl before building from source and was able to reproduced the same output. In non multithreading mode I get the following

```
Running tests...
/usr/bin/ctest --force-new-ctest-process 
Test project /home/sebastien/crocoddyl/cmake
      Start  1: test_states
 1/39 Test  #1: test_states ...................................   Passed    0.14 sec
      Start  2: test_actuation
 2/39 Test  #2: test_actuation ................................   Passed    0.04 sec
      Start  3: test_activations
 3/39 Test  #3: test_activations ..............................   Passed    0.01 sec
      Start  4: test_costs
 4/39 Test  #4: test_costs ....................................   Passed    2.21 sec
      Start  5: test_cost_sum
 5/39 Test  #5: test_cost_sum .................................   Passed    4.01 sec
      Start  6: test_contacts
 6/39 Test  #6: test_contacts .................................   Passed    0.14 sec
      Start  7: test_impulses
 7/39 Test  #7: test_impulses .................................   Passed    0.10 sec
      Start  8: test_multiple_contacts
 8/39 Test  #8: test_multiple_contacts ........................   Passed    1.23 sec
      Start  9: test_multiple_impulses
 9/39 Test  #9: test_multiple_impulses ........................   Passed    1.23 sec
      Start 10: test_actions
10/39 Test #10: test_actions ..................................   Passed    0.02 sec
      Start 11: test_diff_actions
11/39 Test #11: test_diff_actions .............................   Passed    0.39 sec
      Start 12: test_friction_cone
12/39 Test #12: test_friction_cone ............................   Passed    0.00 sec
      Start 13: test_boxqp
13/39 Test #13: test_boxqp ....................................   Passed    0.00 sec
      Start 14: test_solvers
14/39 Test #14: test_solvers ..................................   Passed    6.79 sec
      Start 15: pybinds-states
15/39 Test #15: pybinds-states ................................***Failed    0.24 sec
      Start 16: pybinds-actuations
16/39 Test #16: pybinds-actuations ............................***Failed    0.28 sec
      Start 17: pybinds-actions
17/39 Test #17: pybinds-actions ...............................***Failed    0.29 sec
      Start 18: pybinds-shooting
18/39 Test #18: pybinds-shooting ..............................***Failed    0.31 sec
      Start 19: pybinds-solvers
19/39 Test #19: pybinds-solvers ...............................***Failed    0.32 sec
      Start 20: pybinds-costs
20/39 Test #20: pybinds-costs .................................***Failed    0.34 sec
      Start 21: pybinds-contacts
21/39 Test #21: pybinds-contacts ..............................***Failed    0.32 sec
      Start 22: pybinds-impulses
22/39 Test #22: pybinds-impulses ..............................***Failed    0.32 sec
      Start 23: pybinds-force_cost
23/39 Test #23: pybinds-force_cost ............................***Failed    0.25 sec
      Start 24: pybinds-friction_cone_cost
24/39 Test #24: pybinds-friction_cone_cost ....................***Failed    0.24 sec
      Start 25: pybinds-momemtum_cost
25/39 Test #25: pybinds-momemtum_cost .........................***Failed    0.32 sec
      Start 26: pybinds-impulse_com_cost
26/39 Test #26: pybinds-impulse_com_cost ......................***Failed    0.24 sec
      Start 27: pybinds-squashing
27/39 Test #27: pybinds-squashing .............................   Passed    0.29 sec
      Start 28: example-python-double_pendulum
28/39 Test #28: example-python-double_pendulum ................***Failed    0.29 sec
      Start 29: example-python-boxfddp_vs_boxddp
29/39 Test #29: example-python-boxfddp_vs_boxddp ..............***Failed    0.33 sec
      Start 30: example-python-quadrotor
30/39 Test #30: example-python-quadrotor ......................***Failed    0.23 sec
      Start 31: example-python-quadrotor_ubound
31/39 Test #31: example-python-quadrotor_ubound ...............***Failed    0.23 sec
      Start 32: example-python-arm_manipulation
32/39 Test #32: example-python-arm_manipulation ...............***Failed    0.24 sec
[... 27 lines cut ...]
	 22 - pybinds-impulses (Failed)
	 23 - pybinds-force_cost (Failed)
	 24 - pybinds-friction_cone_cost (Failed)
	 25 - pybinds-momemtum_cost (Failed)
	 26 - pybinds-impulse_com_cost (Failed)
	 28 - example-python-double_pendulum (Failed)
	 29 - example-python-boxfddp_vs_boxddp (Failed)
	 30 - example-python-quadrotor (Failed)
	 31 - example-python-quadrotor_ubound (Failed)
	 32 - example-python-arm_manipulation (Failed)
	 33 - example-python-quadrupedal_gaits (Failed)
	 34 - example-python-quadrupedal_walk_ubound (Failed)
	 35 - example-python-bipedal_walk (Failed)
	 36 - example-python-bipedal_walk_ubound (Failed)
	 37 - example-python-humanoid_manipulation (Failed)
	 38 - example-python-humanoid_manipulation_ubound (Failed)
	 39 - example-python-humanoid_taichi (Failed)
Errors while running CTest
Makefile:109: recipe for target 'test' failed
make: *** [test] Error 8
```

Concerning the multithreading mode, I tried to run a simple program despite the failure of the unitttests and I got memory allocations errors upon instantiation of `ShootingProblem` and `Solver` objects (both in python and C++). Is this related to the known issue you are mentioning ? In this case, could you be more specific about how to circumvent this issue ? 

Besides, before opening this issue I also tried installing from the `master` branch (with and without multithreading). Python unittests failed similarly in both cases, however I was able to use in my simple program without any memory error...

cmastalli, 2020-06-19:
It is hard to know what it's going on, you could run the test with in verbose mode: `ctest -v`. However, given the recent changes in `example-robot-data`, I would say that you need to check that it is updated and working the repository. For instance, could you run the Crocoddyl examples?

> Concerning the multithreading mode, I tried to run a simple program despite the failure of the unitttests and I got memory allocations errors upon instantiation of `ShootingProblem` and `Solver` [...]

Again, it is hard to give you an answer without further details. Which program were you running? Which action model were you trying?

Let me try to explain you the issue. We cannot derived action data in an action model written in Python, and this is not possible to solver it. Therefore, in the action models written in Python, we need to allocate internal data. However, if you create one action model and use it for each node, then when we run a multithreading application, the data used in a given node is updated in parallel by another node. This results in a wrong computation.

So, you need, as an user, to create a copy of the action model written in Python per each node. To do it easily, we need to enable the copying of the action models. This is in my TODO list, but not priority.

skleff1994, 2020-06-22:
Thank you for your detailed answer. So if I understand correctly, this memory allocation issue only affects action models created in Python, and the fix would be to enable copying these action models in order to avoid conflicts in the data when running the computations in parallel. 

> It is hard to know what it's going on, you could run the test with in verbose mode: ctest -v. However, given the recent changes in example-robot-data, I would say that you need to check that it is updated and working the repository. For instance, could you run the Crocoddyl examples?

Let me be try to be more specific : I made sure `example-robot-data` works fine and ran the tests with `ctest -v` (built from `devel` branch with multi-threading on)

```
Test project /home/sebastien/crocoddyl/cmake
      Start  1: test_states
 1/39 Test  #1: test_states ...................................   Passed    0.13 sec
      Start  2: test_actuation
 2/39 Test  #2: test_actuation ................................   Passed    0.04 sec
      Start  3: test_activations
 3/39 Test  #3: test_activations ..............................   Passed    0.01 sec
      Start  4: test_costs
 4/39 Test  #4: test_costs ....................................***Failed    2.18 sec
      Start  5: test_cost_sum
 5/39 Test  #5: test_cost_sum .................................   Passed    4.15 sec
      Start  6: test_contacts
 6/39 Test  #6: test_contacts .................................   Passed    0.14 sec
      Start  7: test_impulses
 7/39 Test  #7: test_impulses .................................   Passed    0.10 sec
      Start  8: test_multiple_contacts
 8/39 Test  #8: test_multiple_contacts ........................   Passed    1.23 sec
      Start  9: test_multiple_impulses
 9/39 Test  #9: test_multiple_impulses ........................   Passed    1.24 sec
      Start 10: test_actions
10/39 Test #10: test_actions ..................................   Passed    0.02 sec
      Start 11: test_diff_actions
11/39 Test #11: test_diff_actions .............................   Passed    0.42 sec
      Start 12: test_friction_cone
12/39 Test #12: test_friction_cone ............................   Passed    0.00 sec
      Start 13: test_boxqp
13/39 Test #13: test_boxqp ....................................   Passed    0.00 sec
      Start 14: test_solvers
14/39 Test #14: test_solvers ..................................   Passed    3.45 sec
      Start 15: pybinds-states
15/39 Test #15: pybinds-states ................................   Passed    1.20 sec
      Start 16: pybinds-actuations
16/39 Test #16: pybinds-actuations ............................***Exception: Child aborted  0.56 sec
      Start 17: pybinds-actions
17/39 Test #17: pybinds-actions ...............................***Exception: SegFault  0.60 sec
      Start 18: pybinds-shooting
18/39 Test #18: pybinds-shooting ..............................***Exception: SegFault  0.49 sec
      Start 19: pybinds-solvers
19/39 Test #19: pybinds-solvers ...............................***Exception: SegFault  0.60 sec
      Start 20: pybinds-costs
20/39 Test #20: pybinds-costs .................................***Exception: SegFault  4.50 sec
      Start 21: pybinds-contacts
21/39 Test #21: pybinds-contacts ..............................***Exception: SegFault  1.55 sec
      Start 22: pybinds-impulses
22/39 Test #22: pybinds-impulses ..............................***Exception: SegFault  1.44 sec
      Start 23: pybinds-force_cost
23/39 Test #23: pybinds-force_cost ............................***Exception: Child aborted  0.73 sec
      Start 24: pybinds-friction_cone_cost
24/39 Test #24: pybinds-friction_cone_cost ....................***Exception: Child aborted  0.71 sec
      Start 25: pybinds-momemtum_cost
25/39 Test #25: pybinds-momemtum_cost .........................   Passed    2.73 sec
      Start 26: pybinds-impulse_com_cost
26/39 Test #26: pybinds-impulse_com_cost ......................***Exception: Child aborted  0.64 sec
      Start 27: pybinds-squashing
27/39 Test #27: pybinds-squashing .............................   Passed    0.25 sec
      Start 28: example-python-double_pendulum
28/39 Test #28: example-python-double_pendulum ................***Exception: Child aborted  0.59 sec
      Start 29: example-python-boxfddp_vs_boxddp
29/39 Test #29: example-python-boxfddp_vs_boxddp ..............***Exception: Child aborted  2.88 sec
      Start 30: example-python-quadrotor
30/39 Test #30: example-python-quadrotor ......................   Passed    0.20 sec
      Start 31: example-python-quadrotor_ubound
31/39 Test #31: example-python-quadrotor_ubound ...............   Passed    0.30 sec
[... 144 lines cut ...]
    runningCostModel.get()->addCost("endeff", goalTrackingCost, cost_weights(0));
    runningCostModel.get()->addCost("stateReg", xRegCost, cost_weights(1));
    runningCostModel.get()->addCost("ctrlReg", uRegCost, cost_weights(2));
    terminalCostModel.get()->addCost("endeff", goalTrackingCost, cost_weights(3));

    // Create action model
    boost::shared_ptr<crocoddyl::ActuationModelFull> actuation = boost::make_shared<crocoddyl::ActuationModelFull>(state);
    boost::shared_ptr<crocoddyl::ActionModelAbstract> runningModel =
        boost::make_shared<crocoddyl::IntegratedActionModelEuler>(boost::make_shared<crocoddyl::DifferentialActionModelFreeFwdDynamics>(state, actuation, runningCostModel), dt);
    boost::shared_ptr<crocoddyl::ActionModelAbstract> terminalModel =
[... cut ...]
This code works fine, except when calling `ddp_->solve(x_traj_, u_traj_, maxit_);` which also causes a segmentation fault error. 

It seems however that my problem is not related to multi-threading, since as I mentioned, I encounter exactly the same issues running the same programs when building without multi-threading mode. But these programs work fine when Crocoddyl is installed from robotpkg, so I guess it's related to my installation somehow... Is there additionnal tests / information I can provide to help spotting where the issue(s) may come from ?

cmastalli, 2020-06-23:
> Thank you for your detailed answer. So if I understand correctly, this memory allocation issue only affects action models created in Python, and the fix would be to enable copying these action models in order to avoid conflicts in the data when running the computations in parallel.

Yes, it is correct.

> The Crocoddyl examples `arm_manipulation` and `quadrotor` work fine but not the other ones. For instance when I run `make examples-humanoid_taichi`

All the examples are passing, with exception of `example-python-double-pendulum`. To be precise, the unittests that should fail, only under multithreading compilation, are:
  - `pybinds-shooting`
  - `example-python-double-pendulum`

> It seems however that my problem is not related to multi-threading, since as I mentioned, I encounter exactly the same issues running the same programs when building without multi-threading mode. [...]

I think you should use `gdb` to find where your issue is happening. This might help us to spot the problem. Another action could be to check which libraries are linked using `ldd` to be sure, for instance, that you don't have multiple copies of Pinocchio.

Btw, could you provide more details of your setup (OS, Pinocchio version, example-robot-data version, etc)?

skleff1994, 2020-06-24:
Thank you for your advices. It appears that re-installing Crocoddyl from scratch solved the problem :

```
Test project /home/sebastien/crocoddyl/cmake
      Start  1: test_states
 1/39 Test  #1: test_states ...................................   Passed    0.15 sec
      Start  2: test_actuation
 2/39 Test  #2: test_actuation ................................   Passed    0.04 sec
      Start  3: test_activations
 3/39 Test  #3: test_activations ..............................   Passed    0.01 sec
      Start  4: test_costs
 4/39 Test  #4: test_costs ....................................   Passed    2.66 sec
      Start  5: test_cost_sum
 5/39 Test  #5: test_cost_sum .................................   Passed    4.10 sec
      Start  6: test_contacts
 6/39 Test  #6: test_contacts .................................   Passed    0.15 sec
      Start  7: test_impulses
 7/39 Test  #7: test_impulses .................................   Passed    0.11 sec
      Start  8: test_multiple_contacts
 8/39 Test  #8: test_multiple_contacts ........................   Passed    1.24 sec
      Start  9: test_multiple_impulses
 9/39 Test  #9: test_multiple_impulses ........................   Passed    1.24 sec
      Start 10: test_actions
10/39 Test #10: test_actions ..................................   Passed    0.03 sec
      Start 11: test_diff_actions
11/39 Test #11: test_diff_actions .............................   Passed    0.39 sec
      Start 12: test_friction_cone
12/39 Test #12: test_friction_cone ............................   Passed    0.01 sec
      Start 13: test_boxqp
13/39 Test #13: test_boxqp ....................................   Passed    0.01 sec
      Start 14: test_solvers
14/39 Test #14: test_solvers ..................................   Passed   32.31 sec
      Start 15: pybinds-states
15/39 Test #15: pybinds-states ................................   Passed    2.19 sec
      Start 16: pybinds-actuations
16/39 Test #16: pybinds-actuations ............................   Passed    2.00 sec
      Start 17: pybinds-actions
17/39 Test #17: pybinds-actions ...............................   Passed    3.74 sec
      Start 18: pybinds-shooting
18/39 Test #18: pybinds-shooting ..............................***Exception: Child aborted  0.83 sec
      Start 19: pybinds-solvers
19/39 Test #19: pybinds-solvers ...............................   Passed    2.57 sec
      Start 20: pybinds-costs
20/39 Test #20: pybinds-costs .................................   Passed    5.48 sec
      Start 21: pybinds-contacts
21/39 Test #21: pybinds-contacts ..............................   Passed    1.32 sec
      Start 22: pybinds-impulses
22/39 Test #22: pybinds-impulses ..............................   Passed    1.75 sec
      Start 23: pybinds-force_cost
23/39 Test #23: pybinds-force_cost ............................   Passed    0.79 sec
      Start 24: pybinds-friction_cone_cost
24/39 Test #24: pybinds-friction_cone_cost ....................   Passed    0.70 sec
      Start 25: pybinds-momemtum_cost
25/39 Test #25: pybinds-momemtum_cost .........................   Passed    2.99 sec
      Start 26: pybinds-impulse_com_cost
26/39 Test #26: pybinds-impulse_com_cost ......................   Passed    0.58 sec
      Start 27: pybinds-squashing
27/39 Test #27: pybinds-squashing .............................   Passed    0.26 sec
      Start 28: example-python-double_pendulum
28/39 Test #28: example-python-double_pendulum ................***Exception: Child aborted  0.63 sec
      Start 29: example-python-boxfddp_vs_boxddp
29/39 Test #29: example-python-boxfddp_vs_boxddp ..............   Passed    5.23 sec
      Start 30: example-python-quadrotor
30/39 Test #30: example-python-quadrotor ......................   Passed    0.23 sec
      Start 31: example-python-quadrotor_ubound
31/39 Test #31: example-python-quadrotor_ubound ...............   Passed    0.31 sec
      Start 32: example-python-arm_manipulation
32/39 Test #32: example-python-arm_manipulation ...............   Passed    0.47 sec
      Start 33: example-python-quadrupedal_gaits
33/39 Test #33: example-python-quadrupedal_gaits ..............   Passed    4.17 sec
      Start 34: example-python-quadrupedal_walk_ubound
34/39 Test #34: example-python-quadrupedal_walk_ubound ........   Passed    3.51 sec
      Start 35: example-python-bipedal_walk
35/39 Test #35: example-python-bipedal_walk ...................   Passed    2.10 sec
      Start 36: example-python-bipedal_walk_ubound
36/39 Test #36: example-python-bipedal_walk_ubound ............   Passed    3.95 sec
      Start 37: example-python-humanoid_manipulation
37/39 Test #37: example-python-humanoid_manipulation ..........   Passed   32.60 sec
      Start 38: example-python-humanoid_manipulation_ubound
38/39 Test #38: example-python-humanoid_manipulation_ubound ...   Passed   26.30 sec
      Start 39: example-python-humanoid_taichi
39/39 Test #39: example-python-humanoid_taichi ................   Passed   91.73 sec

95% tests passed, 2 tests failed out of 39

Total Test time (real) = 238.90 sec

The following tests FAILED:
	 18 - pybinds-shooting (Child aborted)
	 28 - example-python-double_pendulum (Child aborted)
Errors while running CTest
```

I did :

```
make clean
sudo make uninstall
rm -rf cmake
git pull
git submodule update --init
cd cmake/ 
cmake .. -DBUILD_WITH_MULTITHREADS=ON 
make 
make install 
```

My best guess is that I must have forgotten to clean the `cmake` directory in prior installations...

cmastalli, 2020-06-24:
Yes, sometimes `cmake clean` is not enough. I always recommend to to `rm -rf *`.

Since everything seems to be resolved, could you close this issue?

closed 2020-06-24 by skleff1994