loco-3d/crocoddyl #795: Memory leak issue

gfadini, 2020-07-02:
I encountered this problem since I have to solve a great number of crocoddyl fddp problems in loop.
I found that after some iterations during an optimization routine the memory consumption increased significantly.

To test this I'm attaching some examples of the robot manipulation in a loop.
https://gist.github.com/gfadini/df00b6d8a245acf40d68771403946eb8
In one of them the problem is initialized and solved, in the other initialized and assigned to a variable and then solved.
In both case there seem to be the same unexpected behaviour.
To isolate the issue I put the pinocchio model initialization outside the loops and try to call the del() method on the ddp class.
This doesn't same to change the outcome that the memory allocation to the ddp problem is not freed.
If you can reproduce the examples, you should see the increase of memory consumption with top, htop, mprofile or similia.

I'm not sure if it's related to crocoddyl itself or the python bindings but I'd need to have this issue checked since it's quite limiting in my study case.

jcarpent, 2020-07-02:
This might be related to https://github.com/stack-of-tasks/eigenpy/issues/156. Please consider including `eigenpy/eigenpy.hpp` inside a `fwd.hpp` for the Python bindings.
The issue is coming from Boost.Python which has a specific way of destroying the memory of allocated objects. In the present context, we need to overload the destroying of `Eigen::MatrixBase` which is only done by including the mentioned file.

gfadini, 2020-07-02:
@jcarpent Hi, thanks for the quick reply. So to apply this fix I should compile eigenpy from scratch?
At the moment I was still working with the up-to-date binary version of crocoddyl for convenience.
I also mention that in my original problem I also need to continuously modify the pinocchio model in the loop so that too would be another possible source of leak if I don't apply the fix in the issue you linked https://github.com/stack-of-tasks/eigenpy/issues/156
Am I correct on this guess?

nim65s, 2020-07-02:
@gfadini : I can try to reproduce your issue and check @jcarpent fix if you prefer

jcarpent, 2020-07-02:
No, Python bindings of Croco should include the header that I've mentioned.

gfadini, 2020-07-02:
@nim65s ok for me if you can do it it would be the best to quickly check if this solves the problem.
However as discussed I was planning to compile crocoddyl also on the second pc sooner or later

gfadini, 2020-07-02:
@jcarpent is there any reason/downside that you are aware of for which they were not included in the first place?

jcarpent, 2020-07-02:
I don't have any answer for your question.

gfadini, 2020-07-02:
Ok, @jcarpent for the reply. I guess that if what you propose is solving this problem without too much hassle it'll be the way to go.
Thanks

nim65s, 2020-07-03:
Actually, the fix proposed by @jcarpent is already in place, and I can reproduce @gfadini issue with it:

https://github.com/loco-3d/crocoddyl/blob/c31a27432f9f2b365faec31b5e7cb37d90b7abb0/bindings/python/crocoddyl/fwd.hpp#L12

cmastalli, 2020-07-03:
@gfadini I think you might want to try the Python code reported in Pinocchio, just to check that you don't have that issue in your systems. If this passes, then try again at Pinocchio level.

Please report the results in this thread.

gfadini, 2020-07-03:
Hi @cmastalli ,

I freshly installed the system at the lab and everything there is installed with the apt packages provided through robotpkg. I hardly doubt the issue is coming from the configuration there, that would be **extremely** unlikely.

To further support this conclusion I'm able to reproduce this specific issue with 3 different machines at the moment, on my two personal ones with compiled versions and at the lab with the packaged one. I guess that this, together with independent @nim65s test, adds to the fact that the issue is configuration independent already.

In addition to this, on my personal laptop I cannot reproduce https://github.com/stack-of-tasks/eigenpy/issues/156 anymore hinting that the fix that @jcarpent is already in place as @nim65s said. I attach you the plot and the slightly modified version of https://github.com/stack-of-tasks/eigenpy/issues/156 code used for the little test.

https://gist.github.com/gfadini/f8c3922e315e0f90b00d0f2648f3c291
[image]

cmastalli, 2020-07-03:
@gfadini could you conduct three other test? 
1) use an unicycle model instead of the FreeFwdDyn + euler integrator
2) use a DiffLQR instead of FreeFwdDyn one
3) Instead of creating each time a model inside  make_problem, let's use a fixed running and terminal model.

Please report the results of each case

gfadini, 2020-07-04:
@cmastalli 
Sure I can help you to isolate the problem, the scripts tried are here https://gist.github.com/gfadini/df00b6d8a245acf40d68771403946eb8

RESULTS

# Test 1

- no memory leak if the unicycle is solved in the cycles ```test1.py```
[image]

- memory leak if the models are initialized inside the loop ```test1_variant.py```
[image]

# Test 2

- memory leak ```test2.py```
[image]

# Test 3

- memory leak in both cases model outside ```test3.py``` or inside the loop ```test3_original.py```, comparing those it seems that there's no difference in the allocation just in the computation time
[image]
[image]

jcarpent, 2020-07-09:
> Actually, the fix proposed by @jcarpent is already in place, and I can reproduce @gfadini issue with it:
> 
> https://github.com/loco-3d/crocoddyl/blob/c31a27432f9f2b365faec31b5e7cb37d90b7abb0/bindings/python/crocoddyl/fwd.hpp#L12

Sorry @nim65s, but this file is not visible for the function instiation, meaning that all the `bp::def` should see the file `<eigenpy/eigenpy.hpp>`

cmastalli, 2020-07-09:
> Sorry @nim65s, but this file is not visible for the function instiation, meaning that all the `bp::def` should see the file `<eigenpy/eigenpy.hpp>`

So, where should we include it?

nmansard, 2020-07-17:
My understanding is that we should include the fwd declaration "#include <eigenpy/eigenpy.hpp>"  in all crocoddyl bindings, the same way Pinocchio bindings are doing. @jcarpent is that correct?
Is somebody taking care for that in Crocoddyl?

jcarpent, 2020-07-17:
Yes, this is exactly that.

jcarpent, 2020-07-17:
I think anyone is able to do that, this is a very simple task. @gfadini Are you willing to fix this bug?

cmastalli, 2020-07-17:
> I think anyone is able to do that, this is a very simple task. @gfadini Are you willing to fix this bug?

@gfadini I could help with this task since I understood how to solve. But at least I will need that you test it for me.

gfadini, 2020-07-17:
Hi, @jcarpent   @cmastalli  well this bug is something that is preventing me to move forward with some larger scale tests so I'd need it to be solved for sure.
I think that also some other gepetto team members may benefit from this.
However I'm not sure if I'm able to fix it by myself since I've never actively coded with crocoddyl in C++, but I can try my best.
If @cmastalli  understood how to fix the issue that would be ideal and I can help him to test the fix

jcarpent, 2020-07-17:
@gfadini This is a very simple fix. Just put the `#include <eigenpy/eigenpy.hpp>` in all the cpp files of the Croco bindings. Don't be afraid ;)

cmastalli, 2020-07-17:
@jcarpent the solution is shorter by adding the header in specifics files, not all of the `cpp` ones. @gfadini I will handle today for you :)

jcarpent, 2020-07-17:
In fact, you should do as in Pinocchio, including the `fwd.hpp` everywhere. You have to do that for all the functions having an Eigen argument which is something like: `Eigen::MatrixBase<>`

gfadini, 2020-07-20:
Hi, thanks @cmastalli and @jcarpent  for the support. 
I've applied the #800 and run the test as before but there's no improvement with respect to this specific issue. Apparently the only case that is not affected by the issue is still the test1.py where the same ddp unicycle problem is solved in a loop.
Even when the simple unicycle ddp problem is initialized and solved in a loop there's the memory leak. Can this in some way give us some hints on the cause of this? Can we somehow isolate better the root cause of this issue?

gfadini, 2020-07-20:
@cmastalli so I tried what you suggested, and the problem is still showing up in the case the ShootingProblem is created in a loop
[image]

cmastalli, 2020-07-21:
The memory leaks occur in the custom converter from  Python list `std::vector<T>` (the same for dictionary / std map). In Crocoddyl we have many places where we used them:
  - stack of action models / datas in `ShootingProblem`
  - warm-start of the solvers
  - `calc`, `calcDiff` functions of the `ShootingProblem`, among others

**Explanation**
In the custom converter, you could see how I allocate memory here: https://github.com/loco-3d/crocoddyl/blob/master/bindings/python/crocoddyl/utils/vector-converter.hpp#L83. Furthermore, I would expect that BoostPython API de-allocate this memory behind the scene, but contrary to good software engineering practice, BoostPython seems to be badly design. So it seems it doesn't de-allocate the memory.

I don't know how to solve this issue without sacrificing the Pythonic capabilities of Crocoddyl. Note that another alternative could be to use the `vector_indexing_suite` interface. I would prefer to avoid this solution since it will also introduce a change in the Python API.

@jcarpent, @jmirabel, @nim65s do you have any advice?

jcarpent, 2020-07-21:
Use the one from Pinocchio!

cmastalli, 2020-07-21:
> Use the one from Pinocchio!

Yes, indeed this is the alternative that I referred. However, it will break the Python API. I will use it as last resource.
@jcarpent do you have experience building custom converters? I have checked if someone has faced this issue without any success.

jcarpent, 2020-07-21:
In Pinocchio, you can see std::vector as list, and this is the way to do to avoid memory allocation and having shared memory between both. Again, see how I do in Pinocchio.

jcarpent, 2020-07-21:
> Use the one from Pinocchio!

Pinocchio does it well at this level.

cmastalli, 2020-07-21:
> In Pinocchio, you can see std::vector as list, and this is the way to do to avoid memory allocation and having shared memory between both. Again, see how I do in Pinocchio.

As far I know, Pinocchio defines an extra class that has the the option of retrieving a a dictionary. Then if I am correct, the API will change to something like this
```python
import crocoddyl
import numpy as np

model = crocoddyl.ActionModelUnicycle()
x0 = np.array([1, 1, 0])
u0 = np.array([0, 0])
modelList = crocoddyl.StdVec_Action()
for i in range(10):
  modelList.append(model)
problem = crocoddyl.ShootingProblem(x0, modelList, model) # instead of crocoddyl.ShootingProblem(x0, [model] * 10, model)

xsList = crocoddyl.StdVec_VectorX()
usList = crocoddyl.StdVec_VectorX()
for i in range(10):
  xsList.append(x0)
  usList.append(u0)
xsList.append(x0)

crocoddyl.SolverDDP(problem)
crocoddyl.solve(xsList, usList) # instead of crocoddyl.solve([x0] * 11, [us0] * 10)
```

jcarpent, 2020-07-21:
No, not necessarily. There is a mechanism to transform the list as an std::vector and then returning the right results.

cmastalli, 2020-07-22:
This issue was solved in #802.

@jcarpent thank you for the letting me know the additional mechanism developed in Pinocchio. Indeed the PR follows the same approach :)

cmastalli, 2020-07-22:
solved in #802

closed 2020-07-22 by cmastalli