loco-3d/crocoddyl #799: Downcasting classes in python

proyan, 2020-07-16:
We have a good set of python bindings, where we are able to create classes in python and use them as they are. However, there is one problem with the current bindings, where we are not able to downcast from an Abstract class to the class instance that is being developed. 

For example, when creating a class `IntegratedActionModelEuler` in python, the `calc(self, data, x, u)` needs access to the object `differential` in `IntegratedActionDataEuler`. However, `data` object is of type `ActionDataAbstract`, and thus doesn't contain any element of such name. In c++, we get around it by statically casting `data` to its expected type, and using this type to obtain our element. However, in python, such runtime polymorphism is not yet implemented.

The solution for this would be https://www.boost.org/doc/libs/1_55_0/libs/smart_ptr/pointer_cast.html, which would enable us to downcast pointers in python, in a visitor fashion for pre-defined types.

cmastalli, 2020-07-16:
In c++, I have enable the automatic downcast of Python bings. You could observe this but typing:
```python
actionModel.state # retrieve the derived state implemented in c++
```

However, it is right that we cannot derive data classes in Python. If you look the code, I didn't define `createData` as a virtual function for the Python bindings. So, what it happens is we create the base data for any model written in Python.
To be honest, I have tried to defined this function as virtual when I started the project, but I didn't manage to figure out a solution. The main reason is that data structures only define new data objects, not virtual functions.

In short, I don't think the this limitations is related with static casting of shared pointers. Perhaps, you might need to explain it more.

nmansard, 2020-07-17:
I think I agree with your remark, @cmastalli. Here, the problem is that createData is not virtual, and the re-implementation in the inheriting class is not called. Consequently, a ActionData is created instead of an InheritedActionData.

proyan, 2020-07-17:
> The main reason is that data structures only define new data objects, not virtual functions
I don't understand why this would prevent `createData` from being binded as virtual. Could you please explain?

gfadini, 2020-07-17:
Related to this issue I report a minimal case in which it can be seen that the ```ShootingProblem``` is not using the definition of the ```createData``` method in the definition of the abstract class.
While in python this is possible to do and set the differential ad an attribute of a custom defined Data class

```[python]
import crocoddyl, example_robot_data

class IntegratedActionModelBase(crocoddyl.ActionModelAbstract):
    '''
        Same as integrated action model euler
    '''
    def __init__(self, diffModel, timeStep=1e-3, withCostResiduals=True):
            crocoddyl.ActionModelAbstract.__init__(self, crocoddyl.StateVector(diffModel.state.nx), diffModel.nu)
            self.differential = diffModel
            self.timeStep = timeStep
            self.withCostResiduals = withCostResiduals
            if self.timeStep == 0:
                self.enable_integration_ = False
            else:
                self.enable_integration_ = True

    def createData(self):
        '''
            Not called by the Shooting problem
        '''
        data = IntegratedActionDataBase(self)
        print(data, 'But correctly called by python')
        return data

    def calc(self, data, x, u=None):
        print('...')
        return 0, 0

    def calcDiff(self, data, x, u=None):
        print('...')

class IntegratedActionDataBase(crocoddyl.ActionDataAbstract):
    '''
        Ideally would create a data class with differential alternative to the IAMEuler, not working with C++ bindings
    '''
    def __init__(self, am):
        crocoddyl.ActionDataAbstract.__init__(self, am)
        self.differential = am.differential.createData()

robot_model = example_robot_data.loadTalosArm().model
state = crocoddyl.StateMultibody(robot_model)
runningCostModel = crocoddyl.CostModelSum(state)
actuation = crocoddyl.ActuationModelFull(state)
differential=crocoddyl.DifferentialActionModelFreeFwdDynamics(state, actuation, runningCostModel)
IAMBase=IntegratedActionModelBase(differential)
problem=crocoddyl.ShootingProblem(state.rand(), [IAMBase], IAMBase)
myData=IAMBase.createData()
assert hasattr(myData, 'differential'), 'No differential here'
assert hasattr(problem.runningDatas[0], 'differential'), 'No differential here'
```

cmastalli, 2020-07-27:
@gfadini it seems I found the solution. I am working on it, it requires to do work on each single thing: action, diff-action, cost, contacts, impulses, actuation and activation

cmastalli, 2020-07-27:
solved in #808

closed 2020-07-27 by cmastalli