loco-3d/crocoddyl #807: General underactuated root actuation model

gfadini, 2020-07-27:
Hi,

I need a model of a robot for which there's an under-actuated base free to move along a prismatic guide.
Now one solution would be to develop a custom model from the ```ActuationAbstract``` and this would lead to writing a new actuation model either with bindings or in C++.
However I think that all this not necessary, and the class ```ActuationModelFloatingBase``` can be extended to include a more general under-actuated robot class. I guess that the change in this will also have a positive impact for other classes that would have by default the same C++ implementation.
I get the impression that it would be enough removing all the 1st joint DOFs avoiding the check for it to be a free-flyer.
Am I making a wrong assumption? Are there other parts of the code that would be affected by such change or doing internal checks?

cmastalli, 2020-07-27:
What it could be done is to parse the first joint of the URDF?
Then, we could be able to define any kind of floating-base joint (i.e. not only the free-flyer case).
Is this what you need?

gfadini, 2020-07-27:
Hi @cmastalli, yes that's roughly what I had in mind. With the URDF the thing should be the same, just that now I have a pinocchio model built from scratch. The class ```ActuationModelFloatingBase``` by itself can deal with a wider set of robots than what is implemented just by processing the model differently and neglecting always the first joint DOF

cmastalli, 2020-07-27:
When is your deadline? or is this something that you feel you can modify inside Crocoddyl?

gfadini, 2020-07-28:
Hi @cmastalli at the moment I can simply prototype with the bindings,
In the floating-base.hpp there's a check for the first joint to be explicitly of the free flyer type
```
explicit ActuationModelFloatingBaseTpl(boost::shared_ptr<StateMultibody> state) : Base(state, state->get_nv() - 6) {
    pinocchio::JointModelFreeFlyerTpl<Scalar> ff_joint;
    if (state->get_pinocchio()->joints[1].shortname() != ff_joint.shortname()) {
      throw_pretty("Invalid argument: "
                   << "the first joint has to be free-flyer");
    }
  };
```
I think this can be generalized to also for the prismatic types, it was a general idea that would require little effort and that I may try

cmastalli, 2020-07-28:
> I think this can be generalized to also for the prismatic types, it was a general idea that would require little effort and that I may try

Yes, that's correct!
You could create an ongoing PR. With it, we could give you a precised feedback.

gfadini, 2020-07-31:
Fixed in #816

closed 2020-07-31 by gfadini