loco-3d/crocoddyl #810: [NumDiff / Maybe other files as well] calcDiff dependency on calc should be enforced

ksyy, 2020-07-28:
Hi,

I recently ran into some problems while doing the [Cartpole Jupyter notebook](https://github.com/loco-3d/crocoddyl/blob/devel/examples/notebooks/cartpole_swing_up.py) provided as tutorials for Crocoddyl, because of the *not-so-well* documented dependency that the `calcDiff` function of `DifferentialActionModelNumDiff` has on its own `calc` function.

Indeed, if one calls [`calcDiff`](https://github.com/loco-3d/crocoddyl/blob/13fbdc2d2a419a426b6d0ec246d52c1bd1d030e7/include/crocoddyl/core/numdiff/diff-action.hxx#L46) without calling [`calc`](https://github.com/loco-3d/crocoddyl/blob/13fbdc2d2a419a426b6d0ec246d52c1bd1d030e7/include/crocoddyl/core/numdiff/diff-action.hxx#L28) beforehand, `data_0` is zero-initialised and therefore, all computations that follow are wrong.
I have nothing against that except that there is no warning at all, and the [solution provided](https://github.com/loco-3d/crocoddyl/blob/devel/examples/notebooks/solutions/cartpole_fxfu.py) falls into this trap.

I think there should be a `recalc=True` parameter somewhere in this function, or at least something clear to tell the user that he might not receive a meaningful result.

What do you think about that?

cmastalli, 2020-07-28:
Hi @kssy,

We have removed `recalc=True` to avoid branchiation when we code generate a `calcDiff` function. It is expected to run first `calc`. Furthermore, this behavior should be followed in the numdiff routine.

It is truth that we are missing to add documentation in many places of the library. This is a difficult task since we are at the same time pushing forward our research projects (we lack of resources).

Feel free to help us with the documentation, this is an entry task that could be done without knowing the complex implementation details.

cmastalli, 2020-07-28:
I will close this issue since this request goes against our design choice.

Feel free to make questions if you still have doubts about our design choice.

In any case, thank you for making propositions.

nim65s, 2020-07-28:
Well, isn't this still wrong https://github.com/loco-3d/crocoddyl/blob/devel/examples/notebooks/solutions/cartpole_fxfu.py ?

cmastalli, 2020-07-28:
> Well, isn't this still wrong https://github.com/loco-3d/crocoddyl/blob/devel/examples/notebooks/solutions/cartpole_fxfu.py ?

Oh, I didn't see this issue. I understood the idea was to change the design choice. I will fix it immediately so we can integrate it in the release

cmastalli, 2020-07-28:
done. I pushed it in the release branch. I will handle it at the devel branch now

ksyy, 2020-07-29:
> Feel free to help us with the documentation, this is an entry task that could be done without knowing the complex implementation details.

I disagree. Without looking at the implementation details, I could not have known what the problem was with this call to `calcDiff`.

cmastalli, 2020-07-29:
> I disagree. Without looking at the implementation details, I could not have known what the problem was with this call to `calcDiff`.

Now you know it, the proposition was after giving you the explanation :)

nmansard, 2020-07-30:
A minimal check could be to verify (assert) that calc was at least called once before calling calcDiff. It is not covering all the scenarios, but at least makes sure that the user is aware of the existence of the calc function. What do you think?

cmastalli, 2020-07-31:
> A minimal check could be to verify (assert) that calc was at least called once before calling calcDiff. It is not covering all the scenarios, but at least makes sure that the user is aware of the existence of the calc function. What do you think?

I am not sure about this. It will require to define an extra boolean variable for each single data structure (action, multiple contacts / impulses / costs, contacts, ...

In contrast, I believe the solution is having a well documented library. To do that, we need to do:
  1. doxygen documentation of c++ classes
  2. (maybe) revise the documentation in the Python bindings
  3. have notebooks that explain this internal design choices

proyan, 2020-07-31:
I agree with Carlos here. I think what is missing here is the documentation. We have some similar issues in pinocchio, where some function calls are dependant on other functions being called first. However, the documentation clearly specifies that in pinocchio's case.

cmastalli, 2020-10-28:
Added the requested documentation in #876

closed 2020-10-28 by cmastalli