loco-3d/crocoddyl #819: About cancel 'crocoddyl.switchToNumpyMatrix()'

ddliugit, 2020-07-30:
I need to integrate several piece of codes and is it possible to cancel the function `crocoddyl.switchToNumpyMatrix()` after declaring the `crocoddyl.switchToNumpyMatrix()` in the code?

cmastalli, 2020-07-31:
Hi @ddliugit,

We indeed want to deprecate each piece of code using numpy matrix. We have these dependencies in internal code: unittest and examples. I just wanted to point out that the library itself doesn't depend on it.

Saying that, which code are you referring? Are you using the devel branch?

Feel free to help us to migrate some code to numpy array.

ddliugit, 2020-08-07:
Thanks. I am using the code `biped.py` and previously used some code with old version of pinocchio.

I figured that out and I will close it.

closed 2020-08-07 by ddliugit