loco-3d/crocoddyl #827: Add frame translation or rotation bound

ddliugit, 2020-08-07:
Is there a API to add the quadratic barrier function with a upper or lower bound to a specific frame?

cmastalli, 2020-08-07:
> Is there a API to add the quadratic barrier function with a upper or lower bound to a specific frame?

It is possible to add barrier to a given reference frame placement. You need to use the `CostModelFramePlacement` cost and the `ActivationModelQuadraticBarrier` activation.

> Its purpose is to avoid geometry collision with ground or self -collision.

The above solution won't avoid collision with a specific geometry. However, it is possible to do it but this is ongoing work.

Is this answer your questions?

ddliugit, 2020-08-08:
Yes. It solved my problem well.

closed 2020-08-08 by ddliugit