loco-3d/crocoddyl #829: Line contact issue

ddliu365, 2020-08-10:
I want to add a line contact for an endeffector but  I didn't find such a contact type in crocoddyl.
I found that for a quadruped we have a 3D contact `ContactModel3D` and for a humanoid we have a 6D contact `ContactModel6D`.

So I try to use 2-3D contacts to represent a line contact(similiar to a hinge joint on the ground), however, the optimization failed.

Can you help to solve this issue?

cmastalli, 2020-08-11:
A line contact constrains 5 dimensions, it could only rotate along the axis. Using two 3d contacts impose 6 constraints, however, and depending on how they are configured, one (or a few) of them could be redundant.

This certainly will produce a numerical issue with our contact dynamics since the analytical solution cannot remove this extra redundancy. In other words, the analytical solution assumes that there is no redundancy.

Saying that, we always need to build the exact Jacobian, which in your case is 5D. Unfortunately, this is not a trivial task for an entry level user. You need to understand Pinocchio and Crocoddyl (and especially spatial algebra). The good thing is you could prototype in Python before going to c++.

As far I can tell you, nobody is working (have priority) to develop such kind of contact. However, it would be very welcome if you contribute in this topic.

If you want to do it, I could give you basic guidelines of how to implement it in Crocoddyl. I apologize if we cannot give you a stronger support on this topic.

jcarpent, 2020-08-17:
> A line contact constrains 5 dimensions, it could only rotate along the axis. Using two 3d contacts impose 6 constraints, however, and depending on how they are configured, one (or a few) of them could be redundant.

This analysis is wrong. The two 3d contacts should be sufficient to mimic the hinge joint. The issue may come from some numerical issues or from the non-robustness of the optimization solver.

cmastalli, 2020-08-17:
> This analysis is wrong. The two 3d contacts should be sufficient to mimic the hinge joint. The issue may come from some numerical issues or from the non-robustness of the optimization solver.

It is not the optimal control solver, it is the contact dynamics solver. Note that we have an analytical solver for that. 
Furthermore, our analytical contact solver should be able to remove redundant active constraints as in any active-set method.

The analysis is correct, I believe you misunderstood the comment.

cmastalli, 2020-08-17:
It is correct because we have designed like that. Of course, we could think to improve the contact solver in future but this creates an extra complexity in the contact solver which might not be desired.

proyan, 2020-08-17:
I agree that there is redundancy introduced because of the 2 3-d contacts. But this should be dealt with at the contact-dynamics level.

@ddliu365 I think you could try to increase the damping factor (`JMinvJt_damping`) to handle some of this rank-deficiency?

jcarpent, 2020-08-17:
> I agree that there is redundancy introduced because of the 2 3-d contacts. But this should be dealt with at the contact-dynamics level.
> 
> @ddliu365 I think you could try to increase the damping factor (`JMinvJt_damping`) to handle some of this rank-deficiency?

This was the sens of my previous comment.

cmastalli, 2020-08-17:
> @ddliu365 I think you could try to increase the damping factor (`JMinvJt_damping`) to handle some of this rank-deficiency?

In my experience, I don't think the damping procedure is the correct way of dealing. Actually, I believe it will produce numerical issues. I say this due to the fact that QP solvers don't rely on regularization for handling redundancies. The redundancy is removed by using active-set procedure, for instance.

In any case, @ddliu365 could you report the results of the @proyan's suggestion?

cmastalli, 2020-08-17:
There is something else to consider here. I think we are not handling the derivatives of the regularized version of the contact dynamics. This for sure will produce more issues, so it will be important to solve it first.

I just want to everyone know this missing task.

ddliu365, 2020-08-18:
@cmastalli @jcarpent @proyan Thanks for your response. Actually I try to avoid this issue in my problem. I am still working on this problem and will report @proyan suggestion ASAP.

This method is similar to  a technique used in TSID when I got the summer training. For their case, they also used the analytical solution for the optimization. Maybe the solver can automatically remove the redundant constraint.

cmastalli, 2020-08-25:
@ddliu365 did you have time to analyze the effect of damping the contact dynamics?

ddliu365, 2020-08-28:
Sorry for a late response. I am still working on this issue, but the solver seems to work well after I changed the time step. From the optimization process, it seems a line contact constraint is added to the robot because two points are fixed and the robot can rotate around the axis shaped by these two points.

But the whole problem has not been solved properly.

ddliu365, 2020-10-02:
Yes. I think it works well. The only thing we need to take care is the time step when we choose.

closed 2020-10-02 by ddliu365