loco-3d/crocoddyl #832: Improving CI in both: Travis and gitlab-ci

cmastalli, 2020-08-13:
> How about dividing the jobs on Travis then? Currently we do benchmarks, examples, unittests all in one single job. Perhaps it would be better to have _2_ runs, one which compiles and tests examples and unittests, and one which compiles benchmarks and python-binding based unittests. Is that a valid way out? And is it possible to do?

We are not running the benchmarks in any of the jobs neither. So, what we has is
  - for Release jobs: unittest (c++ / python) and examples
  - for Debug jobs: unittest (c++ / python)

However, I agree with you, we could have a better set of tests, however I am not sure what it is the best way to optimize it. For instance, we could do something like this:
  - Job 1: c++ unittest
  - Job 2: python unittest
  - Job 3: examples
  - Job 4: c++ benchmarks
  - Job 5: python benchmarks
  - (in future) Job 6: c++ codegen

Each job could be run in both distros (Bionic and Xenial) and in both modes (Debug and Release). Additionally the python jobs could be run in Python 2 and 3.

From Travis side, I would be incline to define the Debug cases as "allowed to fail". This is due to the fact that most of the time is needed to compile library.

What it doesn't convince me of this path is the need of many jobs in Travis. I suspect this could be contra-productive, @nim65s and @wxmerkt could you give us your feedback?

_Originally posted by @cmastalli in https://github.com/loco-3d/crocoddyl/pull/831#issuecomment-673405557_

cmastalli, 2020-08-13:
@nim65s @proyan and @wxmerkt 
I am very keen to improve this. Could we discuss it here?

Otherwise, this discussion could be lost.

wxmerkt, 2020-08-14:
I believe Guilhem has the most experience and is the best to comment on this. 

I don't know if our builds are amenable to artefact re-use or caching or how splitting jobs would work. I just opened #834 to see if we see any gains in compilation time using ccache (for repeated builds within each distro + build type). Alternatively, we could try having one job build the artefacts, then cache, and another retrieve the cache and run the tests.

> From Travis side, I would be incline to define the Debug cases as "allowed to fail"

Given there is `assert` used in quite a few places (e.g. in the recent #831 for checking user input), I would keep at least one Debug job as required to pass.

cmastalli, 2020-08-14:
> Given there is `assert` used in quite a few places (e.g. in the recent #831 for checking user input), I would keep at least one Debug job as required to pass.

Hi @wxmerkt -- this is already happening but in getgit-ci, are you OK with this approach?
We don't have it in Travis due to the needed time is often reaching the allowed 90 mins per job.

wxmerkt, 2020-08-17:
> this is already happening but in getgit-ci, are you OK with this approach?

I didn't realise it's now integrated and a pre-merge check - this is great. No objections to turn Debug to allowed to fail on Travis.
Sadly, #834 does not work as hoped because the settings (CMAKE_BUILD_TYPE, Distro, etc.) are not re-used. Happy to proceed as you see fit.

cmastalli, 2020-10-02:
@nim65s can we close this issue?

closed 2021-01-14 by cmastalli