loco-3d/crocoddyl #837: Doubt about JIntegrateTransport in IntegratedActionModelEuler::calcDiff()

PepMS, 2020-08-18:
Hi,

I was about to implement a RK4 integrator. That is why I've taken a look into how the Euler's integrator has been implemented. I have some doubts about it.

The first one is in [this line](https://github.com/loco-3d/crocoddyl/blob/1370fa32b4028e93a88f753520362a8a21c3521d/include/crocoddyl/core/integrator/euler.hxx#L113). I'll try to explain the steps I'm following to calculate the Jacobians. May be there's something wrong in this procedure and I'm not aware of it, so I'd appreciate a little bit of help.

This `IntegratedActionModelEuler::calcDiff(x, u)` function is computing the derivatives of the following function:
[image]
where
[image]
Notice that I've used p to indicate position, orientation and joint values and v to indicate all velocities, which lie in the tangent space to the manifold at p_k

Assuming that the state variable `x` that is being passed as an input paramter to the function is `x_k`, as far as I can understand,  the resulting Jacobian in [this line](https://github.com/loco-3d/crocoddyl/blob/1370fa32b4028e93a88f753520362a8a21c3521d/include/crocoddyl/core/integrator/euler.hxx#L108) is the following:
[image]
and, therefore, is an state increment that lies in the tangent space of the Manifold at `x_k`.

If what I say above is true, I get confuse with the use of `dIntegrateTransport`, as according to [this](https://gepettoweb.laas.fr/doc/stack-of-tasks/pinocchio/master/doxygen-html/structpinocchio_1_1LieGroupBase.html#a4886054e140d774b7cff831e51ed4f57), this method is used to transport a matrix that lies in the tangent space of `x_next` to `x_k`. But, isn't the stated Jacobian already in the tangent space of `x_k`?

Probably I'm missing something. I'd appreciate some light in this. Thank you very much!

cmastalli, 2020-08-18:
> But, isn't the stated Jacobian already in the tangent space of `x_k`?

Yes, it lies in the tangent space but at `x_k` state point. `dIntegrateTransport` is applying a parallel transport of this tangent vector (i.e. Jacobian) from `x_k` to `x_next`.

I took this picture from [Calinon's paper](https://calinon.ch/papers/Calinon-RAM2020.pdf), which I believe it is quite communicative.

[image]

In this picture, g and h are different point in the manifold (for us `x_k` and `x_next`). Then, `dIntegrateTransport` is computing `u_||`.

PepMS, 2020-08-18:
Still confused...quoting the Pinocchio's wiki:
"This function performs the parallel transportation of an input matrix whose columns are expressed in the tangent space of the integrated element q+v, to the tangent space at q".

So according to this, what it is really doing is to transport from x_next to x_k, right? I understand that the terminal is x_next and the origin is x_k.

proyan, 2020-08-18:
Hi @PepMS 
>So according to this, what it is really doing is to transport from x_next to x_k, right? I understand that the terminal is x_next and the origin is x_k.

You are right. This function is meant to "move" a matrix from one point to another. Think of the example of defining a velocity at point A and defining the same velocity at point B. You need to "transport" the velocity based on some relation between A and B.

Now, for the case of Integrator, what you want to do is to differentiate (I use (+) to denote the "addition" operation).
x_next = x (+) dx 

If you differentiate this wrt x, you get d(x_next)/dx. Think of differentiation as an operator. Instead of app applying this operator at x_next, wrt x, you move your matrix back to x in line https://github.com/loco-3d/crocoddyl/blob/1370fa32b4028e93a88f753520362a8a21c3521d/include/crocoddyl/core/integrator/euler.hxx#L113 , and then apply this differentiation operator in https://github.com/loco-3d/crocoddyl/blob/1370fa32b4028e93a88f753520362a8a21c3521d/include/crocoddyl/core/integrator/euler.hxx#L114

PepMS, 2020-08-18:
Ok...so, if we have the Jacobian at x_next, that should mean that the x at:
https://github.com/loco-3d/crocoddyl/blob/1370fa32b4028e93a88f753520362a8a21c3521d/include/crocoddyl/core/integrator/euler.hxx#L101
is actually x_next. Isn't it? :sweat_smile:

PepMS, 2020-08-18:
Ah I think that I know where my confusion came from.

You are doing the transport because your dp is actually dp=v_{k+1}*dt (because of the semiImplicit Euler). I was confused because I was looking at the expansion of this equation that turns out to be dp=v_k*dt+a_k*dt^2, so I assumed that dp already was at x as all subindices are k (not k+1).

My bad, sorry for the noise.

proyan, 2020-08-18:
Hi @PepMS,
I hope it was as simple as that (I hate lie algebra). `dp` index does not hold any significance here, since for lie operations on x(+)dx, it does not matter "what" index dx is.

let me write the sum operation x(+)dx as sum(x,dx)

If you look at how we compute the derivative of our state, you notice that it is:
**D(sum(x,dx)/D(x) = D(sum)/D(x) * D(x)/D(x) + D(sum)/D(dx) D(dx)/D(x)**

Just to summarise:

1) `differential_->calcDiff(d->differential, x, u)` computes D(ddq)/Dx (so, D(ddq)/Dq, and D(ddq)/Dv)

2) You use this to compute D(dx)/Dx in lines
```
    d->Fx.topRows(nv).noalias() = da_dx * time_step2_;
    d->Fx.bottomRows(nv).noalias() = da_dx * time_step_;
    d->Fx.topRightCorner(nv, nv).diagonal().array() += Scalar(time_step_);
```

At this point, we have only computed D(dx)/D(x). I don't think that you have to decide in which tangent space our "dx" lies at this point.

3) Now, we see that we need to apply an operation  D(sum)/D(dx) on our D(dx)/d(x) that we calculated. This operation can be thought of as "transporting" a matrix from tangent space at x(+)dx to tangent space at x. So now, we can think of our D(dx)/d(x) as having been defined at x(+)dx, because our formulation needs us to do this product. This does not come because of index of k.
We are transporting our matrix in line 
```differential_->get_state()->JintegrateTransport(x, d->dx, d->Fx, second)```

4) and then adding the first component in line 
```differential_->get_state()->Jintegrate(x, d->dx, d->Fx, d->Fx, first, addto)```

No matter how you define your `dx`, you would need to follow the same procedure as above.

proyan, 2020-08-18:
By the way, for the RK4 Integrator, there is a python version already in crocoddyl:
https://github.com/loco-3d/crocoddyl/blob/4708428676f596f93ffe2df739faeb5cc38d2326/unittest/python/crocoddyl/integrated_action.py#L89

Are you converting this to c++?

PepMS, 2020-08-18:
Ah ok, I think now I get...let's try another time.

So, applying the chain rule to the sum, we get:
D(sum(x,dx)/D(x) = D(sum)/D(x)\*D(x)/D(x) + D(sum)/D(dx)\*D(dx)/D(x)

1. Is the second component (D(x)/D(x)) in the first term the identity?
2. According to the equations in this image,
[image]
D(sum)/D(x) is equation (80) and D(sum)/D(dx) is (81). I guess that the transport operation is left-multiplying D(dx)/D(x) by the Jacobian (81).

I think that I get it now. Thanks for taking the time to build the explanation @proyan 

> Are you converting this to c++?

I'm aware of the code in python. Yes, the idea was to code it in C++.

proyan, 2020-08-18:
> Is the second component (D(x)/D(x)) in the first term the identity?

Yes
> D(sum)/D(x) is equation (80) and D(sum)/D(dx) is (81). I guess that the transport operation is left-multiplying D(dx)/D(x) by the Jacobian (81)

First of all, I'm glad that you are citing a paper with 193 equations to understand lie algebra :). But yes, we were calculating the same Jacobian before in crocoddyl. Since https://github.com/stack-of-tasks/pinocchio/pull/1172, we are using this approach.

proyan, 2020-08-18:
@PepMS Please close this issue if you feel that your doubt is cleared :)

closed 2020-08-18 by PepMS

jcarpent, 2020-08-18:
@PepMS @proyan Just to be a bit more precise: we have started the project Pinocchio (with @jmirabel and @jcarpent for supporting the development) with in mind the wish of supporting correctly the notion of Lie groups with their underlying geometry. Since then, many projects rely on Pinocchio for this support, mainly https://github.com/humanoid-path-planner and of course Croco.

joansola, 2020-08-19:
Hi guys, thanks for sharing this great work and sorry for insisting on this topic still a little further. I have not fully understood the conclusion of this conversation.

Once we have this expression:

```
D(sum(x,dx)/D(x) = D(sum)/D(x) + D(sum)/D(dx)*D(dx)/D(x)
```
and after computing the partials using (80) and (81) [here](https://github.com/loco-3d/crocoddyl/issues/837#issuecomment-675593787), what is the conclusion?

1) we are done and `D(sum)/D(dx)` is equivalent to the transport operation
2) we still have to perform the transport operation anyway

I feel comfortable with 1). But if the answer is 2) this means that I am missing something.

proyan, 2020-08-19:
Hi @joansola 

> we are done and D(sum)/D(dx) is equivalent to the transport operation

The dIntegrateTransport operation is equivalent to the multiplication operator `D(sum)/D(dx)*`. 
So if needed, you can get the `Dsum/D(dx)` from the dIntegrate operation in pinocchio, and dIntegrateTransport operation is reducing this multiplication operation for us.

joansola, 2020-08-19:
I see. Thanks for this clarification!