loco-3d/crocoddyl #849: SolverBoxFDDP bounding the control only in contact phase

gfadini, 2020-09-10:
Hi, I'm experiencing a problem with the ```SolverBoxFDDP``` solver that I was testing in a contact scenario and I'm not fully able to grasp why this is happening.
If I understood correctly this solver aim, ideally the control should be restrained in the bounds defined by the vectors -/+```Model.effortLimit``` in all the time window of the problem regardless of the model underlying dynamics.
However what I see is that the projection of the control on the bounds is applied only in the models that are created with the ```DifferentialActionModelContactFwdDynamics``` model and not for those of the flying phase that are created with the ```DifferentialActionModelFreeFwdDynamics```
Is there any particular reason that could explain this behavior?
I attach an image, in this case the bounds [-2.5, 2.5] Nm are respected only in the first part of the trajectory. The following time interval (with the running model) has some peaks that I guess should be easily flattened by this particular solver
[image]

cmastalli, 2020-09-10:
Hi @gfadini,

You can set manually the torque limits. However, you are right. For some reason we miss to automatically set them. 
Now I am off of office, but you might want to add these routine in the c++ code, see this example https://github.com/loco-3d/crocoddyl/blob/master/include/crocoddyl/multibody/actions/contact-fwddyn.hxx.

Btw, you don't need to use the free fwd-dynamics for flying phases. It is enough with deactivating all the contacts.

gfadini, 2020-09-10:
Thanks @cmastalli , I see now that for all the models in the second phase as you said they're not set, the same is also true for the ```SolverBoxDDP```

```
m=ddp.problem.runningModels[-1]
m.u_lb
array([-inf, -inf])
m.has_control_limits
False
```
I think I can manually set them, for future reference how do I deactivate all the contacts?

cmastalli, 2020-09-10:
There is a function named 'setContactStatus', which you could call by getting the contacts.

gfadini, 2020-09-11:
Closed in #850

closed 2020-09-11 by gfadini