loco-3d/crocoddyl #852: Specifying costs that should not be integrated

PepMS, 2020-09-18:
I wanted to specify a cost for a single node placed in the middle of the problem. Thus, this cost should not be integrated with time. We have the same case with a terminal node, that it is not integrated with time and, instead, the cost considered in the IntegratedActionModel is directly taken from the DifferentialActionModel.

When creating the IntegratedActionModel, we use time_step=0 to indicate that this is a terminal node and, therefore, integration has to be disabled. I'd propose something similar for a specific node placed inside the problem. The difference would be that in this case the dynamics and its derivatives have to be computed like any other running node. Thus,

- Cost computation: https://github.com/loco-3d/crocoddyl/blob/d9eb51312e4d07200cd4a849b50e98ecd9468a09/include/crocoddyl/core/integrator/euler.hxx#L73
- Dynamics computation: https://github.com/loco-3d/crocoddyl/blob/d9eb51312e4d07200cd4a849b50e98ecd9468a09/include/crocoddyl/core/integrator/euler.hxx#L66-L68
- Cost derivatives: https://github.com/loco-3d/crocoddyl/blob/d9eb51312e4d07200cd4a849b50e98ecd9468a09/include/crocoddyl/core/integrator/euler.hxx#L125-L129
- Dynamics derivatives: https://github.com/loco-3d/crocoddyl/blob/d9eb51312e4d07200cd4a849b50e98ecd9468a09/include/crocoddyl/core/integrator/euler.hxx#L104-L115

When writing the issue I've realized that what I've just propose cannot be done if, for example, in a node we need costs that have to be integrated with time and nodes that don't. So, any workaround to achieve this?

Thank you so much in advance.

cmastalli, 2020-09-18:
Hi @PepMS,

Your request doesn't look compatible with standard theory behind optimal control (i.e. Mayer form).

Could you justify a bit more this requirement? For instance, tell us about the formulation.

However, you can work around this issue with current code. For instance, you could set the step integration to 1. Note that we have a trapezoidal integrator in this case.

PepMS, 2020-09-18:
I was thinking of a multi-stage problem, where you concatenate several OCP and then specify a continuity constraint between stages.

In any case, I might be confused. I will try to put my ideas straight and will write back here if needed. I think I'll close this by now. Sorry for the noise.

closed 2020-09-18 by PepMS