loco-3d/crocoddyl #859: No to_python (by-value) converter found for C++ type: boost::shared_ptr<crocoddyl::SolverFDDP>

edantec, 2020-09-25:
When prototyping in Python, I noticed there is no python to C++ binding for a shared pointer of SolverFDDP object. There is one for `boost::shared_ptr<crocoddyl::SolverDDP>` though.

cmastalli, 2020-09-25:
Because we don't need to bind any shared pointer for any solver. The code that you mentioned (for the DDP solver) is indeed not needed.

Are you facing an issue with this?

edantec, 2020-09-27:
I have a C++ function which returns a `boost::shared_ptr<crocoddyl::SolverFDDP>` object, and I have Python bindings for this function. When trying to use this function in Python I encounter the error `No to_python (by-value) converter found for C++ type: boost::shared_ptr<crocoddyl::SolverFDDP>`. I do not have this error when I change the type from SolverFDDP to SolverDDP in C++.

proyan, 2020-09-27:
Are you changing the #include to fddp as well?

edantec, 2020-09-29:
Yes, I included `#include <crocoddyl/core/solvers/fddp.hpp>` but still the same error

cmastalli, 2020-09-29:
> Yes, I included `#include <crocoddyl/core/solvers/fddp.hpp>` but still the same error

I don't understand this comment.  You get this message when you haven't register a converter.
Check out the latest devel branch, it should be solved.

proyan, 2020-09-29:
@edantec could you rebase on top of the current devel? #861 should have solved your issue.

cmastalli, 2020-09-30:
solved by #861

closed 2020-09-30 by cmastalli