loco-3d/crocoddyl #866: Changing API of costModel.reference

proyan, 2020-10-07:
Dear Crocoddyl developers, 

It looks like the `get_reference` function in the costmodel is creating a copy of the object in python ([here](https://github.com/loco-3d/crocoddyl/blob/devel/include/crocoddyl/core/cost-base.hxx#L84)) and returning that copy. We don't want that. We need to return an internal reference.

Unfortunately, the returning of internal references is being made complicated by the current design choice, where the cost reference is not a `VectorX` object, but instead a wrapper around this vector (`FrameForce`, `FrameMotion`, `FramePlacement`) etc. 
Thus, we need to somehow imitate a templated virtual function, which I don't feel is needed.

I'm proposing here to change this and simplify the c++ API to make `costmodel.reference` into a vector object, as it was earlier in the  original python version. This would move the frame-id information into the respective cost classes, and remove the wrappers FrameMotion, FrameForce, FrameWrenchCone etc, which are simply duplicate wrappers on pinocchio classes.
Also, this would rename the individual costmodel references (`fref`, `cref`, `Mref`) into a single name reference.

After the depreciation phase, this API change would affect all users from the future releases, and thus, your opinions are welcome here.

jcarpent, 2020-10-07:
Everything that makes Crocoddyl closer to the Pinocchio API is always, both in terms of maintainability and extensions. I'm in favour of this change.

cmastalli, 2020-10-07:
Hi @proyan,

Each reference cannot fit into a vector, e.g. SE(3) placement.
Furthermore, the reference are also describing the frame index.
So, if you define a vector (and put everything there) then you will need to convert it to the right object.
Note that this conversion is strictly needed since `calc` and `calcDiff`use it.

On the other hand, having to create an object indeed is not desired, I agree with you.
But again, the vector conversion that you propose doesn't seem to be a better idea.
It also sacrifices the usability for a few nanoseconds.

In short, I believe it is better to keep the current implementation.
IMHO, your solution creates too much user-side complexity for a neglectable runtime improvement.

Carlos

proyan, 2020-10-07:
Regarding your points, I'll try to go one by one.

> Each reference cannot fit into a vector, e.g. SE(3) placement.

Yes it can.

>  Furthermore, the reference are also describing the frame index. So, if you define a vector (and put everything there) then you will need to convert it to the right object.

I think you are misunderstanding what the issue is. We should not be defining the frame indices in the reference object in the first place. Currently, the API is 
`CostModelContactForce->fref->id`
`CostModelContactForce->fref->force`

However, this means that if a user wants to update the reference force, he/she would need to update the `id` as well. 
This becomes much more complicated when we go to costs such as `ContactModelWrenchCone` and `ContactCoPPosition`. Everytime we add a new cost, we need to add a new frameobject to add as a reference. 

However, the API that we had before was much easier:
`CostModelContactForce->id`
`CostModelContactForce->fref`
Now, both id and fref are accessible to the user.

>  IMHO, your solution creates too much user-side complexity for a neglectable runtime improvement.

There is no talk of runtime improvement here. This is meant to simplify the API.

cmastalli, 2020-10-07:
It means that you need to create a virtual function for setting index, this is not the case for state and control cost, for instance.

The question is how much are we saving at runtime? Without a number it looks like a unnecessary optimization. It is also good have a user-friendly interface, there is a trade-off to be pay.

There are many more important matters, e.g. codegen of QP solver.

cmastalli, 2020-10-07:
The API is simple, why do you say the opposite?

What it could be a bit more complex is the implementation, not API.

cmastalli, 2020-10-07:
> Everything that makes Crocoddyl closer to the Pinocchio API is always, both in terms of mantainability and extensions. I'm in favour of this change.

@jcarpent what do you mean closer to Pinocchio? Note that the current solution is closer to Pinocchio, i.e. It uses Pinocchio SE(3) and Motion.

proyan, 2020-10-07:
Again, I'll try to go through your points one by one.

> The question is how much are we saving at runtime? Without a number it looks like a unnecessary optimization. It is also good to have a user-friendly interface, there is a trade-off to be payed.

Again, this is not a performance improvement. This is meant to simplify the API where it is consistent through all classes. And to remove unnecessary wrappers which are not needed to be maintained in crocoddyl.

> There are many more important matters, e.g. codegen of QP solver.

This is again not related to codegen.

>  For instance, a user set a Pinocchio SE(3) object, rather than arranging its values, in a magic way, into a vector. It sounds a complex task to learn this magic way, a very easy to make a mistake.

It is definitely good to have overloads, and it is good to have set_reference that can accept both vectorxd and pinocchio objects for example. Currently, what we are doing is to create a wrapper around that pinocchio object, and then ask user to give us this wrapper instead of directly the pinocchio object. This is different from what you are suggesting here.
If this is your main suggestion, then I think we don't have a disagreement here.

cmastalli, 2020-10-08:
@proyan I believe you are not seeing the whole picture. It is not only about the indexes, we also have other terms such as friction coefficient, surface normal, etc. Furthermore, the reference of a cost function could be anything, not only a Pinocchio object.

Said that, the design logic is to have a single function that allows us to set any reference without the need of down casting the function. Each cost function has a specific reference, for instance 
  - state and control cost do not need to pass indexes or
  - friction cone and wrench costs need to pass friction coefficient, normals, etc.

Indeed, a reference instantiates a desired behaviour of our cost function, thus all these mentioned terms are conceptually a reference information.

Unfortunately, your proposition is not making our life easier once we want to set up a reference. I fully disagree with your claim: "this proposition simplifies the API". This is due to the fact that your proposition will impose a down-casting when you want to set a index, a friction coefficient, or any object that is not a Vector or Pinocchio structure.

I have provided more details in my previous replies. Please read carefully for more details :)

nmansard, 2020-10-08:
I don't really understand why the frame objects have been introduced. The argument of @jcarpent is very sound: keeping Pinocchio API as often as possible is very sounds, otherwise rely on plain vector objects. I cannot figure out what you feel not being able to do, @cmastalli, that pushed you to introduce this object.

Could we discuss that orally?

cmastalli, 2020-10-08:
> Could we discuss that orally?

Yes, let's discuss it.
Let's arrange our meeting on riot.

cmastalli, 2021-01-19:
Tailored cost functions will be eventually replace by the residuals as discussed in #899
Therefore, we don't need to change the cost-reference API as these tailored cost functions will be deprecated.
Instead, I am propagating our agreement in the residual function as they contain the reference as well.

I will close this issue as it is not needed!

closed 2021-01-19 by cmastalli