loco-3d/crocoddyl #872: FIx API change introduced from #857

proyan, 2020-10-20:
The change in the header files breaks downstream packages which were using the previous header structure.

@cmastalli could you please take care of this.
Could you please 
1) recreate the original files as empty header files.
2) include the new header files inside the original headers
3) and deprecate the original?

cmastalli, 2020-10-20:
Hi @proyan,

Yes, it could. Do you have a deadline for that?
I hope this is not priority for you because I have many things in my TODO list.

In any case, I promise to handle it before the next release.
Let me know if this is fine for you :)

Best,
Carlos

proyan, 2020-10-20:
There is never a deadline for a broken API. If it isn't fixed, people would be forced to update their code anyways.

cmastalli, 2020-10-20:
> There is never a deadline for a broken API. If it isn't fixed, people would be forced to update their code anyways.

I will do my best.
Hopefully, I will provide a solution before the end of the next week :)

jcarpent, 2020-10-20:
One other fast solution would be to revert the Merge #857 and then publish a new release.
I think it is important to not ask downstream users to have to change their code because some deprecated tags have been forgotten.

nim65s, 2020-10-20:
I need this. I'll deal with it.

proyan, 2020-10-27:
Fixed with 4a3c544

commit 4a3c544c (2020-10-27): Merge pull request #873 from nim65s/devel

add deprecated headers, fix #872
commit d75c3576 (2020-10-27): add deprecated headers, fix #872
closed 2020-10-27 by proyan