loco-3d/crocoddyl #875: Add mass or losing mass

ddliu365, 2020-10-22:
I try to work on a problem with adding or losing mass from a humanoid.

For example, we may give humanoid a book and or it can throw a ball. 

In this case, can we optimize these process with crocoddyl library?

cmastalli, 2020-10-24:
Hi @ddliu365,

A workaround is to define a new URDF model that includes the extra weight in the body that carries out it.
You will need to define a phase of events: e.g. carrying an object and then throwing it, where each phase will have different URDF models (or precisely multibody states).

Your initial idea is not yet clear for me.
In any case, in Crocoddyl, there is no any theoretical limitation if you wish to incorporate the object dynamics into the optimal control problem.
However, these requires development from your side.

Is this clarifies your question?
If so, could you close this issue?

jcarpent, 2020-10-24:
Be simple. Just append an inertia to the inertia of the end effector which is holding the weight. No need at all to define a new URDF.

ddliu365, 2020-10-25:
> Hi @ddliu365,
> 
> A workaround is to define a new URDF model that includes the extra weight in the body that carries it out.
> You will need to define a phase of events: e.g. carrying an object and then throwing it, where each phase will have different URDF models (or precisely multibody states).
> 
> Your initial idea is not yet clear for me.
> In any case, in Crocoddyl, there are no theoretical limitations if you wish to incorporate the object dynamics into the optimal control problem.
> However, it requires development and research from your side.
> 
> Does this answer clarify your doubts?
> If so, could you close this issue?

@cmastalli 

That's the exact thing I want to do. My initial idea is like force control method which doesn't fit to ddp approach.

ddliu365, 2020-10-25:
> Be simple. Just append an inertia to the inertia of the end effector which is holding the weight. No need at all to define a new URDF.

@jcarpent  I think adding inertia is more efficient compared with updating Urdf.
However I am not familiar to the library(Pinocchio or Crocoddyl). Which method or function should I use for adding inertia?

jcarpent, 2020-10-25:
`Model::appendInertia` is your solution.

cmastalli, 2020-10-25:
Yes, It is better to append the inertia. Indeed, I was referring to that but somehow I said "new URDF".

What you really need is a new Pinocchio model ;)

ddliu365, 2020-10-25:
Thanks. I see the point and will implement that.

closed 2020-10-25 by ddliu365