loco-3d/crocoddyl #877: About prismatic joint

ddliu365, 2020-10-26:
I am working on a project which has one underactuated prismatic joint connected to the free floating base.

That is, this prismatic joint will affect the inertia distribution, but no motor is connected.

For the a typical free floating base robot, we have a configuration dimension is (7+n)(if rotation is quaternion, following pinocchio style), but for this robot, we need a configuration dimension (8+n).

Is that possible to define a ddp problem with crocoddyl without changing c++ class?

cmastalli, 2020-10-27:
Crocoddyl should be able to handle this problem since the `ActuationModelFloatingBase` class could deal with different degrees of freedom in the floating base. However, we have not tested this condition, thus if you find an issue then let us know.

There is something not yet clear in your message. If you substitute the free-flyer joint by a prismatic one, then the configuration dimension is `1 + nj`, where `nj` is the number of actuated joints.

If you refer to have an appended floating-base joint (i.e. 8 degrees of freedom), then I am not fully sure if this is possible in Pinocchio. This will be the only limitation in Crocoddyl since we assume that the first joint is the floating-base one.

proyan, 2020-10-27:
Hi @ddliu365 

If you look at the definition of the ActuationModelFloatingBase (https://github.com/loco-3d/crocoddyl/blob/b0eeaa5713166d7e6955454b90aedf0fc940baa1/include/crocoddyl/multibody/actuations/floating-base.hpp#L30), the underactuation is defined to be only present in the first joint (which may or may not be the free-flyer). As of now, crocoddyl doesn't consider underactuation in places other than the initial joint. The model you have, where two subsequent joints are underactuated, seems to be very peculier. If you can explain further your problem, perhaps we can help you use the library better.

ddliu365, 2020-10-27:
@cmastalli @proyan 

Actually, I don't substitute the free-flyer joint by a prismatic one, but I want to add one more prismatic joint to current SE3 free-flyer joint as you can from the draft. 

It is possible to add the second joint as an underactuated joint. I think pinocchio can handle the prismatic joint.

[image]

The above figure shows the simplified model of this project. I plan to add one mass block which can slide along the torso and this will bring another underactuated degree of freedom(maybe there is another way to consider it, and my assumption for model simplification may be stupid). 

Except the extra mass along the torso, it is a actual bipedal robot. The torso is SE3(3 translation DoF and 3 rotation DoF).

I admit this idea is peculiar, maybe there is another interpretation for the problem.

proyan, 2020-10-29:
> The sliding motion is realized by the friction between the block and the torso

You would need to show this friction force in an actuation model. The actuation model would convert your current state + motor actuation + friction to the pinocchio tau control, which you can use to run your dynamics through the action models. 

Since this is a custom problem, the solution would also be a custom code. Crocoddyl is modular enough, that the only thing you would need to do is to create a new actuation model which contains this friction model. Afterwards, crocoddyl and pinocchio have all the tools you need for your problem.

ddliu365, 2020-10-29:
Thank you @proyan 

if `nj` only represents the rotation joint and `nv` should be equal to `nj+6+1`.

I would like to change
https://github.com/loco-3d/crocoddyl/blob/b0eeaa5713166d7e6955454b90aedf0fc940baa1/include/crocoddyl/multibody/actuations/floating-base.hpp#L30
to
`      : Base(state, state->get_nv() - state->get_pinocchio()->joints[1].nv()-1){};`
I would like to change
https://github.com/loco-3d/crocoddyl/blob/b0eeaa5713166d7e6955454b90aedf0fc940baa1/include/crocoddyl/multibody/actuations/floating-base.hpp#L57
to
`data->dtau_du.diagonal(-state->get_pinocchio()->joints[1].nv()-1).fill((Scalar)1.);`

add -1 above because we have one prismatic joint which is underactuated.
If we assume the second joint in pinocchio is this prismatic joint.

Do you think if the above changes can solve my problem?

cmastalli, 2021-02-17:
I will close this issue as there is no activity.

However, you need to consider that you could use the floating-base actuation if you can embed both joints (FF and prismatic) into a single one. I am not sure if this is possible in Pinocchio in term of current implementation or software design.
If this is not possible, then you will need to define your custom actuation model. You could apply the modifications that you described above.

closed 2021-02-17 by cmastalli