loco-3d/crocoddyl #884: Actuation model inside the DifferentialActionModelContactFwdDynamics class

Diiso, 2020-11-10:
Hi,
I am trying to manipulate a robotic arm with crocoddyl and also taking into account contacts.
I am thus using the `DifferentialActionModelContactFwdDynamics` class, but the actuation model that this class requires is an `ActuationModelFloatingBase` (C++ code below) instance, which is not correct for a robotic arm as it doesn't have a free-floating joint.
Is there a way to have the `DifferentialActionModelContactFwdDynamics` class use instances of the `ActuationModelFull` class?

Thanks!

```
 DifferentialActionModelContactFwdDynamicsTpl(boost::shared_ptr<StateMultibody> state,
                                               boost::shared_ptr<ActuationModelFloatingBase> actuation,
                                               boost::shared_ptr<ContactModelMultiple> contacts,
                                               boost::shared_ptr<CostModelSum> costs,
                                               const Scalar& JMinvJt_damping = Scalar(0.),
                                               const bool& enable_force = false);
```

proyan, 2020-11-13:
You are right @Diiso . Thanks for raising this issue. The actuation should be abstract type here, and this is indeed a bug in the design. 

I can try to fix this by next week. Would that be okay for you?

Diiso, 2020-11-13:
Thank you @proyan, yes next week is fine ! :)

cmastalli, 2020-11-13:
It would be also useful to include an example

nmansard, 2020-11-13:
I think it is mostly changing the type of the actuation model inside the contact differential dynamic model. That should be an agnostic actuation, while it is hard-coded as a FF dynamics. 
It might be that @gfadini already solved the issue on a private branch. If this is the case, could you raise it here so that we can evaluate possibility to merge it?

Alternatively, it is mostly changing types and checking the unittest. @Diiso, do you think that you would be in position to handle it? That might be more efficient than waiting for @proyan to have the availability for it.

gfadini, 2020-11-13:
Hi @nmansard in my case, since I was dealing with an under-actuated robot, the solution was to just remove a check in the first link being of floating base type. That way I could simply use the ```ActuationModelFloatingBase``` in place of the custom one. So all the changes were mainly on that side #816 to solve of issue #807. For some tests I also used a fully actuated robot but that was without contact, so I guess that wouldn't work for this problem.

Diiso, 2020-11-16:
Hi @nmansard @proyan, ok I will do the type change and create a PR. Regarding unittests, there isn't one for `DifferentialActionModelContactFwdDynamics` with an actuation other than `ActuationModelFloatingBase` so I can add one with `ActuationModelFull`, using the TalosArm robot.

proyan, 2020-11-19:
Fixed with #886

closed 2020-11-19 by proyan