loco-3d/crocoddyl #895: ActuationModelFull with FreeFlyer

hammoudbilal, 2020-12-28:
Hi, 

I'm trying to create a simple demo for a fully actuated sphere (i.e. my model is only an SE3 joint with some mass) just to move it around and control the orientation too, however ActuationModelFull prevents this
 `Invalid argument: the first joint cannot be free-flyer`
is there a workout around this issue ?

proyan, 2021-01-04:
Hi BIlal,

This is a bug that should not be part of the code. I think this was done to make sure a user does not use ActuationModelFull when controlling a floating base robot. But perhaps we we need to trust our users a bit more. 

We are already using the StateMultibody in our ActuationModel class. So the only thing to do is to remove this throw. Do you think you can make a PR with this, and a unittest of your example in `unittest/factory/actuation`?

hammoudbilal, 2021-01-11:
Hi @proyan 

alright, I will leave the issue open till i get to doing it sometime later this week! thanks

cmastalli, 2021-02-23:
solved in #912

closed 2021-02-23 by cmastalli