loco-3d/crocoddyl #896: Error about omniORB.CORBA.BAD_PARAM?

ddliu365, 2020-12-31:
Hi, I tested the solver, and got this error:

```
---------------------------------------------------------------------------
BAD_PARAM                                 Traceback (most recent call last)
<ipython-input-7-8f02414781a8> in <module>()
      2 
      3 rolloverPlanner = DoubleSupportThrowPlanner(x0, nq, nv, na, 100, release_index=0, timeStep=ctrl_time_step, display=True)
----> 4 tauRolloverTraj,throwTime = rolloverPlanner.forward(m)

/home/jack/repos/throw/src/whole_body_planner.pyc in forward(self, model)
     50         gait = SimpleHumanoidGaitProblem(model.model_o, q0, torso, lFoot, rFoot, rGripper)
     51         problem = gait.createDSPrepThrowProblem(self.x0, timeStep, prepKnots, final=False)
---> 52         x_prep = self.solve(model.robot_o, model.model_o, problem, endEffectors)
     53         x_f_prep = x_prep[:,-1]
     54         x_f_prep = x_f_prep.reshape((x_f_prep.shape[0], 1))

/home/jack/repos/throw/src/planner.pyc in solve(self, robot, model, problem, endEffectors)
     59             for m, d in list(zip(self.ddp.problem.runningModels, self.ddp.problem.runningDatas))
     60         ]
---> 61         done = self.ddp.solve(xs, us, 500, False, 0.1)
     62         print(done)
     63         if self.display==True:

/opt/openrobots/lib/python2.7/site-packages/crocoddyl/__init__.pyc in __call__(self, solver)
    351         if (solver.iter + 1) % self.visualization.freq:
    352             return
--> 353         self.visualization.displayFromSolver(solver)
    354 
    355 

/opt/openrobots/lib/python2.7/site-packages/crocoddyl/__init__.pyc in displayFromSolver(self, solver, factor)
     44         models = solver.problem.runningModels.tolist() + [solver.problem.terminalModel]
     45         dts = [m.dt if hasattr(m, "differential") else 0. for m in models]
---> 46         self.display(solver.xs, fs, ps, dts, factor)
     47         if numpy_conversion:
     48             numpy_conversion = False

/opt/openrobots/lib/python2.7/site-packages/crocoddyl/__init__.pyc in display(self, xs, fs, ps, dts, factor)
    147                         position.rotation = rotationMatrixFromTwoVectors(self.z_axis, f["nsurf"])
    148                         frictionName = self.frictionGroup + "/" + key
--> 149                         self._setConeMu(key, f["mu"])
    150                         self.robot.viewer.gui.applyConfiguration(
    151                             frictionName, list(np.array(pinocchio.SE3ToXYZQUAT(position)).squeeze()))

/opt/openrobots/lib/python2.7/site-packages/crocoddyl/__init__.pyc in _setConeMu(self, coneName, mu)
    310             coneGroup = self.frictionGroup + "/" + coneName
    311 
--> 312             self.robot.viewer.gui.deleteNode(coneGroup + "/lines/0", "")
    313             self.robot.viewer.gui.deleteNode(coneGroup + "/lines/1", "")
    314             self.robot.viewer.gui.deleteNode(coneGroup + "/lines/2", "")

/opt/openrobots/lib/python2.7/site-packages/gepetto/corbaserver/graphical_interface_idl.pyc in deleteNode(self, *args)
    457 
    458     def deleteNode(self, *args):
--> 459         return _omnipy.invoke(self, "deleteNode", _0_gepetto.corbaserver.GraphicalInterface._d_deleteNode, args)
    460 
    461     def getNodeList(self, *args):

BAD_PARAM: CORBA.BAD_PARAM(omniORB.BAD_PARAM_WrongPythonType, CORBA.COMPLETED_NO)
```

cmastalli, 2021-01-04:
Could you report the version used for Gepetto viewer and corba?
I think @jmirabel (or @nim65s) could know the reason.

jmirabel, 2021-01-04:
The second argument of `deleteNode` should be a boolean. @cmastalli the bug is in crocoddyl.

ddliu365, 2021-01-04:
@cmastalli 
```
jack@desktop:~$ dpkg -l | grep robotpkg
ii  robotpkg-collada-dom                            2.4.0r1                                               amd64        C++ object representation of a COLLADA XML instance document
ii  robotpkg-curves                                 0.5.2                                                 amd64        Library for creating curves
ii  robotpkg-dynamic-graph-v3                       4.3.1                                                 amd64        Efficient data-flow C++ library for robotics
ii  robotpkg-eigen-quadprog                         1.0.1r1                                               amd64        QuadProg QP solver using the Eigen3 library
ii  robotpkg-eiquadprog                             1.2.2                                                 amd64        algorithm of Goldfarb and Idnani for the solution of a QP
ii  robotpkg-example-robot-data                     3.7.0                                                 amd64        Robot URDFs for benchmarking and developed examples
ii  robotpkg-hpp-fcl                                1.6.0                                                 amd64        Humanoid Path Planner (collision detection and distance computations)
ii  robotpkg-libccd                                 2.0                                                   amd64        collision detection
ii  robotpkg-octomap                                1.9.0                                                 amd64        An Efficient Probabilistic 3D Mapping Framework Based on Octrees
ii  robotpkg-openscenegraph                         3.6.5r3                                               amd64        open source high performance 3D graphics toolkit
ii  robotpkg-parametric-curves                      1.4.0                                                 amd64        This packages provides a template based parametric curves library.
ii  robotpkg-pinocchio                              2.5.4                                                 amd64        Efficient rigid body dynamics
ii  robotpkg-py27-crocoddyl                         1.5.0r1                                               amd64        Optimal control library for robot under contact sequence
ii  robotpkg-py27-curves                            0.5.2                                                 amd64        Library for creating curves (python bindings)
ii  robotpkg-py27-eigenpy                           2.5.0                                                 amd64        Eigen bindings to python
ii  robotpkg-py27-example-robot-data                3.7.0                                                 amd64        Robot URDFs for benchmarking and developed examples
ii  robotpkg-py27-hpp-fcl                           1.6.0                                                 amd64        Humanoid Path Planner (collision detection and distance computations (python bindingss))
ii  robotpkg-py27-parametric-curves                 1.4.0                                                 amd64        This packages provides a template based parametric curves library (python bindings)
ii  robotpkg-py27-pinocchio                         2.5.4                                                 amd64        Efficient rigid body dynamics (python bindings)
ii  robotpkg-py27-qt4-gepetto-viewer                4.10.1r2                                              amd64        Humanoid Path Planner (Simple viewer based on OpenSceneGraph)
ii  robotpkg-py27-qt4-gepetto-viewer-corba          5.5.1                                                 amd64        Humanoid Path Planner (Corba server for gepetto-viewer)
ii  robotpkg-py27-quadprog                          0.1.6r1                                               amd64        Solve a strictly convex quadratic program
ii  robotpkg-qt4-osgqt                              3.5.7r2                                               amd64        open source high performance 3D graphics toolkit
ii  robotpkg-simple-humanoid-description            1.0.2                                                 amd64        Rigid body model of a simple humanoid robot.
ii  robotpkg-tsid                                   1.4.2                                                 amd64        Efficient Inverse Dynamics based on Pinocchio
```

cmastalli, 2021-01-04:
@ddliu365 the solution was provided by @jmirabel.

We need to replace `""` by `True` here:
```bash
--> 312             self.robot.viewer.gui.deleteNode(coneGroup + "/lines/0", "")
    313             self.robot.viewer.gui.deleteNode(coneGroup + "/lines/1", "")
    314             self.robot.viewer.gui.deleteNode(coneGroup + "/lines/2", "")
```

@ddliu365 could you handle this issue and create a PR?

ddliu365, 2021-01-04:
Sure, see pr #897

closed 2021-01-04 by ddliu365