loco-3d/crocoddyl #901: customized cost for multibody

ddliu365, 2021-01-12:
Hi,

I want to construct a customized frame velocity cost like the following:
[image]

[image] is the frame linear velocity.

I am using a python library. How do I construct a customized cost like this?

cmastalli, 2021-01-12:
Any cost model can be derived in Python similarly to action models.
However, you should consider to develop an activation model as it allows you to reuse our velocity cost.

ddliu365, 2021-01-12:
want to build based on `ActivationModelAbstract` and `ActivationDataAbstract`:
```
from crocoddyl import ActivationModelAbstract, ActivationDataAbstract
class ActivationModelQuad(ActivationModelAbstract):
    def __init__(self):
        self.ActivationDataType = ActivationDataQuad

    def calc(self, data, r):
        '''Return [ a(r_1) ... a(r_n) ] '''
        return r**2 / 2

    def calcDiff(self, data, r, recalc=True):
        if recalc:
            self.calc(data, r)
        '''
        Return [ a'(r_1) ... a'(r_n) ], diag([ a''(r_1) ... a''(r_n) ])
        '''
        return r, np.array([1.] * len(r))[:, None]

class ActivationDataQuad(ActivationDataAbstract):
    def __init__(self, model):
        pass

am = ActivationModelQuad()
ad = am.createData()
x = np.random.rand(1)
```

but found the following error:
```
---------------------------------------------------------------------------
ArgumentError                             Traceback (most recent call last)
<ipython-input-3-a4fc4f6a507f> in <module>()
      1 am = ActivationModelQuad()
----> 2 ad = am.createData()
      3 x = np.random.rand(1)

ArgumentError: Python argument types in
    ActivationModelAbstract.createData(ActivationModelQuad)
did not match C++ signature:
    createData(crocoddyl::python::ActivationModelAbstract_wrap {lvalue} self)
    createData(crocoddyl::python::ActivationModelAbstract_wrap {lvalue} self)
```

Do you know where I made mistakes?

cmastalli, 2021-01-13:
Your code is using a very old API (around 2 year old).

You need to adapt your code as follows:
 1. Construct the `ActivationModelAbstract` (as done in Python derived classes)
 2. Remove `self.ActivationDataType = ActivationDataQuadNew` line
 3. Store the computations of `calc`, `calcDiff` inside `data.r`, and (`data.Ar`, `data.Arr`), respectively. (now they are void functions)
 4. Define `createData` function which returns your new data class
 5. Construct the `ActivationDataAbstract` in `ActivationDataQuadNew`

ddliu365, 2021-01-13:
Thanks. I found the old code in your depreciated folder
I revised the code based on your comments. It works now. Can you help me check?
```
from crocoddyl import ActivationModelAbstract, ActivationDataAbstract
class ActivationModelQuadNew(ActivationModelAbstract):
    def __init__(self):
        pass
        
    def calc(self, data, r):
        '''Return [ a(r_1) ... a(r_n) ] '''
        data.r = r**2 / 2

    def calcDiff(self, data, r, recalc=True):
        if recalc:
            self.calc(data, r)
        '''
        Return [ a'(r_1) ... a'(r_n) ], diag([ a''(r_1) ... a''(r_n) ])
        '''
        data.Ar = r
        data.Arr = np.array([1.] * len(r))[:, None]
    def createData(self):
        data = ActivationDataQuadNew(self)
        return data
    
class ActivationDataQuadNew(ActivationDataAbstract):
    def __init__(self, model):
        pass
am = ActivationModelQuadNew()
ad = am.createData()
import numpy as np
x = np.random.rand(1)
am.calc(ad, x)
```

cmastalli, 2021-01-13:
The message is telling you the problem.
`data.a` is a scalar, not an array

Btw, are you using the master branch?
Note that we have renamed this variable to `data.a_value` in the devel branch.
(I suggest you to work at the devel branch)

ddliu365, 2021-01-13:
yes, I am using the release v 1.5.0. The source code is `data.a`.
I will switch to `devel` branch and build from source later.

I see the problem, but it still pops up errors after I revise:
```
    def calc(self, data, r):
        '''Return [ a(r_1) ... a(r_n) ] '''
        print(r.transpose().dot(r)*0.5)
        data.a =  r.transpose().dot(r)*0.5
```
The error message is:
```
0.009830020482429593
---------------------------------------------------------------------------
ArgumentError                             Traceback (most recent call last)
<ipython-input-8-baffceaa3c4c> in <module>()
      1 import numpy as np
      2 x = np.random.rand(1)
----> 3 am.calc(ad,x)
      4 am.calcDiff(ad,x)

<ipython-input-7-6c70204df63a> in calc(self, data, r)
      7         '''Return [ a(r_1) ... a(r_n) ] '''
      8         print(r.transpose().dot(r)*0.5)
----> 9         data.a =  r.transpose().dot(r)*0.5
     10 
     11     def calcDiff(self, data, r, recalc=True):

ArgumentError: Python argument types in
    None.None(ActivationDataQuadNew, numpy.float64)
did not match C++ signature:
    None(crocoddyl::ActivationDataAbstractTpl<double> {lvalue}, double
```
The same thing happens to `data.Ar` and `data.Arr`.

I don't know what happened here.

cmastalli, 2021-01-14:
It seems you forgot to construct the activation base class.
You code should looks like
```python
import crocoddyl

class ActivationDataQuadNew(ActivationDataAbstract):
    def __init__(self, model):
        crocoddyl.ActivationDataAbstract.__init__(self, model)
       # You more data in this way:
       #      self.mySpecificData = ...
```

ddliu365, 2021-01-14:
great, it works. sorry, I missed that part.

Another quick question: it seems in crocoddyl the activation function's Hessian matrix is diagonal by default.

But in my case, its Hessian matrix is not  a diagonal matrix anymore, how to solve this problem?

[image]

cmastalli, 2021-01-14:
Yes, we have temporarily defined as diagonal to improve performance. However, we need to work on a better solution.

In this case, I suggest you to write the whole cost function.

ddliu365, 2021-01-14:
do you think it is feasible to redefine the another activation base by removing the diagonal type declaration for `Arr` on my side?

cmastalli, 2021-01-14:
> do you think it is feasible to redefine the another activation base by removing the diagonal type declaration for `Arr` on my side?

We can only define a single base class, anything else is an inherited class.
This base class assumes that the structure of the Hessian is diagonal.

There are different workarounds, but I need to evaluate them before moving forward.

For now, you will need to develop the entire cost function as this method does not do this assumption. The way of doing it is very similar, so it should not be harder.

ddliu365, 2021-01-14:
I see. Can I start to develop by inheriting the `CostModelAbstract` class and `CostDataAbstract` in python?

proyan, 2021-01-14:
Hi @ddliu365 

We use activation, cost, etc as separate models, since the same activation can be used in multiple cost models.
In your case, you need a highly customized cost. You don't need an activation model in your custom cost. You can define the residual, and the cost directly inside your costmodel.

ddliu365, 2021-01-14:
thanks. I don't customize the cost before. Is there a minimal template to do that? If I don't inherit any class in crocoddyl, can the solver integrate the customized function from the coding view?

proyan, 2021-01-15:
I don't understand what you mean. You only have to inherit from the 
CostModelAbstract. Then it is up to you to define calc(costdata, x,u) 
and calcDiff(costdata, x, u) functions. You can create these functions 
in the manner that you want. The only thing that is used is 
costdata->cost and costdata->Lx,Lu,Lxx,Luu,Lxu derivatives.

cmastalli, 2021-01-15:
@ddliu365 -- I think the previous answers are not clear for new users.

The cost functions can be used for any action model; however, if you use them with one of our action models, then you need to consider that all the computations are carried out in the action model. This is our design choice, to avoid unnecessary extra computations. All our action models computes Pinocchio quantities; however, some of them might also include contact/impulse quantities. Finally, the results of these computations are stored in a shared data.

Coming back to your problem, it seems your cost model depends on the frame velocities. If so, you need to be aware that those Pinocchio computations are already computed and stored by our action models (all of them). Then, if this is your case, you should get inspired by this code that I developed: https://github.com/loco-3d/crocoddyl/blob/devel/bindings/python/crocoddyl/utils/__init__.py#L802-L835

I hope this answer helps you. If so, could I close this issue?

(With our action models I meant: `DifferentialModelFwdDynamics`, `DifferentialModelContactFwdDynamics` and `ActionModelImpulseFwdDynamics`)

ddliu365, 2021-01-15:
@cmastalli @proyan 

Thanks for your timely response.

This is the exact code base I want. I will work on that then.

We can close this issue.

closed 2021-01-15 by cmastalli