loco-3d/crocoddyl #906: double support cop cost

ddliu365, 2021-01-17:
Hi,

I am wondering if we have a way to define a double support cop cost based on current libraries.

#792 defines a single rectangular foot hull for cop constraint, but for the double support case, it no longer holds.

for example,
[image]

cmastalli, 2021-01-17:
You are interpreted wrongly the CoP cost. It implements a cost penalization producedure to keep the foot's CoP within a rectangular.

It is not the whole CoP of the system, neither the ZMP point.

This cost holds for two feet on ground.

ddliu365, 2021-01-17:
sorry, I think I mixed up cop and zmp here. cop should always be in or on the boundary of each foot, but zmp could be near somewhere between two feet. 

do you think it is possible to implement zmp cost for a double support based on current libraries

cmastalli, 2021-01-18:
> do you think it is possible to implement zmp cost for a double support based on current libraries

It is an unnecessary development. By mechanical principles, if you constrain both CoP to be within their respective sole, then, as direct consequence, the ZMP is going to be within the support region in a flat contact conditions. The current approach in Crocoddyl is also universal, in contrast to the ZMP approach that you're suggesting, as it works for noncoplanar surface as well.

You suggestion only makes sense if you define an inverted pendulum (or any variant of it) model. However, in Crocoddyl, we are not now interested in such kind of models.

I will close this issue as I believe it provides a clear answer to your question.

closed 2021-01-18 by cmastalli