loco-3d/crocoddyl #913: Access to different cost values

amrn9674, 2021-02-18:
Hello there, I am new to crocoddyl and pinocchio, and I am trying to learn these packages by controling different aspects of a robot (panda in this case). So far, I've become familiar with controlling a frame position (gripper_pose) using crocoddyl and it works fast and fine, however, when I try to add cost about the position of center of mass, it does not affect the previous trajectory (i.e, the trajectory is as same as when there was no CostModelCoMPosition). To debug this and see what is going underline the code, I need to see how much the values of different costs are (e.g., gripper_pose cost, state_costs, ComTrack cost, etc.), but unfortunately I cannot find out how to do so. I would be so grateful if you could help me through.

proyan, 2021-02-19:
Use 
```
cost_val = solver.problem.runningDatas[ i ].differential.costs.costs["gripper_pose_cost"].cost`
```
to get the value of `gripper_pose_cost` for the `i`th problem node in your `solver`.

Use 
```
cost_weight = solver.problem.runningModels[ i ].differential.costs.costs["gripper_pose_cost"].weight
```
to get the value of the cost weight. 

The total cost that is being optimized = `cost_weight * cost_val`

Does this answer your question?

amrn9674, 2021-02-19:
Thanks for your reply, 
Some how yes, I mean I did want to find cost of all node, but I can do it in a for loop. However, in my case, I had to make some changes to receive the cost value:
`cost_val = solver.problem.runningDatas[ i ].differential[a number, in my case between 0 and 4].costs.costs["cost_name"].cost`
Can you explain me what that number is? (the number that I have to provide after differential)

cmastalli, 2021-02-19:
> Any way, in my case, I had to make some changes to receive the cost value:
> `cost_val = solver.problem.runningDatas[ i ].differential[a number, in my case between 0 and 3].costs.costs["cost_name"].cost`

This doesn't make sense.
Note that the `differential` returns all the data contained in the differential action model.

Let us know if this clarifies your doubts, if so, could you close this issue?

proyan, 2021-02-19:
can you return the output of 
`dir(solver.problem.runningDatas[ i ].differential)`?

amrn9674, 2021-02-19:
yes, it returns:
['__class__',
 '__contains__',
 '__delattr__',
 '__delitem__',
 '__dict__',
 '__dir__',
 '__doc__',
 '__eq__',
 '__format__',
 '__ge__',
 '__getattribute__',
 '__getinitargs__',
 '__getitem__',
 '__getstate__',
 '__gt__',
 '__hash__',
 '__init__',
 '__instance_size__',
 '__iter__',
 '__le__',
 '__len__',
 '__lt__',
 '__module__',
 '__ne__',
 '__new__',
 '__reduce__',
 '__reduce_ex__',
 '__repr__',
 '__safe_for_unpickling__',
 '__setattr__',
 '__setitem__',
 '__setstate__',
 '__sizeof__',
 '__str__',
 '__subclasshook__',
 '__weakref__',
 'append',
 'extend',
 'tolist']

gfadini, 2021-02-19:
Hi, I think I saw what @amrn9674 is referring to when trying out the RK4 integrator. In that case there is indeed a list of data to perform the necessary computations.
```
In [10]: ddp.problem.runningDatas[500].differential[3]
Out[10]: <crocoddyl.libcrocoddyl_pywrap.DifferentialActionDataFreeFwdDynamics at 0x7fe5ec10a820>
```
If this is the case, then note that with the Euler integration method you perform a simple step, while with the RK4 method you need to compute multiple intermediate steps to perform a single integration step. The datas hence should refer to those intermediate calculations. I hope this helps

amrn9674, 2021-02-19:
@gfadini yes, it makes sense. Thanks for your answer.

closed 2021-02-19 by amrn9674

gfadini, 2021-02-19:
your're welcome @amrn9674

proyan, 2021-02-19:
Yup, that's right @gfadini. Thanks.
@amrn9674 Since you are looking at the cost values, in an RK4 integrator, you would have to do 
`gripper_cost = (cost[0] + 2. * cost[1] + Scalar(2.) * cost[2] + cost[3])/ 6.`
to get the final cost. `cost[ i ]` corresponds to the data inside `differential [ i ]`