loco-3d/crocoddyl #914: Control of center of Mass Position

amrn9674, 2021-02-19:
Dear all,

I am trying to control the position of a robot's Center of Mass (CoMPosition) using crocoddyl. Unfortunately, I found that adding a cost with CostMoelCoMPosition has no effect on the trajectory (it remains as same as when there is no comTracking cost). Checking with CoMPositionCostModelDerived, I found that all elements of "data.shared.pinocchio.com[0]" and "data.shared.pinocchio.Jcom" are zero. I replaced these functions with "pin.centerOfMass(rmodel,rdata,x[:rmodel.nq,:])" and "pin.jacobianCenterOfMass(rmodel,rdata,x[:rmodel.nq,:])", respectively, and now the code works fine, but it is slow. Two questions:
1- Why do those first functions always return zero matrices? Is there any problem on my side?
2- Is the reason that the code is slow because I am using the new functions or finding a trajectory for com position is intrinsically time consuming task?

Thank you  in advance.

proyan, 2021-02-19:
> Why do those first functions always return zero matrices? Is there any problem on my side?

What do you mean by first functions here? Could you please specify which differential action model you are using?

amrn9674, 2021-02-19:
Sorry for my bad referring. By first functions, I meant "data.shared.pinocchio.com[0]" and "data.shared.pinocchio.Jcom".

cmastalli, 2021-02-19:
> Sorry for my bad referring. By first functions, I meant "data.shared.pinocchio.com[0]" and "data.shared.pinocchio.Jcom", and I use crocoddyl.DifferentialActionModelFreeFwdDynamics as my differential action model

By design, we don't update the whole-body CoM information in the free dynamics. Therefore, it is expected to have zero values. Furthermore, you might be aware that the CoM is always controlled by the contact forces (c.f., the centroidal dynamics).

In short, I don't see a real motivation of updating the whole-body CoM information in the free dynamics.

Let me know if this helps you

amrn9674, 2021-02-19:
Sure, that was very helpful. Thanks for your answer.

cmastalli, 2021-02-19:
You're welcome.
I will close this issue as it seems to be resolved.

Btw, thank you for playing with Crocoddyl!

closed 2021-02-19 by cmastalli