loco-3d/crocoddyl #919: CoP constraint with Z rotation

ddliu365, 2021-03-02:
I checked #907 and found an argument of `FrameCoPSupport` has not been synchronized with new `CoPSupport` by taking rotation matrix.
I guess we may follow this style:
`copsupport = crocoddyl.CoPSupport(foot_rotation,foot_box)
copframe = crocoddyl.FrameCoPSupport(i, copsupport)
CoP = crocoddyl.CostModelContactCoPPosition(state, copframe, nu)
costModel.addCost("CoP", CoP, 1e1)`
However, the function `crocoddyl.FrameCoPSupport(i, copsupport)` has not been updated yet.

cmastalli, 2021-03-02:
Hi @ddliu365,

I did not update `FrameCoPSupport` (in #907) as this function will be deprecated soon.
Indeed, we are in the process of deprecating all `FrameXXX` structures as they add an unnecessary API complexity.
Part of this effort has been developed in #899.

I will close this issue as it won't lead to any action from our side.

closed 2021-03-02 by cmastalli

ddliu365, 2021-03-02:
ok, is there a way to apply cop cost with rotation matrix in current code?
or should I wait for another update

cmastalli, 2021-03-02:
It would be possible after merging #899.
This PR might take us around 2 weeks for merging it (we need to check that everything is working properly).

ddliu365, 2021-04-26:
Can I directly build the devel branch for this feature?
Is there an alternative way I can obtain this feature