loco-3d/crocoddyl #921: Is the friction cone constraint correct?

mayataka, 2021-03-16:
Dear all,

I have a question about the friction cone constraint, `CostModelContactFrictionConeTpl`.
This class imposes a penalty based on the contact force `d->contact->jMf.actInv(d->contact->f).linear()`.
However, this force seems represented in the end-effector's frame.
I wonder there needs some transformation of the axis to impose the friction cone constraints correctly.

Thank you.

proyan, 2021-03-16:
Hi @mayataka 
Welcome to crocoddyl!
The contact force, as well as the friction cone are defined in the local frame. 
If you wish to move them to the world frame, you would need to move the friction cone as well. This corresponds to the `A` matrix in the `FrameFrictionCone` class. Does this answer your question?

mayataka, 2021-03-17:
@proyan 
Thank you for replying!
I understand that the friction cone currently provided is considered in the local frame.
With my understanding, the reference frame of the friction cone should be the contact surface, e.g., its axes should be identical to the world frame (LOCAL_WORLD_ALIGNED in Pinoccho's convention) for flat ground.
I also understand that if I want to use the friction cone defined in the world frame, I should implement it.

cmastalli, 2021-03-17:
@mayataka -- Thank you for being interested in our project.

We indeed perform to such kind of rotation but at the level of the friction cone matrix. We also follow the same pattern for the contact wrench cone. Is this clarifying your doubts?

mayataka, 2021-03-17:
@cmastalli 
Thank you for replying.
I clearly understand what is implemented.
Thank you very much again for @proyan @cmastalli.

closed 2021-03-17 by mayataka