loco-3d/crocoddyl #922: MacOS installation through Homebrew

Neotriple, 2021-03-16:
Hi all,

Just curious if there are any plans of having Crocoddyl installed through the Homebrew package for installation through MacOS (similar to Pinocchio).

jcarpent, 2021-03-16:
Conda is your best friend at this time!

Neotriple, 2021-03-16:
Thanks, I didn't see that it was available through Conda either. Do you know how well this plays well with using it as a C++ library? I've only ever used Conda for Python environments.

jcarpent, 2021-03-16:
It works like a charm. I'm now using Conda as the base source for all my project, and importantly, for developing Pinocchio! ;)

Neotriple, 2021-03-16:
Great, thanks!

closed 2021-03-16 by Neotriple

Danfoa, 2023-02-15:
> It works like a charm. I'm now using Conda as the base source for all my projects, and importantly, for developing Pinocchio! ;)

@jcarpent Would you mind explaining how are you using conda to do source installs (i.e., being able to modify/compile and contribute to pinocchio/crocoddyl).

Though conda I can only install latest release not devel branches and furthermore packages are not editable, so no developing can be done.

jcarpent, 2023-02-15:
You can simply do:
```
conda install crocoddyl -c conda-forge --only-deps
```
and in your crocoddyl build directory do:
```
cd build
cmake .. -DCMAKE_BUILD_TYPE=Release -DCMAKE_INSTALL_PREFIX=$CONDA_PREFIX -DPYTHON_EXECUTABLE=$(which python)
make install
```