loco-3d/crocoddyl #923: MacOS installation through Conda

ddliu365, 2021-03-18:
I try to install the crocoddyl through Conda.

It comes with issues as shown in this log.
[crocoddyl_log.txt](https://github.com/loco-3d/crocoddyl/files/6165524/crocoddyl_log.txt)

proyan, 2021-03-18:
Hi @ddliu365 This looks like a problem in your conda installation, and not crocoddyl.

closed 2021-03-18 by proyan

jcarpent, 2021-03-18:
I confirmed, croco is working well with conda install on macs.
The problem is on your side.

ddliu365, 2021-03-18:
is there any specific version of Conda I should use?

I am using this python 3.8 [anaconda](https://repo.anaconda.com/archive/Anaconda3-2020.11-MacOSX-x86_64.sh)
```
Anaconda3-2020.11-MacOSX-x86_64
```

I also installed pinocchio which works well.

ddliu365, 2021-03-18:
@jcarpent @proyan 
Thanks for your quick response.

I will check my side.

jcarpent, 2021-03-18:
Should work with all versions. People from conda are doing an excellent job at this level.
We are not the best persons to answer your local computer problems.