loco-3d/crocoddyl #924: Enabling bounds on control

jlandais, 2021-04-06:
Hello,
I am new to Crocoddyl and I have a little question about how to use the bounds on control, it's possibly silly but I haven't figured out how to solve it yet. Once I have defined the lower and upper bounds on control of the Differential Action Model and the Integrated Action Model that I use, what should I do to make the solver take these bounds into account ?

It will be clearer with an example, here is the cartpole swing up script that comes with Crocoddyl. Let's say I want the torque to be in [-2.5, 2.5]  :

```
import numpy as np
import matplotlib.pyplot as plt
import crocoddyl

class DifferentialActionModelCartpole(crocoddyl.DifferentialActionModelAbstract):
    def __init__(self):
        crocoddyl.DifferentialActionModelAbstract.__init__(self, crocoddyl.StateVector(4), 1, 6)  # nu = 1; nr = 6
        self.unone = np.zeros(self.nu)

        self.m1 = 1.
        self.m2 = .1
        self.l = .5
        self.g = 9.81
        self.costWeights = [1., 1., 0.1, 0.001, 0.001, 1.]  # sin, 1-cos, x, xdot, thdot, f
        self.u_lb = np.array([-2.5])
        self.u_ub = np.array([2.5])

    def calc(self, data, x, u=None):
        if u is None: u = model.unone
        # Getting the state and control variables
        y, th, ydot, thdot = np.asscalar(x[0]), np.asscalar(x[1]), np.asscalar(x[2]), np.asscalar(x[3])
        f = np.asscalar(u[0])

        # Shortname for system parameters
        m1, m2, l, g = self.m1, self.m2, self.l, self.g
        s, c = np.sin(th), np.cos(th)

        # Defining the equation of motions
        m = m1 + m2
        mu = m1 + m2 * s**2
        xddot = (f + m2 * c * s * g - m2 * l * s * thdot**2) / mu
        thddot = (c * f / l + m * g * s / l - m2 * c * s * thdot**2) / mu
        data.xout = np.matrix([xddot, thddot]).T

        # Computing the cost residual and value
        data.r = np.matrix(self.costWeights * np.array([s, 1 - c, y, ydot, thdot, f])).T
        data.cost = .5 * np.asscalar(sum(np.asarray(data.r)**2))

    def calcDiff(self, data, x, u=None):
        # Advance user might implement the derivatives
        pass

# Creating the DAM for the cartpole
cartpoleDAM = DifferentialActionModelCartpole()
cartpoleData = cartpoleDAM.createData()
cartpoleDAM = model = DifferentialActionModelCartpole()

# Using NumDiff for computing the derivatives. We specify the
# withGaussApprox=True to have approximation of the Hessian based on the
# Jacobian of the cost residuals.
cartpoleND = crocoddyl.DifferentialActionModelNumDiff(cartpoleDAM, True)

# Getting the IAM using the simpletic Euler rule
timeStep = 5e-2
cartpoleIAM = crocoddyl.IntegratedActionModelEuler(cartpoleND, timeStep)
cartpoleIAM.u_lb = np.array([-2.5])
cartpoleIAM.u_ub = np.array([2.5])
# Creating the shooting problem
x0 = np.array([0., 3.14, 0., 0.])
T = 50

terminalCartpole = DifferentialActionModelCartpole()
terminalCartpoleDAM = crocoddyl.DifferentialActionModelNumDiff(terminalCartpole, True)
terminalCartpoleIAM = crocoddyl.IntegratedActionModelEuler(terminalCartpoleDAM)
terminalCartpoleIAM.u_lb = np.array([-2.5])
terminalCartpoleIAM.u_ub = np.array([2.5])

terminalCartpole.costWeights[0] = 100
terminalCartpole.costWeights[1] = 100
terminalCartpole.costWeights[2] = 1.
terminalCartpole.costWeights[3] = 0.1
terminalCartpole.costWeights[4] = 0.01
terminalCartpole.costWeights[5] = 0.0001
problem = crocoddyl.ShootingProblem(x0, [cartpoleIAM] * T, terminalCartpoleIAM)

# Solving it using DDP
ddp = crocoddyl.SolverDDP(problem)
ddp.setCallbacks([crocoddyl.CallbackVerbose()])
ddp.solve([], [], 300)

plt.plot(ddp.us)
plt.title("Torque (Nm)")
plt.show()
```
I added the lines
```
self.u_lb = np.array([-2.5])
self.u_ub = np.array([2.5])
```
```
cartpoleIAM.u_lb = np.array([-2.5])
cartpoleIAM.u_ub = np.array([2.5])
```
```
terminalCartpoleIAM.u_lb = np.array([-2.5])
terminalCartpoleIAM.u_ub = np.array([2.5])
```
to the original script to set the control bounds, but the torque found by Crocoddyl goes up to 10Nm, as you can see with the plot at the end. So the bounds are clearly not considered by the solver.

I apologize if the answer is obvious, I haven't found it in the notebook examples.

Thank you !

cmastalli, 2021-04-06:
You are using the DDP solver which by definition it doesn't handle control limits. Instead, you should use the BoxFDDP solver. For more details about this latter algorithm, you should read https://arxiv.org/pdf/2010.00411

Let me know if you have further questions. Otherwise, I would kindly ask you to close this issue ;)

jlandais, 2021-04-07:
Thank you very much !

closed 2021-04-07 by jlandais