loco-3d/crocoddyl #926: goal state specifed by target position & velocity

ndehio, 2021-04-15:
Hello, I am new to crocoddyl.

Is it possible to plan a trajectory with crocoddyl by specifying a goal state that comprises the desired end-effector pose and the desired end-effector velocity? In my case, I want to optimize the robot's joint position and joint velocity profiles simultaneously.
Thanks for your help
Niels

cmastalli, 2021-04-15:
Hi @ndehio,

Thank you for being interested in Crocoddyl.
Currently, we do not have an algorithm that explicitly handles terminal constraint (nether equality or inequality ones).
However, it is possible to softly constrain the terminal state, both using equality or inequality, by using a quadratic penalty function.

We are currently working in general algorithms that handle these (and other) constraints. But this is an ongoing research topic for us.

ndehio, 2021-04-15:
Thank you for the quick answer. Softly constraining the terminal state via a quadratic cost function should be sufficient for my purpose.

This [paragraph](https://gepettoweb.laas.fr/doc/loco-3d/crocoddyl/master/doxygen-html/index.html#OverviewConclu) mentioned some examples but I could not find them. Can you provide me a link?

proyan, 2021-04-15:
Hi Niels

Good to see you here.

The examples are in the examples folder https://www.github.com/loco-3d/crocoddyl/tree/master/examples, and detail the setups for simple (like double pendulum) and some complex (like humanoid legs backflipping) traj optimation problems

ndehio, 2021-04-16:
Great, I managed to install crocoddyl from source and tried to extend the talos-arm example, [see attached file](https://github.com/loco-3d/crocoddyl/files/6327051/reachtest.txt) (github doesn't allow me to upload a python script...)

I have two questions:

- I tried to compute (and plot) the end-effector position, but it seems it does not correspond to the quantity minimized by the solver for reaching a position. I guess there is a mismatch with my "definition" of the end-effector?
```
    pinocchio.forwardKinematics(robot_model,data,qFinal,qdFinal)
    eePosFinal = data.oMi[ee_joint_id].copy()
    eeVelFinal = data.ov[ee_joint_id].copy()
```
- How do I compute the end-effector velocity correctly such that it matches the end-effector velocity task?
```
    Vref = crocoddyl.FrameMotion(robot_model.getFrameId(ee_joint), eeVelDesired)
    goalTrackingCost2 = crocoddyl.CostModelFrameVelocity(state, Vref)
```

cmastalli, 2021-04-16:
With Pinocchio, you are computing the placement and velocity of the body at its joint frame. Instead, you want to compute these terms at the frame coordinates. You could either map these quantities as 'jMf.act(.)', or compute them using frame FK.

Regarding your second question, to be honest, I don't understand it. You could describe in which frame your are expressing the desired spatial velocity. We follow Pinocchio standard for that, i.e., 'LOCAL', 'WORLD' and 'LOCAL_WORLD_ALIGNED'.

ndehio, 2021-04-16:
Where can I find an example demonstrating frame FK?

proyan, 2021-04-17:
Hi Niels, 

The doc for getting the frame velocity is: 
https://gepettoweb.laas.fr/doc/stack-of-tasks/pinocchio/master/doxygen-html/namespacepinocchio.html#aeff1f1c50eaa6a706ac2271a4d70a2cd

If I understand, you are defining the final eePos as `data.oMi[ee_joint_id]`. Normally, all joints are automatically registered as frames. If your joint is indeed your end-effector, the value stored in data.oMi[ee_joint_id] should correspond to that in data.oMf

A simple example:
```python
import pinocchio
from example_robot_data import loadTalos
r = loadTalos()
joint_name = "gripper_left_joint"
q0 = pinocchio.randomConfiguration(r.model)

# Get the Joint Position
pinocchio.forwardKinematics(r.model, r.data, q0)
oMi_val = r.data.oMi[r.model.getJointId(joint_name)].copy()

#And the frame position:
pinocchio.framesForwardKinematics(r.model, r.data, q0)
oMf_val = r.data.oMf[r.model.getFrameId(joint_name)].copy()

# Both match
assert (oMf_val == oMi_val)
```

ndehio, 2021-04-19:
Thank you all so much for your support!
I almost solved my planning problem today. It seems crocoddyl is an easy-to-use library for solving sophisticated problems.
Are there maybe some lecture slides or something similar available to learn more about crocoddyl (besides the official publications)?

Getting more familar with crocoddyl, and studdying the different examples, I have few more naive questions:

1. Is it possible to define upper and lower limits for the control inputs u? I noticed there are torque limits stored in `robot_model.effortLimit`. Are these automatically transformed into a cost (with some specific weight)?

2. What is the difference between `robot_dataA` and `robot_dataB`? When shall I use which?
```
robot_arm = example_robot_data.load(robot_name)
robot_dataA = robot_arm.model.createData()
robot_dataB = robot_arm.data
```

3. What is the difference between `stateBoundsA` and `stateBoundsB` (the additional parameter=1...)?
```
stateBoundsA = crocoddyl.ActivationBounds(np.concatenate([ql, qdotl]), np.concatenate([qu, qdotu])) 
stateBoundsB = crocoddyl.ActivationBounds(np.concatenate([ql, qdotl]), np.concatenate([qu, qdotu]), 1) 
```

4. What is the difference between `stateLimitCostA` and `stateLimitCostB`?
```
stateLimitCostA = crocoddyl.CostModelState(state, activationStateBounds, 0 * stateInitial)
stateLimitCostB = crocoddyl.CostModelState(state, activationStateBounds, stateInitial, numJoints)
```

5. What is the meaning of these `armature` values? 
```
runningModel = crocoddyl.IntegratedActionModelEuler(runningModel, dt)
runningModel.differential.armature = 0.1*np.ones(1, numJoints)
```
In an example with a motion sequence no `armature` values are defined?
```
seqs = [[crocoddyl.IntegratedActionModelEuler(runningModel, dt)] * T +
        [crocoddyl.IntegratedActionModelEuler(terminalModel)]
        for runningModel, terminalModel in zip(runningModels, terminalModels)]
problem = crocoddyl.ShootingProblem(stateInitial, sum(seqs, [])[:-1], seqs[-1][-1])
```

6. How can I specify a joint-posture task, such that the robot moves to its home posture at the end of a motion sequence? My idea was to use state bounds with lower=upper limits but maybe there is a better solution?
```
restStateBounds = crocoddyl.ActivationBounds(np.concatenate([qPosture, qdRest]), np.concatenate([qPosture, qdRest]))
```

Thank you again!

cmastalli, 2021-04-20:
>     1. Is it possible to define upper and lower limits for the control inputs u? I noticed there are torque limits stored in `robot_model.effortLimit`. Are these automatically transformed into a cost (with some specific weight)?

Yes, we have a solver that handles control bounds, its name is `Box-FDDP` (see this paper: https://arxiv.org/pdf/2010.00411.pdf). These effort limits are automatically parse, and only consider if you use `Box-FDDP`.

>     2\. What is the difference between `robot_dataA` and `robot_dataB`? When shall I use which?

This question is more related to Pinocchio; however, I could provide a short answer. Both data objects are created for the respective Pinocchio model; however, the second one is used internally in the RobotWrapper class. Note that Crocoddyl doesn't use neither of them, the action model creates its own data.

> 3. What is the difference between `stateBoundsA` and `stateBoundsB` (the additional parameter=1...)?

It is a safety margin that you could include in the bounds. See the Crocoddyl documentation for more details.

> 4. What is the difference between `stateLimitCostA` and `stateLimitCostB`?

You are passing different reference values for the cost function. See the Crocoddyl documentation for more details.

> 5. What is the meaning of these `armature` values?

It is the rotor inertia of an electrical motor. Note that wee have not a definite API for handling arbitrary actuators. Most of the time you don't want to define the armature values.

>     6. How can I specify a joint-posture task, such that the robot moves to its home posture at the end of a motion sequence? My idea was to use state bounds with lower=upper limits but maybe there is a better solution?

If you know what it is the desired posture, you must want to use a quadratic function to track  it. You don't need to define upper and lower bounds of the joint configuration.

cmastalli, 2021-04-20:
Let us know if we can close this issue.

ndehio, 2021-04-20:
Thanks for the detailed explanation!

closed 2021-04-20 by ndehio