loco-3d/crocoddyl #929: GepettoVisualizer object has no attribute viewer

ndehio, 2021-04-22:
Hi,
so far I was using crocoddyl without visualizing the motion.

After having some [installation troubles](https://github.com/Gepetto/gepetto-viewer/issues/170) I did `sudo apt-get install openscenegraph-3.4 libopenscenegraph-3.4-dev` and then I was able to install osgQt, gepetto-viewer and gepetto-viewer-corba.

Next I tried to run a crocoddyl example with `export CROCODDYL_DISPLAY=1` and `export CROCODDYL_PLOT=1` and got this problem:
```
ndehio@ndehio-5540:~/src/crocoddyl/examples$ python arm_manipulation.py 
/usr/local/lib/python2.7/dist-packages/pinocchio/robot_wrapper.py:272: UserWarning: Error while starting the viewer client.
Check whether gepetto-viewer is properly started
  self.viz.initViewer(*args, **kwargs)
Traceback (most recent call last):
  File "arm_manipulation.py", line 69, in <module>
    display = crocoddyl.GepettoDisplay(talos_arm, 4, 4, cameraTF)
  File "/usr/local/lib/python2.7/dist-packages/crocoddyl/__init__.py", line 100, in __init__
    self._addRobot()
  File "/usr/local/lib/python2.7/dist-packages/crocoddyl/__init__.py", line 239, in _addRobot
    self.robot.loadViewerModel(rootNodeName="robot")
  File "/usr/local/lib/python2.7/dist-packages/pinocchio/robot_wrapper.py", line 294, in loadViewerModel
    self.viz.loadViewerModel(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/pinocchio/visualize/gepetto_visualizer.py", line 137, in loadViewerModel
    gui = self.viewer.gui
AttributeError: 'GepettoVisualizer' object has no attribute 'viewer'
```

cmastalli, 2021-04-22:
Did you run the Gepetto GUI (i.e., `gepetto-gui` command) in another terminal?

ndehio, 2021-04-22:
Oh I didn't know about this. Running `gepetto-gui` command in another terminal produces this output:
```
ndehio@ndehio-5540:~/src/crocoddyl/examples$ python arm_manipulation.py 
/usr/local/lib/python2.7/dist-packages/pinocchio/visualize/gepetto_visualizer.py:154: UserWarning: Error while loading geometry object: arm_left_1_link_0
Error message:
CORBA.COMM_FAILURE(omniORB.COMM_FAILURE_WaitingForReply, CORBA.COMPLETED_MAYBE)
  self.loadViewerGeometryObject(collision,pin.GeometryType.COLLISION)
/usr/local/lib/python2.7/dist-packages/pinocchio/visualize/gepetto_visualizer.py:154: UserWarning: Error while loading geometry object: arm_left_2_link_0
Error message:
CORBA.TRANSIENT(omniORB.TRANSIENT_ConnectFailed, CORBA.COMPLETED_NO)
  self.loadViewerGeometryObject(collision,pin.GeometryType.COLLISION)
/usr/local/lib/python2.7/dist-packages/pinocchio/visualize/gepetto_visualizer.py:154: UserWarning: Error while loading geometry object: arm_left_3_link_0
Error message:
CORBA.TRANSIENT(omniORB.TRANSIENT_ConnectFailed, CORBA.COMPLETED_NO)
  self.loadViewerGeometryObject(collision,pin.GeometryType.COLLISION)
/usr/local/lib/python2.7/dist-packages/pinocchio/visualize/gepetto_visualizer.py:154: UserWarning: Error while loading geometry object: arm_left_4_link_0
Error message:
...
Error message:
CORBA.TRANSIENT(omniORB.TRANSIENT_ConnectFailed, CORBA.COMPLETED_NO)
  self.loadViewerGeometryObject(collision,pin.GeometryType.COLLISION)
Traceback (most recent call last):
  File "arm_manipulation.py", line 69, in <module>
    display = crocoddyl.GepettoDisplay(talos_arm, 4, 4, cameraTF)
  File "/usr/local/lib/python2.7/dist-packages/crocoddyl/__init__.py", line 100, in __init__
    self._addRobot()
  File "/usr/local/lib/python2.7/dist-packages/crocoddyl/__init__.py", line 239, in _addRobot
    self.robot.loadViewerModel(rootNodeName="robot")
  File "/usr/local/lib/python2.7/dist-packages/pinocchio/robot_wrapper.py", line 294, in loadViewerModel
    self.viz.loadViewerModel(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/pinocchio/visualize/gepetto_visualizer.py", line 156, in loadViewerModel
    self.displayCollisions(self.collision_model is not None and self.visual_model is None)
  File "/usr/local/lib/python2.7/dist-packages/pinocchio/visualize/gepetto_visualizer.py", line 205, in displayCollisions
    gui.setVisibility(nodeName,visibility_mode)
  File "/usr/local/lib/python2.7/dist-packages/gepetto/corbaserver/graphical_interface_idl.py", line 513, in setVisibility
    return self._obj.invoke("setVisibility", _0_gepetto.corbaserver.GraphicalInterface._d_setVisibility, args)
omniORB.CORBA.TRANSIENT: CORBA.TRANSIENT(omniORB.TRANSIENT_ConnectFailed, CORBA.COMPLETED_NO)
```
and 
```
$ gepetto-gui
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
Unknow keyboard layout
StatsHandler::StatsHandler() Setting up GL2 compatible shaders
Unknow keyboard layout
gepetto-gui: symbol lookup error: /usr/lib/x86_64-linux-gnu/libosgViewer.so.131: undefined symbol: _ZN11OpenThreads35SetProcessorAffinityOfCurrentThreadEj
```

jcarpent, 2021-04-22:
`conda install -c conda-forge gepetto-viewer-corba crocoddyl` is the best option

cmastalli, 2021-04-22:
You could also use `sudo apt robotpkg-qt4-gepetto-viewer*`.

I will close this issue as it is related to https://github.com/Gepetto/gepetto-viewer

closed 2021-04-22 by cmastalli

cmastalli, 2023-11-27:
@ndehio and @fukangl -- Please start using Meshcat. We might deprecate GV in future.

Later, I will create support for Rviz via this plugin: https://github.com/loco-3d/whole_body_state_rviz_plugin.
This is useful when working on the robot, we use daily basis in our MPC.

yyeunggg, 2024-02-14:
@cmastalli Hi, will meshcat/other visualizer be supported in pinocchio's cpp? We cannot get GV working, with the same issue in https://github.com/loco-3d/crocoddyl/issues/929#issuecomment-824711450.