loco-3d/crocoddyl #934: Moving to a more recent C++ standard

wxmerkt, 2021-04-23:
With 16.04 reaching EOL next month and C++11 about ten years old, I wonder if there are requirements to continue to support C++98 in the community? In practice we _implicitly_ already require C++11 support (e.g. for URDF parsing in Pinocchio which translates to Crocoddyl) and officially moving the minimum supported standard to a newer version could simplify development by benefitting from new features (e.g. using `chrono` in #931). I'd love to hear thoughts and plans on this topic for future developments :-).

cmastalli, 2021-04-23:
I am in with moving to c++11

nim65s, 2021-04-24:
I think we could even move directly to C++14. 

I've seen many cases where developers think they are using a C++11 feature, while it is in fact a C++14 feature (eg. using `auto` as a function return type), and it works on their computer because the default standard is >= 14, but raises boring issues when they want to share their work.

I think the oldest system we support is 16.04, and activating C++11 or C++14 on 16.04 is the same thing.

To answer about the implicit requirement, I don't think this is the case. On gitlab, pinocchio is using urdfdom v0.4 provided by ubuntu, which doesn't require C++11.

proyan, 2021-04-27:
I've already stated my preference for updating our base support to c++11. If we can move to c++14, that is even better from my perspective. Crocoddyl is supposed to be on the working on the robot, and I don't think any current robots still operate on 12.04.
And Ubuntu 14.04 has c++14 support.  (https://www.gnu.org/software/gcc/projects/cxx-status.html#cxx14)
So I am okay with moving directly to c++14.

cmastalli, 2021-05-12:
I think we can close this issue (see #949 for more details)

closed 2021-05-12 by cmastalli