loco-3d/crocoddyl #939: Crocoddyl errors when building in c++ (Installed through Conda)

Neotriple, 2021-04-26:
Hi there,

I recently installed croc. through a conda environment. However, when I try to build files using c++, I get the following error:

/home/pshah/anaconda3/envs/biconvex_mpc/lib/libcrocoddyl.so: undefined reference to `std::__cxx11::basic_ostringstream<char, std::char_traits<char>, std::allocator<char> >::basic_ostringstream()@GLIBCXX_3.4.26'
/home/pshah/anaconda3/envs/biconvex_mpc/lib/libcrocoddyl.so: undefined reference to `std::__cxx11::basic_stringstream<char, std::char_traits<char>, std::allocator<char> >::basic_stringstream()@GLIBCXX_3.4.26'

Any recommendations on how to fix this?

Neotriple, 2021-04-27:
PS: This seems to be an issue across multiple OpenRobot based packages (e.g. hpp-fcl, and pinocchio). 

A quick google seems to make me think this is something with the precompiled libraries not being made with the same gcc/g++: On my setup, I'm using version 7.5. I tried also defining:

#define _GLIBCXX_USE_CXX11_ABI 0

which did not seem to help

Neotriple, 2021-04-27:
This seems link to have fixed the issue, but it might be worth looking into as well:

https://github.com/apache/incubator-mxnet/issues/10187#issuecomment-414912008

Essentially I needed to create a link between the libstdc++ in my anaconda3 environment and the one located on my 'non-virtual' machine

jcarpent, 2021-04-27:
You need to compile your project with C++11.

closed 2021-04-27 by jcarpent