loco-3d/crocoddyl #955: BUILD_TESTING still builds unittests

proyan, 2021-05-29:
We are not using BUILD_TESTING in the correct way in crocoddyl, and as a result unittests are built even when it is turned off. 

The issue is assigned to @nim65s , but if someone has the time to fix it first, please change the assignment and go ahead.

cmastalli, 2021-05-29:
Please remove the cmake files and try again.

proyan, 2021-05-29:
Of course I tried with a clean build system.

The issue is that we are building the unittests along with the main repository, while the BUILD_TESTING is supposed to build the unittests separately by `make test`. This could be corrected by using the macro `ADD_UNIT_TEST`, but with the factory files, we would need to accomodate them in the testing, which is not how this macro is designed to be used.

nim65s, 2021-06-04:
Hi @proyan 

I can't reproduce your issue.

I did this:
```
mkdir build
cd build
cmake -DBUILD_TESTING=OFF ..
make
```

After this, I tried:
```
$ find . -name \*test\*
./CMakeFiles/build_tests.dir
```

I also tried: 
```
$ make test
Running tests...
/usr/bin/ctest --force-new-ctest-process 
Test project /home/nim/local/loco-3d/crocoddyl/build
      Start  1: example-python-double_pendulum
      Start  2: example-python-boxfddp_vs_boxddp
      Start  3: example-python-quadrotor
      Start  4: example-python-quadrotor_ubound
 1/12 Test  #3: example-python-quadrotor ......................   Passed    0.36 sec
      Start  5: example-python-arm_manipulation
 2/12 Test  #4: example-python-quadrotor_ubound ...............   Passed    0.40 sec
      Start  6: example-python-quadrupedal_gaits
 3/12 Test  #1: example-python-double_pendulum ................   Passed    0.95 sec
      Start  7: example-python-quadrupedal_walk_ubound
 4/12 Test  #5: example-python-arm_manipulation ...............   Passed    0.62 sec
      Start  8: example-python-bipedal_walk
 5/12 Test  #8: example-python-bipedal_walk ...................   Passed    2.79 sec
      Start  9: example-python-bipedal_walk_ubound
 6/12 Test  #6: example-python-quadrupedal_gaits ..............   Passed    3.72 sec
      Start 10: example-python-humanoid_manipulation
 7/12 Test  #7: example-python-quadrupedal_walk_ubound ........   Passed    3.81 sec
      Start 11: example-python-humanoid_manipulation_ubound
 8/12 Test  #2: example-python-boxfddp_vs_boxddp ..............   Passed    5.05 sec
      Start 12: example-python-humanoid_taichi
 9/12 Test  #9: example-python-bipedal_walk_ubound ............   Passed    3.10 sec
10/12 Test #10: example-python-humanoid_manipulation ..........   Passed    4.89 sec
11/12 Test #11: example-python-humanoid_manipulation_ubound ...   Passed    9.39 sec
12/12 Test #12: example-python-humanoid_taichi ................   Passed   12.65 sec

100% tests passed, 0 tests failed out of 12

Total Test time (real) =  17.70 sec
```

To me, its seems that unittests were not built when `BUILD_TESTING` was turned `OFF`. And this is the behavior I was expecting.

Now, I agree that one could expect that `make test`will actually build the unittests and launch them afterwards. This is what the `ADD_UNIT_TEST` macro provides. But as you mentioned, we are not using this macro, and instead we are purely deactivating the whole unittest CMake subtree when `BUILD_TESTING`is `OFF`.

So in order to build unit tests after a `-DBUILD_TESTING=OFF`, one can easily `cmake -DBUILD_TESTING=ON ..`, and `make` again. This time, unit tests are available: `find . -name \*test\* shows tons on results, and `make test` runs everything.

Could you detail your needs on this matter ? Is there anything that you can't do as you would like to, or anything that is done behind your back while you would prefer it not to be done ?

proyan, 2021-06-04:
I turned off all options, however, this is the build that I receive:
```sh
Script started on 2021-06-04 12:11:23+02:00 [TERM="xterm-256color" TTY="/dev/pts/3" COLUMNS="211" LINES="53"]
]0;rbudhira@pl834-pro: ~/devel/src/loco3d/crocoddyl[01;32mrbudhira@pl834-pro[00m:[01;34m~/devel/src/loco3d/crocoddyl[00m$ builds
-- Package version (ROS package.xml): 1.7.0
-- Contact RObot COntrol by Differential DYnamic programming Library (Crocoddyl), version 1.7.0
-- Copyright (C) 2018-2021 CNRS-LAAS, University of Edinburgh, INRIA
-- All rights reserved.
-- Released under the BSD 3-Clause License.
-- The CXX compiler identification is Clang 11.0.0
-- Check for working CXX compiler: /usr/bin/clang++
-- Check for working CXX compiler: /usr/bin/clang++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
CMake Warning (dev) at /usr/share/cmake-3.16/Modules/CTest.cmake:50 (option):
  Policy CMP0077 is not set: option() honors normal variables.  Run "cmake
  --help-policy CMP0077" for policy details.  Use the cmake_policy command to
  set the policy and suppress this warning.

  For compatibility with older versions of CMake, option is clearing the
  normal variable 'BUILD_TESTING'.
Call Stack (most recent call first):
  cmake/post-project.cmake:45 (INCLUDE)
  CMakeLists.txt:59 (PROJECT)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Performing Test R-pedantic
-- Performing Test R-pedantic - Success
-- Performing Test R-Wno-long-long
-- Performing Test R-Wno-long-long - Success
-- Performing Test R-Wall
-- Performing Test R-Wall - Success
-- Performing Test R-Wextra
-- Performing Test R-Wextra - Success
-- Performing Test R-Wcast-align
-- Performing Test R-Wcast-align - Success
-- Performing Test R-Wcast-qual
-- Performing Test R-Wcast-qual - Success
-- Performing Test R-Wformat
-- Performing Test R-Wformat - Success
-- Performing Test R-Wwrite-strings
-- Performing Test R-Wwrite-strings - Success
-- Performing Test R-Wconversion
-- Performing Test R-Wconversion - Success
-- Found Doxygen: /usr/bin/doxygen (found version "1.8.17") found components: doxygen missing components: dot
-- currently used  C++ standard: 201402
-- minimal C++ standard upgraded to 14
-- pinocchio FOUND. pinocchio at /local/rbudhira/devel/install/sot/lib/libpinocchio.so
-- boost_filesystem FOUND. boost_filesystem at /usr/lib/x86_64-linux-gnu/libboost_filesystem.so
-- boost_serialization FOUND. boost_serialization at /usr/lib/x86_64-linux-gnu/libboost_serialization.so
-- boost_system FOUND. boost_system at /usr/lib/x86_64-linux-gnu/libboost_system.so
-- Found Boost: /usr/include (found version "1.71.0") found components: filesystem serialization system 
-- eigenpy FOUND. eigenpy at /local/rbudhira/devel/install/sot/lib/libeigenpy.so
-- hpp-fcl FOUND. hpp-fcl at /local/rbudhira/devel/install/sot/lib/libhpp-fcl.so
-- Found Boost: /usr/include (found version "1.71.0") found components: chrono serialization 
-- scipy is optional.
-- Checking for module 'scipy'
--   No package 'scipy' found
-- Found Boost: /usr/include (found version "1.71.0") found components: filesystem serialization system 
-- Found Boost: /usr/include (found version "1.71.0") found components: unit_test_framework 
-- Doxygen rendering: using MathJax backend
-- Configuring done
-- Generating done
-- Build files have been written to: /local/rbudhira/devel/build/loco3d/crocoddyl/_build-RELEASE
make: Entering directory '/local/rbudhira/devel/build/loco3d/crocoddyl/_build-RELEASE'
/usr/bin/cmake -S/home/rbudhira/devel/src/loco3d/crocoddyl -B/local/rbudhira/devel/build/loco3d/crocoddyl/_build-RELEASE --check-build-system CMakeFiles/Makefile.cmake 0
/usr/bin/cmake -E cmake_progress_start /local/rbudhira/devel/build/loco3d/crocoddyl/_build-RELEASE/CMakeFiles /local/rbudhira/devel/build/loco3d/crocoddyl/_build-RELEASE/CMakeFiles/progress.marks
[... 371 lines cut ...]
/usr/bin/cmake -E cmake_progress_start /local/rbudhira/devel/build/loco3d/crocoddyl/_build-RELEASE/CMakeFiles 0

exit
```

As you can see, The unittests are getting built. I turn off all options in CMakeLists.txt:

```cmake
# Create different building options                                                                                                                                                                                
OPTION(ENABLE_VECTORIZATION "Enable vectorization and further processor-related optimizations" OFF)                                                                                                                
OPTION(BUILD_PYTHON_INTERFACE "Build the python binding" OFF)                                                                                                                                                      
OPTION(BUILD_BENCHMARK "Build the benchmark" OFF)                                                                                                                                                                  
OPTION(BUILD_EXAMPLES "Build the examples" OFF)                                                                                                                                                                    
[... cut ...]
ENDIF(DEFINED BUILD_UNIT_TESTS)                                                                                                                                                                                    
SET(BUILD_TESTING OFF)                                                                                                                                                                                             

```

I think the `SET(BUILD_TESTING OFF)` isn't working very well. Do you have an insight?

nim65s, 2021-06-04:
Can you `grep -r BUILD_TESTING` in your build dir ?

proyan, 2021-06-04:
Interesting. I get 
```
CMakeCache.txt:BUILD_TESTING:BOOL=ON
```
So I'm not able to set BUILD_TESTING from inside the CMakeLists. This is unexpected

nim65s, 2021-06-04:
https://github.com/Kitware/CMake/blob/master/Modules/CTest.cmake#L50 is a 

```cmake
option(BUILD_TESTING "Build the testing tree." ON)
```

In CMake, options are a shortcut to cached bool variables. And you can't update a cached variable with only `set(var val)`.

I'm sorry that CMake is such a pain sometimes, but C++ doesn't come with anything better :/

nim65s, 2021-06-04:
I strongly advise anybody against doing any local configuration by changing the `CMakeLists.txt`. In case you really want that footgun to work, you can try `set(BUILD_TESTING "VALUE" CACHE BOOL OFF FORCE)`, but you've been warn: please don't ;)

proyan, 2021-06-04:
I don't intend to force set variables. However, I expect to be able to set options in CMakeLists during development, you need different segments during different times.

Since we are not using BUILD_TESTING with the intended `ADD_UNIT_TEST` macro, I don't see any advantage of using `SET` here instead of `OPTION`. I think it would be better to change to 
```cmake
OPTION(BUILD_TESTING ${BUILD_UNIT_TESTS})
```
instead. What do you think?

proyan, 2021-06-04:
hmmm, thanks for the link.
It looks like `OPTION` doesn't create a copy variable, and insteads works on the upstream variable.  So `ADD_UNIT_TEST` is working on the copy variable, while the current build is processing with the original variable.

We are using the `OPTION` behaviour, while setting with `SET` instead.

> I'm sorry that CMake is such a pain sometimes, but C++ doesn't come with anything better :/
Agreed!

proyan, 2021-06-11:
Fixed with #958

PR #958 (2021-06-07): [cmake] use OPTION to set BUILD_TESTING
Closes #955
closed 2021-06-11 by proyan