loco-3d/crocoddyl #959: accuracy issue implemented by Python interface

ddliu365, 2021-06-05:
Hi,

I wrote down my own cost function based on `CostModelAbstract` in python. However, the optimization results are different when I test the code on different machines. 

I guess the difference might be caused by the interface between c++ and python.

Is there a way to fix this issue?

cmastalli, 2021-06-07:
Hi @ddliu365,

Thank you for being part of our community.

With the description of your issue, we cannot help you so much. Please take care when you write issues as we do this ad honorem and all of us are very busy.

Said that, you need to provide more details and relevant ones. You should try to inspect and understand your issue first.

In any case, there are obvious details that you should let us know:
 1. What are exactly the difference that you describe? Could you provide us some numbers?
 2. Which branch are you using in both PCs?
 3. Which OS are you using?
 4. Which compile flags are you using in both PCs?

Yours,
Carlos

proyan, 2021-06-11:
Since there is no further news on this, I'll close it.

closed 2021-06-11 by proyan