loco-3d/crocoddyl #970: ArgumentError when creating a differential action model

quentinll, 2021-06-28:
Hello,
I installed crocoddyl from conda and I am trying to implement a DifferentialActionModel by following the example from https://github.com/loco-3d/crocoddyl/blob/master/examples/notebooks/introduction_to_crocoddyl.ipynb .

When I run the following code:
```python
import crocoddyl
import numpy as np
import example_robot_data
import pinocchio

talos_arm = example_robot_data.load('talos_arm')
robot_model = talos_arm.model # getting the Pinocchio model

# Defining a initial state
q0 = np.array([0.173046, 1., -0.52366, 0., 0., 0.1, -0.005])
x0 = np.concatenate([q0, np.zeros(talos_arm.model.nv)])

class DifferentialFwdDynamics(crocoddyl.DifferentialActionModelAbstract):
    def __init__(self, state, costModel):
        crocoddyl.DifferentialActionModelAbstract.__init__(self, state, state.nv, costModel.nr)
        self.costs = costModel
        self.enable_force = True
        self.armature = np.zeros(0)

    def calc(self, data, x, u=None):
        if u is None:
            u = self.unone
        q, v = x[:self.state.nq], x[-self.state.nv:]
        # Computing the dynamics using ABA or manually for armature case
        if self.enable_force:
            data.xout = pinocchio.aba(self.state.pinocchio, data.pinocchio, q, v, u)
        else:
            pinocchio.computeAllTerms(self.state.pinocchio, data.pinocchio, q, v)
            data.M = data.pinocchio.M
            if self.armature.size == self.state.nv:
                data.M[range(self.state.nv), range(self.state.nv)] += self.armature
            data.Minv = np.linalg.inv(data.M)
            data.xout = data.Minv * (u - data.pinocchio.nle)
        # Computing the cost value and residuals
        pinocchio.forwardKinematics(self.state.pinocchio, data.pinocchio, q, v)
        pinocchio.updateFramePlacements(self.state.pinocchio, data.pinocchio)
        self.costs.calc(data.costs, x, u)
        data.cost = data.costs.cost

    def calcDiff(self, data, x, u=None):
        q, v = x[:self.state.nq], x[-self.state.nv:]
        if u is None:
            u = self.unone
        if True:
            self.calc(data, x, u)
        # Computing the dynamics derivatives
        if self.enable_force:
            pinocchio.computeABADerivatives(self.state.pinocchio, data.pinocchio, q, v, u)
            data.Fx = np.hstack([data.pinocchio.ddq_dq, data.pinocchio.ddq_dv])
            data.Fu = data.pinocchio.Minv
        else:
            pinocchio.computeRNEADerivatives(self.state.pinocchio, data.pinocchio, q, v, data.xout)
            data.Fx = -np.hstack([data.Minv * data.pinocchio.dtau_dq, data.Minv * data.pinocchio.dtau_dv])
            data.Fu = data.Minv
        # Computing the cost derivatives
        self.costs.calcDiff(data.costs, x, u)

    def set_armature(self, armature):
        if armature.size is not self.state.nv:
            print('The armature dimension is wrong, we cannot set it.')
        else:
            self.enable_force = False
            self.armature = armature.T

    def createData(self):
        data = crocoddyl.DifferentialActionModelAbstract.createData(self)
        data.pinocchio = pinocchio.Data(self.state.pinocchio)
        data.costs = self.costs.createData(data.pinocchio)
        data.costs.shareMemory(data) # this allows us to share the memory of cost-terms of action model
        return data

# Create the cost functions
target = np.array([0.4, 0., .4])
Mref = crocoddyl.FrameTranslation(robot_model.getFrameId("gripper_left_joint"), target)
state = crocoddyl.StateMultibody(robot_model)
goalTrackingCost = crocoddyl.CostModelFrameTranslation(state, Mref)
xRegCost = crocoddyl.CostModelState(state)
uRegCost = crocoddyl.CostModelControl(state)

# Create cost model per each action model
runningCostModel = crocoddyl.CostModelSum(state)
terminalCostModel = crocoddyl.CostModelSum(state)

# Then let's added the running and terminal cost functions
runningCostModel.addCost("gripperPose", goalTrackingCost, 1e2)
runningCostModel.addCost("stateReg", xRegCost, 1e-4)
runningCostModel.addCost("ctrlReg", uRegCost, 1e-7)
terminalCostModel.addCost("gripperPose", goalTrackingCost, 1e5)
terminalCostModel.addCost("stateReg", xRegCost, 1e-4)
terminalCostModel.addCost("ctrlReg", uRegCost, 1e-7)

# Running and terminal action models
DT = 1e-3
actuationModel = crocoddyl.ActuationModelFull(state)
runningModel = crocoddyl.IntegratedActionModelEuler(
    DifferentialFwdDynamics(state, runningCostModel), DT)
terminalModel = crocoddyl.IntegratedActionModelEuler(
    DifferentialFwdDynamics(state, terminalCostModel), 0.)

T = 250
problem = crocoddyl.ShootingProblem(x0, [runningModel] * T, terminalModel)

```
I get the following error:

```python
---------------------------------------------------------------------------
ArgumentError                             Traceback (most recent call last)
<ipython-input-5-d169cd962bf4> in <module>
      2 # models) plus a terminal knot
      3 T = 250
----> 4 problem = crocoddyl.ShootingProblem(x0, [runningModel] * T, terminalModel)
      5 

<ipython-input-2-58f8016a9eef> in createData(self)
     57         data = crocoddyl.DifferentialActionModelAbstract.createData(self)
     58         data.pinocchio = pinocchio.Data(self.state.pinocchio)
---> 59         data.costs = self.costs.createData(data.pinocchio)
     60         data.costs.shareMemory(data) # this allows us to share the memory of cost-terms of action model
     61         return data

ArgumentError: Python argument types in
    CostModelSum.createData(CostModelSum, Data)
did not match C++ signature:
    createData(crocoddyl::CostModelSumTpl<double> {lvalue} self, crocoddyl::DataCollectorAbstractTpl<double>* data)
```

How should I initialize data for the cost model ?

cmastalli, 2021-06-28:
Thank you for reporting an issue in our notebook. Unfortunately, we don't have a way of unittesting them.

I have provided a solution in #971 in which you can inspect quickly the differential action model.
As I short answer, you need to pass a data collector object when you create a cost data in line 59.
This data collector helps us to share data (without recomputing it) across cost functions (and in future constraints ones).
Note that you can also implement your own data-collector structure.

closed 2021-06-28 by cmastalli

proyan, 2021-07-02:
@quentinll The notebook would work for the devel installation. If you prefer to keep to conda, please wait a bit while the notebook is fixed for master branch :)