loco-3d/crocoddyl #985: URDF not represented correctly in gepetto-GUI

VasilyRakche, 2021-08-08:
Hi, 
I am working on my own legged robot and wanted to try out your solution for control, but I have problems when I try to use [my urdf and srdf files.](https://github.com/Legged-robot/robot_description_package) I have made quite simple robot representation and verified that it is correctly shown in RVIZ:
[image]

I used your script `example_robot_data` to upload my robot with this configurations:
```
class FOREloader(RobotLoader):
    path = 'robot_simple_description'
    urdf_filename = "robot_simplified.urdf"
    srdf_filename = "robot_simplified.srdf"
    ref_posture = "standing"
    free_flyer = True
```
But what I get is that all links are placed at the origin:
[image]

Also, I created my URDF based on the file provided by [ANYbotics anymal_b_simple_description ](https://github.com/ANYbotics/anymal_b_simple_description) and when I use [your adopted version from pinnochio](https://github.com/Gepetto/example-robot-data/tree/cac155e7fb266edfd57010df1c97b80e2473b89a/robots/anymal_b_simple_description), this is what I get (looks like the behavior is similar to what I get with my model, so I was wondering why): 

[image]

VasilyRakche, 2021-08-08:
I think all is good, it is expected from me to run 
```
robot.display(robot.q0)
```
Afterwards I get what I was expecting (still would be interested in short feedback from you if you think everything looks ok on the picture - one leg is modeled since I will be experimenting with one leg stand):
[image]

I just wander about one thing, why you [scale all the models with 1](https://github.com/Gepetto/example-robot-data/blob/cac155e7fb266edfd57010df1c97b80e2473b89a/robots/anymal_b_simple_description/robots/anymal.urdf#L169), but [in original it was scaled with 0.001](https://github.com/ANYbotics/anymal_b_simple_description/blob/988b5df22b84761bdf08111b1c2ccc883793f456/urdf/anymal.urdf#L93)? I hope it wont be too problematic to answer here so I wont be opening new issue somewhere else. Thank you.

cmastalli, 2021-08-08:
Hey @VasilyRakche,

This issue doesn't belong to Crocoddyl.

Please post your issues in the corresponding project.

closed 2021-08-08 by cmastalli