loco-3d/crocoddyl #986: Prismatic root_joint - controlling the Robot leg on the teststand

VasilyRakche, 2021-08-14:
### Setup
I am trying to create a good  control for the robotic leg with the teststand: 
[image]
In particular I want to create some interesting jumping motions so I can test them on the stand.

### Problem 
I have set up the crocoddyl and been able to run simulation with my urdf model with some modifications I did to the _**"Jumping"**_ of the [quadrupedal_gaits.py](https://github.com/loco-3d/crocoddyl/blob/9e79f80c6d52b2c6c000106570811d449341cad3/examples/quadrupedal_gaits.py) example : 

https://user-images.githubusercontent.com/30003101/129442784-baf327a0-ae41-42e8-8d76-36e3f2a26ae4.mp4

The problem here is that I want to constraint movement of the body to only be along z (vertical) axis, so I could focus on other parts of the problem and maybe make the algorithm work faster. Also here the algorithm Fails since it can not track referenced COM, which shouldn't be moving along x and y axis. 

### I have tried
Without changing urdf description, I have tried to use `pinocchio.JointModelPZ()` when using  `BuildFromUrdf`. I also  decided to keep the `ActuationModelFloatingBase` - I think it should work. If I understood [definition right,](https://github.com/loco-3d/crocoddyl/blob/9e79f80c6d52b2c6c000106570811d449341cad3/bindings/python/crocoddyl/multibody/actuations/floating-base.cpp#L21) it just doesnt try to control the `root_joint` as can be seen in [this line of the code](
https://github.com/loco-3d/crocoddyl/blob/9e79f80c6d52b2c6c000106570811d449341cad3/include/crocoddyl/multibody/actuations/floating-base.hpp#L30). `BuildFromUrdf` did its job correctly regarding defining the number of DOF but when running the robot is not moving at all.  What I am getting can be seen here:

https://user-images.githubusercontent.com/30003101/129443498-b109feb3-fd87-43a7-b7dc-d31cd55bfe9b.mp4

### Check out implementation
You can check it out by running the `fore_jumping.py` in the [z_constraint branch](https://github.com/Legged-robot/control_and_simulation/tree/z_constraint) (dont forget to check out to the **_develop_** branch in the submodule for the right urdf and srdf) and also to the **_z_constraint_** branch in original module. 
You can switch between prismatic and freeflyer by changing the [global variable in `fore_jumping.py`](https://github.com/Legged-robot/control_and_simulation/blob/c1629d4a1bc150d7b82d9667915a1b3df562abbf/fore_jumping.py#L9) to True or False:
```
PRISMATIC = False # freeflyer configuration
PRISMATIC = True # prismatic configuration
```

cmastalli, 2021-08-16:
> Without changing urdf description, I have tried to use `pinocchio.JointModelPZ()` when using `BuildFromUrdf`. I also decided to keep the `ActuationModelFloatingBase` - I think it should work. If I [...]

Yes, you got it. Thanks for reporting the unclear Python documentation. I have fixed in the following PR: https://github.com/loco-3d/crocoddyl/pull/987.

> but when running the robot is not moving at all. What I am getting can be seen here:

It seems you're using a 3d contact model, but this is a 2d motion (i.e., jumping in the frontal plane).
In this case, you're over-constraining the contact dynamics which leads to unsuccessful solutions (as your problem formulation doesn't satisfy the required constraint qualifications).
The quick solution here is to use the 2d contact model.

I will close this issue as it the most help that I could provide you.

I am happy that you have chosen to use our software.
We would appreciate any acknowledgment to our work.

closed 2021-08-16 by cmastalli

VasilyRakche, 2021-08-17:
Hey @cmastalli, thanks for the answer - that was causing the problem. However, I am not sure how to define the cost for **2d contact model**, all `ResidualModelContact*` that I found resonable to use require **3d contact model**. What would be the best way to setup the cost function?

cmastalli, 2021-08-18:
>  However, I am not sure how to define the cost for **2d contact model**

You need to substitute the 3d contact by a 2d one: e.g., using `crocoddyl.ContactModel2d` here: https://github.com/loco-3d/crocoddyl/blob/master/bindings/python/crocoddyl/utils/quadruped.py#L415.

The `ResidualModelContactFrictionCone` and `ResidualModelContactForce` indeed needs to be extended in such a way that they can handle 2d contacts. I could handle this task, but after September 15. Is this might be OK for you?

VasilyRakche, 2021-08-19:
Thanks for the feedback, its hard for me to tell if that will be late or not. The thing is that I expect I would like to experiment with some code and adjusting some math, but it will be hard to do if I dont understand some aspects of it related to a problem setup, robot representation and so on. Better solution would be to guide with some links and so on, so I try to do it on my own. I am following with the literature and links from your answer on my other question for now, hope I would able to clarify some concepts.

cmastalli, 2021-08-19:
> Better solution would be to guide with some links and so on, so I try to do it on my own.

I understand you, there are notions used in Crocoddyl that are not straightforward to grasp them.
Unfortunately, transferring the theoretical and software knowledge requires significant supervision.

You might also want to inspect the prepared material for recent summer school here: https://memory-of-motion.github.io/summer-school/materials

In any case, I would try to enable 2d contact as soon as possible or would be keen to review your code (through a PR to the devel branch).

I hope you can grasp things quickly :)