loco-3d/crocoddyl #988: Documentation on Mathematical background of implemented functions

VasilyRakche, 2021-08-17:
Hey,
 
I really like your software, and want to go deeper so I can modify, adjust and create new functions for my needs, but the bottleneck is that I can not understand the equations used. For example, why the ContactModel3DTpl calc and calcDiff looks the way it is, or what about ResidualModelContactFrictionConeTpl. And that with many other things I stumble upon. I understand that for you its probably obvious, but its not like that for me. I would be happy to dig in the math behind, but dont know where to look for. 

Best,

cmastalli, 2021-08-18:
Hey @VasilyRakche 

We appreciate that you found our software useful.
We are aware of the lack in the documentation, this is due to lack of resources.
Our priorities are currently focused on improving many aspects on the software

I assuming you have read our Crocoddyl paper (written in the README file).
But I am also aware that this does not provide the requested details.

Generally speaking, `calc` computes the contact Jacobian and acceleration drift (as explained in Eq. 2 of Crocoddyl paper).
Instead, `calcDiff` computes the partial derivatives of the acceleration drift.
Note that this last term is used to compute the derivatives of the contact dynamics (Eq. 5 of Crocoddyl paper).

The rest of the lines are related with the needed Spatial-algebra operations to compute the local acceleration into the contact frame. The same for the derivatives where the skew matrices act as differentiation operators (very known operator in screw theory).

Explaining the Spatial algebra goes beyond the scope of Crocoddyl.
However, I suggested you to read this book: https://www.springer.com/gp/book/9780387743141

Finally, the residual for the friction cone is a linearized friction cone, which depends on the contact forces.
Again, the contact forces and their derivatives are computed as described in Eq. 2 and 5, respectively.

I hope this explanation might help you to keep digging in this topic.
I will close this issue, but do not hesitate to ask further questions regarding my comments.

closed 2021-08-18 by cmastalli

VasilyRakche, 2021-09-07:
Hey, 
I went back home for couple of weeks, so I took relaxed approach with reading some literature and trying to understand some concepts. I will try to catch up ;)
Thanks a lot for the literature suggestions and for the explanations, they really helped me out. I understood how you use skew matrices and spatial algebra, so I have mostly questions concerned to the physics part (not mathematics :) ) For future reader of this thread, I would suggest checking [Modern Robotics Kevin M. Lynch and Frank C. Park](http://hades.mech.northwestern.edu/images/7/7f/MR.pdf) book first to understand some basics of Spatial-algebra. 

I was analyzing [contact-3d.hxx ](https://github.com/loco-3d/crocoddyl/blob/master/include/crocoddyl/multibody/contacts/contact-3d.hxx) mainly and I have several questions:

1. Should you use `nq` instead of `nv` whenever you refer to [`d->da0_dx.leftCols`](https://github.com/loco-3d/crocoddyl/blob/9e79f80c6d52b2c6c000106570811d449341cad3/include/crocoddyl/multibody/contacts/contact-3d.hxx#L85)?
2. Isnt [`d->fXj.template topRows<3>() * d->v_partial_dq`](https://github.com/loco-3d/crocoddyl/blob/9e79f80c6d52b2c6c000106570811d449341cad3/include/crocoddyl/multibody/contacts/contact-3d.hxx#L96) the same as: 
```
d->fXjdv_dq.template topRows<3>() 
```
3. Why you first calculate all the velocities in joint frame [here](https://github.com/loco-3d/crocoddyl/blob/9e79f80c6d52b2c6c000106570811d449341cad3/include/crocoddyl/multibody/contacts/contact-3d.hxx#L77) and then transforming them to the contact frame with `fXj`, would there be logical to have formula in pinnochio to compute **Frame acceleration derivatives**?   
4. Can you check if my equations are correct for the 3d contact - checking if my understanding of the code is correct (I couldnt think of a better way of writing formulas, please let me know how to do it for the future). I am not explicitly writing that its in contact frame, only in case its otherwise I mark it:
[image]

5. Can you send me some references to understand why the derivatives equations look the way they do - how robot physics relate to this representation (I checked contact-2d and I see that you are just extracting the **x** and **z** axis, so I probably wont explicitly need to understand this to define cost functions, its just that if I came till here, I would like to get to know this). I think I understand the formula for `a0` in `calc`, classical formula for the acceleration of the body with added PD gains.