loco-3d/crocoddyl #989: ImportError: libhpp-fcl.so: undefined symbol

gianluigigrandesso, 2021-08-24:
Hi, 

I'm having troubles with the installation of crocoddyl on a Ubuntu 18.04 machine with Python 3.6.
I've actually installed it without errors through robotpkg, but when I try to import it I get following error:

`File "/opt/openrobots/lib/python3.6/site-packages/crocoddyl/__init__.py", line 1, in <module>
     from .libcrocoddyl_pywrap import *
ImportError: /opt/openrobots/lib/libhpp-fcl.so: undefined symbol: _ZN8octomath6Pose6DC1ERKS0_` 

This is the list of my robotpkg packages:

> ii  robotpkg-collada-dom                       2.4.0r1                                          amd64        C++ object representation of a COLLADA XML instance document
ii  robotpkg-eiquadprog                        1.2.2                                            amd64        algorithm of Goldfarb and Idnani for the solution of a QP
ii  robotpkg-example-robot-data                3.6.1                                            amd64        Robot URDFs for benchmarking and developed examples
ii  robotpkg-hpp-fcl                           1.7.5                                            amd64        Humanoid Path Planner (collision detection and distance computations)
ii  robotpkg-libccd                            2.0                                              amd64        collision detection
ii  robotpkg-octomap                           1.9.0                                            amd64        An Efficient Probabilistic 3D Mapping Framework Based on Octrees
ii  robotpkg-omniorb                           4.2.4                                            amd64        ORB that implements the 2.6 specification of the CORBA
ii  robotpkg-openscenegraph                    3.6.5r3                                          amd64        open source high performance 3D graphics toolkit
ii  robotpkg-pinocchio                         2.6.3                                            amd64        Efficient rigid body dynamics
ii  robotpkg-py36-crocoddyl                    1.8.1                                            amd64        Optimal control library for robot under contact sequence
ii  robotpkg-py36-eigenpy                      2.6.7                                            amd64        Eigen bindings to python
ii  robotpkg-py36-example-robot-data           3.10.0                                           amd64        Robot URDFs for benchmarking and developed examples
ii  robotpkg-py36-hpp-fcl                      1.7.5                                            amd64        Humanoid Path Planner (collision detection and distance computations (python bindingss))
ii  robotpkg-py36-omniorbpy                    4.2.4                                            amd64        Robust high-performance CORBA ORB for Python
ii  robotpkg-py36-pinocchio                    2.6.3                                            amd64        Efficient rigid body dynamics (python bindings)
ii  robotpkg-py36-qt4-gepetto-viewer+doc       4.10.1r2                                         amd64        Humanoid Path Planner (Simple viewer based on OpenSceneGraph)
ii  robotpkg-py36-qt4-gepetto-viewer-corba     5.5.1                                            amd64        Humanoid Path Planner (Corba server for gepetto-viewer)
ii  robotpkg-py36-quadprog                     0.1.6r1                                          amd64        Solve a strictly convex quadratic program
ii  robotpkg-qt4-osgqt                         3.5.7r2                                          amd64        open source high performance 3D graphics toolkit
ii  robotpkg-urdfdom                           1.0.3                                            amd64        The URDF (U-Robot Description Format) library
ii  robotpkg-urdfdom-headers                   1.0.4                                            amd64        The URDF (U-Robot Description Format) headers

There seems to be troubles with the library hpp-fcl, so I've tried to reinstall it through apt install --reinstall but nothing has changed. Could anyone help me solve this problem please?
Thanks!

jcarpent, 2021-08-24:
If you need a quick fix, you can rely on conda installation:
```
conda install crocoddyl -c conda-forge
```

nim65s, 2021-08-24:
Hi,

This issue comes from multiple liboctomap on your system and in your environment, probably from ros-melodic. Can you try to remove `/opt/ros/melodic/lib` from your `LD_LIBRARY_PATH` and check if this fix your issue ?

gianluigigrandesso, 2021-08-25:
Thanks @jcarpent and @nim65s for the suggestions. I would like to avoid using conda and unfortunately not having `/opt/ros/melodic/lib` in my `LD_LIBRARY_PATH` doesn't fix the issue. I've also tried to reinstall robotpkg-octomap but nothing has changed.

nim65s, 2021-08-25:
Ok, can you show us the output of `ldd /opt/openrobots/lib/libhpp-fcl.so` ?

gianluigigrandesso, 2021-08-25:
This is the output:

> 	linux-vdso.so.1 (0x00007ffe79dca000)
	libassimp.so.4 => /usr/lib/x86_64-linux-gnu/libassimp.so.4 (0x00007f8408d72000)
	libqhull_r.so.7 => /usr/lib/x86_64-linux-gnu/libqhull_r.so.7 (0x00007f8408b18000)
	liboctomap.so.1.9 => /opt/openrobots/lib/liboctomap.so.1.9 (0x00007f84088a9000)
	liboctomath.so.1.9 => /opt/openrobots/lib/liboctomath.so.1.9 (0x00007f84086a2000)
	libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f8408319000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f8407f7b000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f8407d63000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f8407972000)
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007f8407755000)
	libminizip.so.1 => /usr/lib/x86_64-linux-gnu/libminizip.so.1 (0x00007f840754a000)
	librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007f8407342000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f8409b86000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f8407123000)

nim65s, 2021-08-25:
Thanks, this looks good, so I can't explain your issue. What about `objdump -x /opt/openrobots/lib/liboctomath.so.1.9 | grep _ZN8octomath6Pose6DC1ERKS0_` ?

gianluigigrandesso, 2021-08-25:
Thank you so much for your assistance  @nim65s . I tried `objdump -x /opt/openrobots/lib/liboctomath.so.1.9 | grep _ZN8octomath6Pose6DC1ERKS0_` but its output is null...

nim65s, 2021-08-25:
ok, then this file has an issue, as on a clean system I get `0000000000004240 g     F .text	000000000000002a              _ZN8octomath6Pose6DC1ERKS0_`.  What about `dpkg -l robotpkg-\*`?

gianluigigrandesso, 2021-08-25:
Ok, got it. The output of `dpkg -l robotpkg-\*` is:

>Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name                                                           Version                              Architecture                         Description
+++-==============================================================-====================================-====================================-==================================================================================================================================
ii  robotpkg-collada-dom                                           2.4.0r1                              amd64                                C++ object representation of a COLLADA XML instance document
ii  robotpkg-eiquadprog                                            1.2.2                                amd64                                algorithm of Goldfarb and Idnani for the solution of a QP
ii  robotpkg-example-robot-data                                    3.6.1                                amd64                                Robot URDFs for benchmarking and developed examples
un  robotpkg-gepetto-viewer                                        <none>                               <none>                               (no description available)
un  robotpkg-gepetto-viewer-corba                                  <none>                               <none>                               (no description available)
ii  robotpkg-hpp-fcl                                               1.7.5                                amd64                                Humanoid Path Planner (collision detection and distance computations)
un  robotpkg-hpp-fcl+doc                                           <none>                               <none>                               (no description available)
un  robotpkg-hpp-fcl04                                             <none>                               <none>                               (no description available)
ii  robotpkg-libccd                                                2.0                                  amd64                                collision detection
ii  robotpkg-octomap                                               1.9.0                                amd64                                An Efficient Probabilistic 3D Mapping Framework Based on Octrees
ii  robotpkg-omniorb                                               4.2.4                                amd64                                ORB that implements the 2.6 specification of the CORBA
un  robotpkg-omniorbpy                                             <none>                               <none>                               (no description available)
ii  robotpkg-openscenegraph                                        3.6.5r3                              amd64                                open source high performance 3D graphics toolkit
ii  robotpkg-pinocchio                                             2.6.3                                amd64                                Efficient rigid body dynamics
un  robotpkg-py27-crocoddyl                                        <none>                               <none>                               (no description available)
un  robotpkg-py27-eigenpy                                          <none>                               <none>                               (no description available)
un  robotpkg-py27-omniorbpy                                        <none>                               <none>                               (no description available)
un  robotpkg-py27-qt4-gepetto-viewer                               <none>                               <none>                               (no description available)
un  robotpkg-py27-qt4-gepetto-viewer+doc                           <none>                               <none>                               (no description available)
un  robotpkg-py27-qt4-gepetto-viewer-corba                         <none>                               <none>                               (no description available)
un  robotpkg-py27-qt4-gepetto-viewer-corba+doc                     <none>                               <none>                               (no description available)
un  robotpkg-py27-qt5-gepetto-viewer                               <none>                               <none>                               (no description available)
un  robotpkg-py27-qt5-gepetto-viewer+doc                           <none>                               <none>                               (no description available)
un  robotpkg-py27-qt5-gepetto-viewer-corba                         <none>                               <none>                               (no description available)
un  robotpkg-py27-qt5-gepetto-viewer-corba+doc                     <none>                               <none>                               (no description available)
ii  robotpkg-py36-crocoddyl                                        1.8.1                                amd64                                Optimal control library for robot under contact sequence
ii  robotpkg-py36-eigenpy                                          2.6.7                                amd64                                Eigen bindings to python
[... cut ...]
ii  robotpkg-qt4-osgqt                                             3.5.7r2                              amd64                                open source high performance 3D graphics toolkit
ii  robotpkg-urdfdom                                               1.0.3                                amd64                                The URDF (U-Robot Description Format) library
ii  robotpkg-urdfdom-headers                                       1.0.4                                amd64                                The URDF (U-Robot Description Format) headers

nim65s, 2021-08-25:
Ok, this ` robotpkg-octomap 1.9.0` looks like it's the issue. The current version is 1.9.6. Can you `sudo apt update && sudo apt upgrade` ?

cmastalli, 2021-08-25:
You have a messy setup.
You should not install packages with Python 2.7 or 3.6 (e.g., `robotpkg-py27-crocoddyl` and `robotpkg-py36-crocoddyl`).

Please use a single Python standard, and I suggest you to choose your distro standard (i.e., Python 2)

gianluigigrandesso, 2021-08-25:
Thanks a lot @nim65s , the issue was exactly the old version of  `robotpkg-octomap`. I've updated it to `robotpkg-octomap 1.9.6` and now everything works, so I'll close the issue. Thanks again!

closed 2021-08-25 by gianluigigrandesso