loco-3d/crocoddyl #997: Best practices for building and debuging new code

VasilyRakche, 2021-09-09:
Hey

I want to start making some changes in [contact-friction-cone.hxx](https://github.com/loco-3d/crocoddyl/blob/9e79f80c6d52b2c6c000106570811d449341cad3/include/crocoddyl/multibody/costs/contact-friction-cone.hxx) (to introduce 2d-contact) and test them out, but I am not sure what would be the best way to build it and debug the code. 

Currently I am using the latest master which I built in Debug mode with `-DCMAKE_BUILD_TYPE=Debug` flag. But that takes too long and my computer is barely able to handle it. Couldn't imagine doing it every time I make some change. 

Also, debugging from python isn't that pleasant experience. I assume you do it natively in c++. Just wondered if you have some tips there as-well.

cmastalli, 2021-09-09:
Hey @VasilyRakche!

First of all, you need to start to develop from the latest devel branch. Second, there is no choice, if you want to debug your code then you need to compile in debug mode (this is c++).

But, what I suggest you is to don't develop scientific code in c++ first. You can do it only if you really understand the math.
Saying that, you might want to start prototyping in Python.

cmastalli, 2021-09-09:
I will close this thread as this is not an issue in Crocoddyl.

closed 2021-09-09 by cmastalli

VasilyRakche, 2021-09-10:
Hey @cmastalli , 

I planned to do it in the reference of Issue #988. It took me some time to roughly understand math, now I wanted to try out some things. I understand that build will be checking to rebuild changes only, but that takes lots of time on my laptop. I tried rebuilding it with smaller changes and my RAM usage reached the limits quite quickly (I have i5, 16Gb ram, Nvidia 1050 graphics). Is there a way I can turn off some things, or do something to speed it up, otherwise this is really unbearable for development. Thats why I thought I must be doing something wrong.

nim65s, 2021-09-10:
I think the best way would be to install a pre-built version of crocoddyl, and develop another package on top of that, to compile just what you need, in DEBUG. here is a (slightly out of date) example: https://github.com/Gepetto/example-adder/commit/512ecfe7ec5ecd23192859049a3cd7abb11309b1

cmastalli, 2021-09-10:
Please also read this webpage: https://github.com/loco-3d/crocoddyl/wiki/Developer-Guide