loco-3d/crocoddyl #999: True hessians for CostModelResidual

gfadini, 2021-09-16:
Hi,
I noticed that the ```CostModelResidual``` is using the Gauss-Newton approximation of the hessians by default.
I guess that this also holds for all the subclasses, so basically almost all the built-in ```CostModels```.
I was wondering if this is an explicit design choice and if the ```CostModelAbstract``` be deprecated any time soon.
I ask this because the ```CostModelAbstract``` allowed to freely set the ```data.Lxx```, ```data.Lxu``` and ```data.Luu``` from its functions. Switching will imply to lose this feature since the ```ResidualModelAbstract``` computes just ```data.Ru``` and ```data.Rx``` which then are used in the aforementioned approximation.
This may not be ideal in general, especially when using custom cost functions.

proyan, 2021-09-16:
The idea is to not deprecate CostModelAbstract, but other existing cost models. You should still be able to develop on top of CostmodelAbstract

gfadini, 2021-09-16:
Thanks @proyan for the quick answer, that's clear enough

closed 2021-09-16 by proyan

cmastalli, 2021-09-16:
> I noticed that the `CostModelResidual` is using the Gauss-Newton approximation of the hessians by default.

This is correct, but there is something more. `CostModelResidual` depends on the notion *activation models*.
These models, as you might know, assume a diagonal structure in their Hessians.
This particular choice was made to boost the algorithm performance.