loco-3d/crocoddyl #1001: Help with formating source code with docker image

VasilyRakche, 2021-09-20:
Hi, 
I think the docker image is really convenient tool for checking, but I am having a problem when trying to check the repo (it modifies a lot of files in the repository for some reason). I am following the [Developer guide](https://github.com/loco-3d/crocoddyl/wiki/Developer-Guide) and using the latest docker image. You can find short video of what I am getting, would really appreciate some help

https://user-images.githubusercontent.com/30003101/133973382-a49fba27-1a38-43d0-9931-8818a2759dac.mp4

.

nim65s, 2021-09-20:
Hi @VasilyRakche,

I updated the image with new linters, but this repository was not yet updated with it, and the developer guide is missing that information. 

Could you add `--clang-6` to the end of the `docker run` command line, and try again ?

VasilyRakche, 2021-09-20:
Thanks for the feedback, adding the `--clang-6` worked! Just to be careful, it didnt work when I was adding it to the command indicated in [Developer Guide](https://github.com/loco-3d/crocoddyl/wiki/Developer-Guide):
```
docker run --rm -v $PWD:/app -w /app -it gepetto/linters --clang-6
```
But it did work when I added to the command from [DockerHub](https://hub.docker.com/r/gepetto/linters):
```
docker run --rm -v $PWD:/root/src -it gepetto/linters --clang-6
```

closed 2021-09-20 by VasilyRakche